MarlinFirmware/Marlin #7083: RFE: UBL Graduated Bed Compensation

Andrei-Pozolotin, 2017-06-17:
this is enhancement request to incorporate Graduated Bed Compensation into UBL:

http://reprap.org/wiki/Graduated_bed_compensation

> The second approach applies the unevenness in the bed to the amount of extrusion done in the first one or two layers. In this procedure the Z axis remains level, but more material is extruded in the "troughs" and "valleys" of the measured mesh, while less material is extruded at the "peaks." The result is similar to what you get if you let liquid settle anyplace on Earth. The bottom may be as uneven as you like, but the surface is nice and flat, and should be as smooth as glass.

mperdue, 2017-06-17:
I would think that setting a reasonable fade height would result is the same end results.

Roxy-3D, 2017-06-17:
Yes...   Give the Fade Height option a try.   I think you will find it does exactly this.

Andrei-Pozolotin, 2017-06-18:
thank you
so far I can not measure / see the difference when I introduce ```M420 Z0.4; first 2 layers of 0.2```
is there a way to enable logging to see how current active ```G29 T``` mesh values
are being applied during first 2 layers of print, up to the ```Fade Height 0.4```, 
when all must be even (and I presume, show flat/zeros in the logs)?

Roxy-3D, 2017-06-18:
The fade happens after the Z-Correction is calculated.   The fade adjustment happens according to:
```cpp

      /**
       * This function sets the Z leveling fade factor based on the given Z height,
       * only re-calculating when necessary.
       *
       *  Returns 1.0 if planner.z_fade_height is 0.0.
       *  Returns 0.0 if Z is past the specified 'Fade Height'.
       */
      #if ENABLED(ENABLE_LEVELING_FADE_HEIGHT)
        static inline float fade_scaling_factor_for_z(const float &lz) {
          if (planner.z_fade_height == 0.0) return 1.0;
          static float fade_scaling_factor = 1.0;
          const float rz = RAW_Z_POSITION(lz);
          if (last_specified_z != rz) {
            last_specified_z = rz;
            fade_scaling_factor =
              rz < planner.z_fade_height
                ? 1.0 - (rz * planner.inverse_z_fade_height)
                : 0.0;
          }
          return fade_scaling_factor;
        }
      #else
```
If you have the Fade Height set at 10.0 mm and the nozzle is currently at 9.0 mm, you will get 10% of the full Z-Correction added to the nozzle's position.

FHeilmann, 2017-06-19:
I think what he means (wants?) is that the fade is not only applied to the z-axis, but also to the e-axis, so more  material is extruded in the regions where the nozzle is further away from the previous layer due to z-fade and less material is extruded in regions where the nozzle is closer to the previous layer due to z-fade.

mperdue, 2017-06-19:
@FHeilmann Yes, I think you are right. The thing is, in the final print I don't see where it will make a noticeable difference.

FHeilmann, 2017-06-19:
@mperdue because this depends on how uneven your bed is in relation to your z-fade-height.

If you have an "unevenness" of 1mm and your fade-height is 5mm at a 0.2mm layer height, the worst deviation will be 0.04mm between two layers, which you're very unlikely to see given the inaccuracies of FDM printers.

If you set your fade-height to 1mm, two layers may be as far as 0.2mm apart, which you will notice right away.  

The key here is to have a z-fade height sufficiently high so the unevenness of your bed can be corrected over a large amount of layers, given your layer-height and the maximum deviation between two points on your bed.

Roxy-3D, 2017-06-19:
There are self correcting variables in the equation.  If the nozzle is too close to the previous layer, slightly less plastic is extruded.   And the opposite is true too.    What ends up happening is you can't really see difference.   I'm sure there is more or less internal stress on the part as the plastic cools and contracts.  But it is very hard for the naked eye to see the difference between layers where correction is happening.

closed 2017-11-03 by thinkyhead