MarlinFirmware/Marlin #7084: Ramps reset when move z axis (or homing) after enable LCD

ZeeVoid, 2017-06-17:
Hi,
strange behaviour after enabling REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.
Homing Z axis reset m256+ramps  aprox. after few second (5 or 10) after z axis start moving. 
Every LCD type i enable from firmware do same random reset.

I tried with LCD plugged in/unplugged. Same behaviour .
My machine:
CoreXY
mega2560+Ramps 1.4+drv 8825+REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
(dual Z steppers)

Any idea?

martindb, 2017-06-17:
If your drv8825 are in 1/32 steps you are using to much cpu to move your z axis. Try a lower feedrate. Or put your drv in 1/16 steps.

ZeeVoid, 2017-06-18:
Thanks for your prompt response.
Indeed modify FEEDRATE solved my problem.
From
#define HOMING_FEEDRATE_Z  (4*60) //default feedrate in Marlin v 1.1.3
to 3...3.5 max= no problem
after 3.5 causing reset
Thanks again

ZeeVoid, 2017-06-18:
Also need to modify:
from
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 } //default Z feedrate=5
to
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 3.5, 25 }

Bob-the-Kuhn, 2017-06-18:
Something else is wrong/unusual.  At 3.5 mm/s (210 mm/m feedrate) the Z axis is SLOWWW.  Are you OK with that?

What is your:
DEFAULT_AXIS_STEPS_PER_UNIT
microstep setting on the stepper drivers

What flavor of Marlin are you using?   [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x) is the latest.  Make sure you transfer all your machine specific items to the new config files.  That's been known to cause unusual problems.

Please ZIP up your config files & drop the ZIP on your reply.

ZeeVoid, 2017-06-19:
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 320, 320, 10240, 200 }
1/32 on 400 steps / nema17
Marlin 1.1.3

for some reasons my zip files is not accepted

Bob-the-Kuhn, 2017-06-19:
Some days GIT can be ornery.  When that happens I use a service like DROPBOX to get a URL.

10240 steps per unit is the reason that you have to run such slow Z feedrates.  

I recommend you change your microstepping to 4 or 8.  Microstepping doesn't increase your positional resolution.  All it does is make the steppers quieter.

ZeeVoid, 2017-06-19:
Thanks for the reply.
Seems arduino can`t handle so much calculations.
 I will test later 1/16 1/8 1/4 microstepps.

ZeeVoid, 2017-06-20:
Switched from 1/32 to 1/16 works ok with default feedrate/max_feedrate.

Thanks again for help.

closed 2018-02-07 by thinkyhead