MarlinFirmware/Marlin #7093: Bilinear auto leveling does not work

Hamulus, 2017-06-19:
I configured Marlin 1.1.3 for Autoleveling with fix mounted sensor connected to Z endstop pin.

I have the following settings in my config:

```
#define X_PROBE_OFFSET_FROM_EXTRUDER 25  
#define Y_PROBE_OFFSET_FROM_EXTRUDER 20  
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]

#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points

#define AUTO_BED_LEVELING_BILINEAR

#define GRID_MAX_POINTS_X 7
#define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

// The Z probe minimum outer margin (to validate G29 parameters).
#define MIN_PROBE_EDGE 10

// Set the boundaries for probing (where the probe can reach).
#define LEFT_PROBE_BED_POSITION 25
#define RIGHT_PROBE_BED_POSITION 150
#define FRONT_PROBE_BED_POSITION 30
#define BACK_PROBE_BED_POSITION 180
```
I Enter M111 S38 to enable LEVELING+INFO+ERROR debugging. Then I enter G28 to home all axes and then enter G29 to start auto leveling. The auto leveling starts successfully and after finishing I see the scan grid in console:

```
12:41:35.983 : Bilinear Leveling Grid:
12:41:35.983 : 0      1      2      3      4      5      6
12:41:35.983 : 0 -3.127 -3.405 -3.405 -3.292 -3.595 -3.487 -3.537
12:41:35.983 : 1 -3.110 -3.367 -3.337 -3.220 -3.470 -3.350 -3.365
12:41:35.983 : 2 -3.138 -3.367 -3.330 -3.215 -3.442 -3.345 -3.385
12:41:35.984 : 3 -3.013 -3.225 -3.182 -3.047 -3.225 -3.132 -3.150
12:41:35.984 : 4 -2.970 -3.165 -3.097 -2.972 -3.160 -3.045 -3.065
12:41:35.984 : 5 -2.875 -3.075 -3.005 -2.847 -2.990 -2.872 -2.875
12:41:35.984 : 6 -2.680 -2.845 -2.755 -2.615 -2.753 -2.617 -2.622
12:41:35.985 : G29 uncorrected Z:10.00
12:41:35.985 : corrected Z:12.85
12:41:35.985 : <<< gcode_G29
12:41:35.985 : X:120.00 Y:160.00 Z:12.85 E:0.00 Count X:9600 Y:12800 Z:4000
12:41:35.985 : current_position=(120.00, 160.00, 12.85) : sync_plan_position
So the autoleveling scanning seems to be successfull.
```

Naturally I **DO NOT** enter G28 after the scanning. I enter G0 Z1 to down nozzle almost to table. But when I enter for example G0 X0 Y0 and look at level and then I enter G0 X150 Y150 I see the big difference between nozzle levels relatively to table in these 2 points. So it seems that height compensation does not work. I expect that Z axis would lift up or down depending on auto leveling results but Z motor do not work when I move X/Y.

By the way I tried 3 point autoleveling. It was pretty rough but Z axis corrected it's level when I moved X ans Y axes using G0 commands. And to avoid questions "why my table is inclined so much?" I deliberately inclined the table to be sure that auto leveling works. Btw I tried different Z_PROBE_OFFSET_FROM_EXTRUDER so the current 0 setting is not actual but situation is the same with any value of this option.

So what I am doing wrong?

BlackLightUK, 2017-06-20:
Generating and storing the mesh is one thing - enabling it is another. Are you doing an M420 S1 to enable the bed levelling after you've got the mesh ?

Hamulus, 2017-06-20:
M420 S1 helped me, thanks
btw I have suggestion to automatically enable bed leveling after G29 because it is very logical to expect it:)

BlackLightUK, 2017-06-20:
UBL levelling remembers the state of the levelling system before a g28 and applies it if necessary afterwards. It's the only one that does it and it's considered kinda controversial as is (but have no idea why)

closed 2017-11-03 by thinkyhead

rsemenoff, 2017-12-02:
I've got version 1.1.6, and g29 does nothing even with enabling it via m420 s1. When I g1 to a corner then g1 z0.4 it is way off (about a mm). When I looked at the bed readings they look about right.  As the op, I made the bed intentionally a little sloped...

Roxy-3D, 2017-12-02:
What does your mesh look like?   G26 should be working (as of today!!!) with Bi-Linear.   A G26 pattern would tell us a lot about how well your mesh describes the bed's surface.

rsemenoff, 2017-12-02:
Unfortunately I can't extrude right now because I messed up my Bowden tube clamp... But I will try out the g26 with bilinear and video it.

rsemenoff, 2017-12-03:
I suspect my probe is less accurate at the edges, so I tried probe_manually and lcd_bed_leveling as a sanity check. When I go through the routine it's not following the probe boundaries. Its going off the bed (v1.1.6) Also when going to the next point, it doesn't stop at the limit, it's going to where it's assuming z=0 is at...pushing the nozzle into the bed at some points..is that what's supposed to happen?  I'd appreciate if someone can confirm that manual bed probing does work...with bilinear.  
Thanks

infernocs, 2017-12-11:
I'm having big problems with bilinear and my 3dtouch, seems whatever I do the left side is too high off the bed and the right side is too low. (scraping the bed) Seems like the compensation has a +- multiplier built in that dampens the effect leaving the compensation short on one side and doing too much on the other. Affecting both X and Y axis. 
I see lots of people on fb groups complaining on this same issue, the fault in height is about 0,1mm/100mm
I'm running Anet A8 with stock board and mechanics, been testing this for a week now to see that the problem is not mechanical or something in the config files.
If any of the developers could shed some light onto how to calculations are made.

-Carl-

thinkyhead, 2017-12-16:
@infernocs See if `bugfix-1.1.x` still exhibits the issue.

infernocs, 2017-12-19:
I installed 1.1.7 yesterday, would that have the upgrades?

thinkyhead, 2017-12-25:
> It's the only one that does it and it's considered kinda controversial as is (but have no idea why)

Because the `G28` command is _**specified**_ to turn leveling off and leave it off. It's the "homing" command, not the "home and re-enable my pet feature" command.

> Seems like the compensation has a +- multiplier built in that dampens the effect leaving the compensation short on one side and doing too much on the other.

It has no such thing. Marlin does a very simple interpolation between Z values in a mesh, and doesn't care where on the bed it is. There's nothing in the code that would cause it to be higher on "one side" because it just doesn't care what "side" it's on.

If you're very concerned about it, do some probing and take careful note whether the Z compensation at various points matches the mesh.

- `M420 V` will print out your mesh.
- With leveling enabled, move around the bed with `G1` and then use `G29 Q` to inquire about the leveling compensation being applied at the current position.

If you discover that the nozzle is moving farther or other than the distance reported by `G29 Q`, or if the values reported by `G29 Q` seem out of sync with the mesh reported by `M420 V` then it would merit a closer look. So far in my testing I have not seen anything unusual.