MarlinFirmware/Marlin #7099: discount reprap full graphic display issue

Gigglebear1, 2017-06-20:
im using u8glib 1.18.0 and i have tried 1.19.0 with no luck. trying to figure out if this is a display issue or a firmware issue.

here the symptoms. disaply starts and displays proper for about 10-15 min then it does something like this

[image]

then after about 2 hours it goes complete bonkers and looking like this 

[image]

i downloaded the u8glib 1.18.0 and 1.19.0 from the website listed in marlin 1.1.3.

only thing that points me to a firmware issue is that it is fine for 10-15 minutes

i have the full graphic discount uncommented as well as the sdsupport(which works fine even whe display is not displaying properly )

please help and thanks you in advance. 👍

Bob-the-Kuhn, 2017-06-20:
Chances are good that you have a bad LCD.  My unit looked like your first photo after a couple of hours, then 1 hour and eventually after 2-3 minutes and now dead.  A new unit works just fine.

This was with bugfix-1.1.x a month ago without downloading/using any newer u8glib revisions than what is in the Arduino IDE.

pmbroth, 2017-06-21:
looks like a bad lcd, I have the same one connected along with the mkstft 32 and do not have any issues.

luisqcosta, 2017-06-22:
I have the exact same issue with my lcd. Most likely it is a defective one... I tried every combination of libs in the hope that it would fix the issue, but no success. Anyway, it came from a Chinese vendor who refunded me (lucky me!). Time to get a new one and play the lcd lottery again!

closed 2018-02-09 by thinkyhead