MarlinFirmware/Marlin #7101: RFE: more user menus

Andrei-Pozolotin, 2017-06-20:
here
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration_adv.h#L1246
there is a limit of 5 entries

is it too much to ask for 5 more? :-)
or, perhaps, even an array?

Roxy-3D, 2017-06-20:
Let's just see how useful 5 entries is.   Most people should only need 1 or 2.  

Now let's have a more contentious discussion...   I added the Double-Click for Z-BabyStepping to make it very easy (and quick) to jump into the Z-BabyStep menu.    I don't want to give that up.   But it would be nice to have a quick way to jump into this menu.

Maybe we could do something like 2 quick clicks gets you to this menu.  And if a 3rd click comes in within 1/2 second the LCD Panel jumps to Z-BabyStepping ?

BlackLightUK, 2017-06-20:
@Roxy-3D For some reason, whenever I doubleclick it tends not to go to z-babystep, but when I triple-click, it goes there every time.... No idea why, but just thought i'd bring it up.

Roxy-3D, 2017-06-20:
There is some lag detecting the Encoder Wheel clicks.   The longer you have the LCD_UPDATE_INTERVAL option set for...  The slower you have to click the Encoder Wheel for it to be seen.   (And you don't want to make that number super short because the LCD Panel already eats up a lot of CPU cycles.)    Please try clicking a little bit slower.    You can't double click as fast as you would with a mouse button.

Tannoo, 2017-06-23:
Or, clicking too fast. Slower is better. Not a computer-like double-click.

benlye, 2017-06-23:
> Now let's have a more contentious discussion... I added the Double-Click for Z-BabyStepping to make it very easy (and quick) to jump into the Z-BabyStep menu. I don't want to give that up. But it would be nice to have a quick way to jump into this menu.

Another way 'jumps' might be added - is it possible to detect a long press on the encoder button?  If you do it now it gets detected as multiple clicks, and the LCD cycles through baby stepping then back to the info screen.  (Of course this means with a bit of practice (and good timing) you can currently use a single long press to get to babystepping.)

boelle, 2019-02-19:
i would say that 2 menu's is enough, i dont even use them

who does not agree? :-D

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-12 by Andrei-Pozolotin