MarlinFirmware/Marlin #7102: Non G29 Z offset?

Sungod3000, 2017-06-20:
Hi,

Im looking for a software z offset that isnt connected to G29. As far as I understand it M851 only works in conjuction with G29.

I tried M206 but the command never gets confirmed and it also shows up empty in the eeprom settings. 
Should autobedlevel be completly deactived in order for M206 to work or is there a different setting or Gcode that has this z offset function?

Cheers

Roxy-3D, 2017-06-20:
What are you trying to accomplish?

Sungod3000, 2017-06-20:
Im printing on different materials like paper and cardboard that I tape to my normal bed. To not always have to move the z endstop (which has no simple adjustment screw) I would like to offset the Z axis by software.

Roxy-3D, 2017-06-20:
But isn't that the intended purpose of a Z-Probe?  It doesn't really care where the bed is...  It just finds it and sets the zero point from that information?

Sungod3000, 2017-06-20:
Im working on that seperatly but I havent had much succes with that, so for now I work without z probe, just manual levelling.

CantSayIHave, 2017-06-21:
I believe you're looking for M212. You can do M212Z-0.1 to offset the extruder 0.1mm below measured 0. I use this every time I have a material change on my printrbot

CantSayIHave, 2017-06-21:
To add to that, you can check your offsets with M114 and store them with M500

Sungod3000, 2017-06-21:
yes i saw M212 as well but its printerbot specific.

thinkyhead, 2018-02-07:
Either `M206` or `G92` will offset the coordinate space.

closed 2018-02-07 by thinkyhead