MarlinFirmware/Marlin #7104: Need to stop Z from raising before homing X and Y

CantSayIHave, 2017-06-21:
Hi, I'm sorry if the solution to this is glaringly obvious, but Marlin has quite a lot of options and I've been trying to fix this all day.

I have a MakerGear M2, so my z axis moves upward to meet the fixed-z extruder. For some reason, whenever I try to home X and Y, it tries to "raise" the Z by probably 10mm, which on my machine tries to lower the bed past the limit it's already at. The z lift is gravity-based, so it always starts at the bottom. How do I make it stop moving z before homing x and y?

Thank you

Bob-the-Kuhn, 2017-06-21:
Try commenting BOTH of these out.
Z_CLEARANCE_BETWEEN_PROBES
Z_HOMING_HEIGHT

If that doesn't work then try homing in the + direction (`#define Z_HOME_DIR 1`)  and setting up a Z_MAX endstop.  That'll cause Z to home first.  

If you end up homing Z in the + direction than the setting of Z_MAX_POS becomes important. Set it so the Z0 is about 10mm from the nozzle.  The bed leveling will take care of this "gap".

CantSayIHave, 2017-06-21:
Alright, so I came up with a solution before I saw your post, however I don't know if what I did is recommended.

I ended up commenting out this:

    if (home_all || homeX || homeY) {
        // Raise Z before homing any other axes and z is not already high enough (never lower z)
        destination[Z_AXIS] = LOGICAL_Z_POSITION(Z_HOMING_HEIGHT);
        if (destination[Z_AXIS] > current_position[Z_AXIS]) {

          #if ENABLED(DEBUG_LEVELING_FEATURE)
            if (DEBUGGING(LEVELING))
              SERIAL_ECHOLNPAIR("Raise Z (before homing) to ", destination[Z_AXIS]);
          #endif

          do_blocking_move_to_z(destination[Z_AXIS]);
        }
      }

in the Marlin_main.cpp file, which made it stop raising Z before X and Y homing. I also found it would raise Z before Z homing, so I also commented this line:

    do_probe_raise(_Z_CLEARANCE_DEPLOY_PROBE);

from one of the long `#if` `#elif` chains.

both problems are fixed now, and it homes correctly, but do you think I should try the above mentioned solution instead?

Bob-the-Kuhn, 2017-06-21:
The advantage of my proposals is you don't have to modify the code if you ever update Marlin.

I'd say keep what works & investigate the others if you ever update.

closed 2018-02-07 by thinkyhead