MarlinFirmware/Marlin #7105: Compile Error PlatformIO

Grogyan, 2017-06-21:
PlatformIO reports compile errors of 
Missing binary operator before "("

Way I tested this is to open up a fresh copy of Marlin 1.1.3 in PlatformIO
Adjust the temperature sensor from 1 to 2 in Configuration.h this is so the Linter actually shows that there are compile errors
Eg
#if ENABLED(SWITCHING_EXTRUDER)
The "ENABLED" part appears to be unsupported, producing, Missing binary operator before "("  error

p3p, 2017-06-21:
The linter errors, as far as I can tell, are a problem with the PlatofrmIO linter, The project compiles successfully. I'm unsure why the linter can't see the macros.h file, perhaps it is unable to parse preprocessor stringification (stringified?) macros correctly, I'm not sure there is anything we can do about them but I could be wrong.

Grogyan, 2017-06-22:
It will need to be addressed shortly as 32bit support via PlatformIO is just around the corner as the baseline compiler methinks. 
But it also fails the same regardless of board

Roxy-3D, 2017-06-22:
It would seem we should spend some time and get all the lint errors to go away...

thinkyhead, 2017-06-22:
Linters operate on each file individually, so if a given single file can't compile as a standalone unit, the linter will miss some symbols. This is also true of any `.h` files. Marlin has some careful ordering of headers and config files, as enforced by `MarlinConfig.h`. For most files to compile within the linter, they must at least have `MarlinConfig.h` explicitly included.

The new file layout will be favoring `.h` files and retaining the same basic compilation units, rather than (for example) putting every gcode handler into a separate `.cpp` file. This is for expediency. In future, gcode handlers may be in individual `.cpp` files. But this will require exposing all the globals of `Marlin_main.cpp` and various other changes.

The code organization continues to evolve. We can keep in mind ways to make it more linter-friendly as we go forward.

Phr3d13, 2019-06-10:
Can this be closed?

boelle, 2019-06-22:
without knowing i think yes as its seems not much related to marlin itself

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

GadgetAngel, 2019-11-26:
How did you fix this problem? Atom/PlatformIO will generate a firmware.hex file but I get all these errors from LINT how do I fix this

[image]

Please re-open this or tell me how to fix the errors.

Jice128, 2020-03-07:
I got same trouble(( Just install atom and PlatformIO and same mistakes as on skreen above

Vertabreak, 2020-03-07:
fairly sure the use of atom was deprecated in favor of platformio.

thinkyhead, 2020-03-08:
The [Auto Build Marlin](https://marketplace.visualstudio.com/items?itemName=MarlinFirmware.auto-build) extension in VSCode is the way of the future.
[image]

Jice128, 2020-03-08:
Same to me. Tried to use visual code - get no warnings and errors and compile good. Problem solved. Sad that the atom be like this(

thinkyhead, 2020-03-08:
Atom really should get a rewrite. But then it would just end up being VSCode.