MarlinFirmware/Marlin #7106: 1.1.3 EEPROM_SETTINGS G28/Home

jaburges, 2017-06-21:
Hi there, 
I ran into too many issues using other's firmware so decided to compile my own.
Got things up and running on a Tevo Black Widow, but for some reason when I uncomment EEPROM_SETTINGS Autohome from the LCD, or G28 puts the printer into a busy state (echo:busy: processing' but nothing would happen.

If I comment it out, it works fine ?

Roxy-3D, 2017-06-21:
When you uncomment EEPROM_SETTINGS the firmware gets its settings from the EEPROM.  But you probably never put reasonable values there.   Boot up and do a M502 and M500.   The next time you boot things will probably be more reasonable.

jaburges, 2017-06-21:
Thanks - That seemed a little odd as M501 gave all the correct results - but this did the trick!

closed 2017-06-21 by jaburges