MarlinFirmware/Marlin #7110: Printrboard Bleutooth Serial

TunaLatte, 2017-06-21:
Hello,

Firmware only compile when the following headers are added to serial.h after enabling bluetooth serial:
```
#include "MarlinSerial.h"
#include"Arduino.h"
#include"macros.h"

```

Firmware after this modification and compilation works ok.

Bob-the-Kuhn, 2017-06-22:
Try these changes in serial.h.  For me PRINTRBOARD compiled with Bluetooth , compiled & ran without bluetooth and RAMPS compiled & ran just fine.  I don't have the ability to test the Bluetooth option.
```
#include "MarlinSerial.h"  // added
//#include"Arduino.h"  // not needed
//#include"macros.h"  // not needed

#ifdef USBCON
//  #include "HardwareSerial.h"  // not needed
  #if ENABLED(BLUETOOTH)
    #define MYSERIAL bluetoothSerial
  #else
    #define MYSERIAL Serial
  #endif // BLUETOOTH
#else
//  #include "MarlinSerial.h"   // not needed
  #define MYSERIAL customizedSerial
#endif
```

thinkyhead, 2018-02-07:
Now patched in 1.1.x.

closed 2018-02-07 by thinkyhead