MarlinFirmware/Marlin #7111: G33 P7 V1 stops early on bugfix-1.1.x

joris57, 2017-06-21:
G33 P7 V1 stops after 2 iterations at a std dev of 0.44 where previously on the same printer a std dev of 0.016 was achieved (after several more iterations).

thinkyhead, 2017-06-22:
@LVD-AC Do you know what might be going on here?

PatDW, 2017-06-25:
Experiencing the same issue with G33 P3, stops at 2 iterations with std. dev. 0.404. My pull from 06.09.2017 is working fine and continues to probe.

LVD-AC, 2017-06-26:
Sometimes there is a bump in the road towards the ideal solution. Setting the height off by 1mm with M665 Hx.x. and rerun G33 usually helps to get over it... Anyhow in my experience when this happens the solution found at higher the std. dev. is very close to the one found when the routine gets to a 'normal' std. dev. for the printer.

I've been cracking my brain for a while to detect this automatically... In the mean time I could introduce a Fn parameter for that purpose that forces the routine to do at least n iterations.

joris57, 2017-06-26:
I don't think this will solve it. I think something is wrong with the probing (see the issue of the 'shrinking delta'.

Below you'll find the G33 output from the same calibrated printer, the first one from bugfix-1.1.x, the 2nd one from your own 1.1.3 repository. bugfix-1.1.x reports a height discrepancy where 1.1.3 AC finds the printer calibrated (as I would expect).

```
Send: G33 P4 V2
Recv: G33 Auto Calibrate
[...]
Recv: Checking... AC
Recv: .Height:303.60    Ex:-6.34  Ey:+0.00  Ez:-3.01    Radius:99.16
Recv: .Tower angle :    Tx:-0.24  Ty:+0.06  Tz:+0.00
[...]
Recv: .      c:+0.57     x:+0.54   y:+0.54   z:+0.55
Recv: .                 yz:+0.56  zx:+0.57  xy:+0.57
Recv: Iteration : 01                                    std dev:0.556
Recv: .Height:303.01    Ex:-6.34  Ey:+0.00  Ez:-3.01    Radius:99.10
Recv: .Tower angle :    Tx:-0.25  Ty:+0.04  Tz:+0.00
Recv: 
[...]
Recv: .      c:+0.56     x:+0.55   y:+0.56   z:+0.56
Recv: .                 yz:+0.58  zx:+0.57  xy:+0.58
Recv: Calibration OK                                    rolling back.
Recv: .Height:303.60    Ex:-6.34  Ey:+0.00  Ez:-3.01    Radius:99.16
Recv: .Tower angle :    Tx:-0.24  Ty:+0.06  Tz:+0.00
Recv: Save with M500 and/or copy to Configuration.h
```
```
Send: G33 P4 V2
Recv: G33 Auto Calibrate
Recv: Checking... AC
Recv: .Height:303.60    Ex:-6.34  Ey:+0.00  Ez:-3.01    Radius:99.16
Recv: .Tower angle :    Tx:-0.24  Ty:+0.06  Tz:+0.00
[...]
Recv: .      c:+0.00     x:-0.02   y:-0.02   z:-0.01
Recv: .                 yz:+0.01  zx:-0.00  xy:+0.00
Recv: Iteration : 01                                    std dev:0.014
Recv: .Height:303.60    Ex:-6.34  Ey:+0.00  Ez:-3.00    Radius:99.12
Recv: .Tower angle :    Tx:-0.22  Ty:+0.06  Tz:+0.00
Recv: 
[...]
Recv: .      c:-0.01     x:-0.03   y:-0.01   z:-0.01
Recv: .                 yz:+0.01  zx:-0.01  xy:-0.01
Recv: Calibration OK                                    rolling back.
Recv: .Height:303.60    Ex:-6.34  Ey:+0.00  Ez:-3.01    Radius:99.16
Recv: .Tower angle :    Tx:-0.24  Ty:+0.06  Tz:+0.00
Recv: Save with M500 and/or copy to Configuration.h
```

LVD-AC, 2017-06-26:
Did you set a Z-offset? Maybe in bugfix someone changed the way it takes that into account when probing???

LVD-AC, 2017-06-26:
Bug 'fixed' in #7148 by reverting to previous state. But someone needs to check that change from current_position to CURRENT_POSITION_RAW

joris57, 2017-06-26:
Some further testing : looks like the delta observed in subsequent G33 P1 is equal to (z probe offset (as set with M851) minus 0.04).

LVD-AC, 2017-06-26:
Bug fixed more permanent for Delta's; probing did report false results with G29 G33 as well on Delta's. Someone needs to check this on Cartesians as well.

PS: error was home_offset[Z_AXIS]; for Delta's this is DELTA_HEIGHT from Configuration.h - delta_height set with M665 or by G33 calibration; on Cartesians home_offset is set with M206 or M428 and is used for a different purpose but IMO will result in the same Z-error.

LVD-AC, 2017-06-30:
Issue closed?

joris57, 2017-06-30:
Let's assume so. At the moment, I am using your 1.1.3-AC branch. Thanks for looking into this.

closed 2017-06-30 by joris57