MarlinFirmware/Marlin #7112: G33 : no difference between verbosity level V1 and V2 (bugfix-1.1.x)

joris57, 2017-06-21:
Verbosity level V2 (and V0) should give 'probe results', but does not.

thinkyhead, 2017-06-22:
Are you testing the current `bugfix-1.1.x`? The output code in `gcode_G33` appears correct:
```cpp
// print report

if (verbose_level != 1) {
  SERIAL_PROTOCOLPGM(".    ");
  print_signed_float(PSTR("c"), z_at_pt[0]);
  if (_4p_towers_points || _7p_calibration) {
    print_signed_float(PSTR("   x"), z_at_pt[1]);
    print_signed_float(PSTR(" y"), z_at_pt[5]);
    print_signed_float(PSTR(" z"), z_at_pt[9]);
  }
  if (!_4p_opposite_points) SERIAL_EOL();
  if ((_4p_opposite_points) || _7p_calibration) {
    if (_7p_calibration) {
      SERIAL_CHAR('.');
      SERIAL_PROTOCOL_SP(13);
    }
    print_signed_float(PSTR("  yz"), z_at_pt[7]);
    print_signed_float(PSTR("zx"), z_at_pt[11]);
    print_signed_float(PSTR("xy"), z_at_pt[3]);
    SERIAL_EOL();
  }
}
```

joris57, 2017-06-23:
This was tested with the 'then current' bugfix (i.e. 2 days ago).

Sample output below ... 

I  do see output that fits the format of the code above, but would have expected details of each of the probed points, where only one set of values is displayed for each iteration (with no explanation what they mean).

So obviously no a bug, but 'works as designed'. However, could do with some more clarity.

I will close the issue later today, to give you an opportunity to read the feedback.

Thanks anyway for looking into this.
```
Send: G33 P7 V2
Recv: G33 Auto Calibrate
Recv: Checking... AC
Recv: .Height:300.00    Ex:-6.43  Ey:+0.00  Ez:-2.34    Radius:98.85
Recv: .Tower angle :    Tx:-0.27  Ty:-0.15  Tz:+0.00
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: .      c:+1.21     x:+1.40   y:+1.45   z:+1.02
Recv: .                 yz:+1.20  zx:+1.20  xy:+1.53
Recv: Iteration : 01                                    std dev:1.298
Recv: .Height:298.60    Ex:-6.45  Ey:+0.00  Ez:-2.74    Radius:99.18
Recv: .Tower angle :    Tx:-0.39  Ty:-0.15  Tz:+0.00
Recv: 
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: .      c:+1.36     x:+1.39   y:+1.43   z:+1.21
Recv: .                 yz:+1.32  zx:+1.28  xy:+1.47
Recv: Calibration OK                                    rolling back.
Recv: .Height:300.00    Ex:-6.43  Ey:+0.00  Ez:-2.34    Radius:98.85
Recv: .Tower angle :    Tx:-0.27  Ty:-0.15  Tz:+0.00
Recv: Save with M500 and/or copy to Configuration.h
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: ok
```

closed 2017-06-24 by joris57