MarlinFirmware/Marlin #7113: [Support] 2 motor Z-axis leveling switches and probe for homing

mperdue, 2017-06-21:
I seem to remember reading about this here awhile back and would like to find out if it's a doable thing.

I have two motors and drivers on my Z axis. It works very well with my BLTouch for bed leveling but over time the two sides drift away from each other. Normally this is caused by running into something but sometimes I have no clue what the cause is (probably running into something when I'm not there.) Anyway, it would be nice to have two switches set up where each motor could be driven until it's respective switch is tripped to auto reset the main bed level. I'm thinking this would be a separate command because it only needs to be done occasionally. I would still want to use the BLTouch for homing and probing.

So, is this something that is already supported? If yes, how is it enabled? Someone point me in the right direction.

Bob-the-Kuhn, 2017-06-21:
It's called Z_DUAL_ENDSTOPS.  You'll find it in configuration_adv.h

On my machine I home in the + direction with the Z_MIN and the Z_MAX endstops at the top.  I set Z_MAX_POS so that the nozzle is about 10mm above the bed at Z0.

Here's my setup:
```
#define Z_HOME_DIR  1
#define Z_MAX_POS 425
    #define Z_DUAL_STEPPER_DRIVERS
    
    #if ENABLED(Z_DUAL_STEPPER_DRIVERS)
      
      // Z_DUAL_ENDSTOPS is a feature to enable the use of 2 endstops for both Z steppers - Let's call them Z stepper and Z2 stepper.
      // That way the machine is capable to align the bed during home, since both Z steppers are homed.
      // There is also an implementation of M666 (software endstops adjustment) to this feature.
      // After Z homing, this adjustment is applied to just one of the steppers in order to align the bed.
      // One just need to home the Z axis and measure the distance difference between both Z axis and apply the math: Z adjust = Z - Z2.
      // If the Z stepper axis is closer to the bed, the measure Z > Z2 (yes, it is.. think about it) and the Z adjust would be positive.
      // Play a little bit with small adjustments (0.5mm) and check the behaviour.
      // The M119 (endstops report) will start reporting the Z2 Endstop as well.
      
      #define Z_DUAL_ENDSTOPS
      
      #if ENABLED(Z_DUAL_ENDSTOPS)
        #define Z2_USE_ENDSTOP _ZMIN_
        #define Z_DUAL_ENDSTOPS_ADJUSTMENT -11  // use M666 command to determine/test this value
      #endif
```

mperdue, 2017-06-21:
As I said, I don't need to do this every time I home but sometimes it would be nice to be able to verify that the bed is flat. with that in mind, is this process a part of the G28 command and done every time? I'd hate to have to drive my bed to the opposite limit and back every time I start a print- it takes awhile for the bed to move the 300mm or so there and then back again.

Bob-the-Kuhn, 2017-06-22:
Yes, it would be done every time G28 is run.

You might see if it'll work if homing in the - direction.  You'd have to change a few of the defines and disable ENDSTOPS_ALWAYS_ON_DEFAULT.  I haven't tried that method.

mperdue, 2017-06-22:
Thanks, I'll take a look at it.

mperdue, 2017-06-24:
OK, I got sidetracked by a couple printing jobs so I had en't had a chance to look at this yet. But I do have questions?

When you home, does it use the two z axis switches? What about the probe, when does it come into play? I guess I just don't understand the homing process as it is currently implemented with dual motors and end stops.

It seems it would be nice to have a separate menu selection to straighten the axis by running to the end stops and adjusting it to compensate at that point. It doesn't seem like it would require that any offsets be saved, just move each axis motor till the associated end stop trips and zero them there. This could be set up for any axis with two drive motors.

Bob-the-Kuhn, 2017-06-24:
You can home the Z axis with either the probe or the endstops.  

To home with the probe enable` Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`.  The Z_MIN endstop will NOT be used.  You must be homing in the - direction.

To home using the Z endstop enable `Z_MIN_PROBE_ENDSTOP`. 
- If `Z_DUAL_ENDSTOPS` is disabled then only one of the Z endstops are used when homing.  If homing in the - direction then the Z_MIN endstop is used.  If homing in the + direction then the Z_MAX endstop is used.
- If `Z_DUAL_ENDSTOPS` is enabled then two endstops are used.  Homing can be done in either the - or the + direction.  The endstops are now called Z1 and Z2.  When homing the Z axis is moved until either of the Z1 or the Z2 endstops is activated.  The motor for the activated endstop is stopped and the other Z motor continues until the second endstop is activated.  The second Z motor reverses and travels the `Z_DUAL_ENDSTOPS_ADJUSTMENT` distance.  Homing is now completed.

Z1 & Z2 definitions:
- If homing in the - direction then Z1 is the Z_MIN endstop.  Any of the other five endstops can be assigned to Z2.  The re-assigned (Z2) endstop has to be physically attached to one of the Z axis rails.
- If homing in the + direction then Z1 is the Z_MAX endstop.  Any of the other five endstops can be assigned to Z2.  The re-assigned (Z2) endstop has to be physically attached to one of the Z axis rails.

The Z axis is homed last if the direction is -.  It's homed first if the direction is +.

closed 2017-07-07 by mperdue

Generalx5, 2017-10-06:
where do I go to change the direction of the stepper motors for dual Z? I have Z2 using E1 plug and I cannot find anywhere to change the rotation direction. I tried the E1 direction in config.h but that didn't work.

I then looked at config_adv.h and found that the dual X and dual Y had the reverse motor option. However, right below that the Z didn't have it. 

**Here below is an example of the Dual Ys with the invert direction option:**
###     // Dual Y Steppers
    // Uncomment this option to drive two Y axis motors.
    // The next unused E driver will be assigned to the second Y stepper.
    //#define Y_DUAL_STEPPER_DRIVERS
    #if ENABLED(Y_DUAL_STEPPER_DRIVERS)
     // Set true if the two Y motors need to rotate in opposite directions
         #define INVERT_Y2_VS_Y_DIR true
    #endif

**And here is the section for dual Z right after, but without invert direction option as seen above.**
#if ENABLED(DUAL_Z_STEPPER)
  #define Z_DUAL_STEPPER_DRIVERS
#endif

AnHardt, 2017-10-06:
Workaround:
Swap the two connectors of one coil to reverse the steppers direction.

aurion55, 2018-09-26:
> You can home the Z axis with either the probe or the endstops.
>
> To home with the probe enable`Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`. The Z_MIN endstop will NOT be used. You must be homing in the - [...]

Hi Bob,
I was wondering if you can help, I love Marlin but cannot get my head around this Dual Z limit switches.
I'm running BLTouch, plug into End stop min, and would like to add this feature, to my Marlin 1.19, but don't understand the full setup, as it so confusing, can you post what needs to be altered in Config.h and Config.adv.h, please if you have the time.
I've asked Scott Lahteine, but just seem to understand what he talking about.
There needs to be a tutorial on how to do this, to some degree.
I would like to use optical micro switches and make up some flags, so there no interference if the bed move up or down.

dough29, 2018-11-05:
As far as I understand this subject, when having 2 Z drivers, you need to home Z with 2 end stop switches (Z1 & Z2).

When using a bed leveling sensor you'll not be able to put those switches to Z min, because you are using the leveling sensor, so you need to put them to Z MAX.

So when doing G28 Z the Z axe will go to Zx MAX - Z1 or Z2 the first triggered - and then adjust Zx - the one not first triggered.

You'll then be able to UBL with G29.

It looks like, for the moment, it is not possible to home Z1 + Z2 using a leveling sensor.

thinkyhead, 2018-11-05:
> As far as I understand this subject, when having 2 Z drivers, you need to home Z with 2 end stop switches (Z1 & Z2).

Nope. That is optional. A single Z endstop is supported.

> It looks like, for the moment, it is not possible to home Z1 + Z2 using a leveling sensor.

For homing, you are correct. However, `bugfix-2.0.x` now has `Z_STEPPER_AUTO_ALIGN` which adds a `G34` command to align the ends of the X axis to the bed surface using dual Z steppers and the attached probe.