MarlinFirmware/Marlin #7115: No such file or directory #include <LCD.h>

martinotu94, 2017-06-21:
As soo ad I uncomment "//#define LCD_I2C_SAINSMART_YWROBOT" to try to use a LCM1602 ywrobot LCD screen with marlin i obtain this error:

fatal error: LCD.h: No such file or directory
     #include <LCD.h>
                     ^
compilation terminated.

MacOs latest Marlin version and Arduino IDE also
Someone can help. Thanks a lot!

Bob-the-Kuhn, 2017-06-21:
I've been able to reproduce the problem in a Widows environment.

I'll need to do some research before I can propose a solution.

Bob-the-Kuhn, 2017-06-22:
Looks like you need a new "liquidcrystal" library.  I was able to get it to compile once I had the correct one for this card installed.

Go to **https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home** and install that library in Arduino.

I had a devil of a time getting it to work because I needed to clear out all the other "liquidcrystal" libraries from the Arduino search path.  My suggestion is:
1) search your computer for all files that start with **liquidcrystal**.
2) move the **directories** that contain those files to a temporary location.  This temp location can **NOT** be a subdirectory of any of the original paths.
3) Download & install the new library per the instructions in the above link.
4) Compile Marlin
5) if all is well then delete the temp location.

martinotu94, 2017-06-25:
Thank you a lot! i will try as soon as i can but i'm confident it will work.

closed 2018-02-07 by thinkyhead

sj1519, 2021-10-02:
#include <avr/io.h>
#include <avr/interrupt.h>
#include "LCD.h"

ISR(INT0_vect)
{
  LCD_GoToLineOne();
  LCD_DIsplayString(" Hi INT Zero");
}

ISR(INT1_vect)
{
  LCD_GoToLineOne();
  LCD_DisplayString("Hi INT One");
}

int main(void)
{
  GICR |= ( 1 <<INT0);
  GICR |= ( 1 <<INT1);
  MCUCR = 0x0f;
  Lcd_init();
  
  sei();
    while (1) 
    {
    }
}
plz help me error of the code

ellensp, 2021-10-02:
@sj1519 Do your own home work. 
Not Marlin related

jaisonvj, 2022-01-21:
can you please send lcd.h library files

jaisonvj, 2022-01-21:
above link is not working

ellensp, 2022-01-21:
way back machine has lthe files https://web.archive.org/web/20190528143519mp_/https://bitbucket.org/fmalpartida/new-liquidcrystal/downloads/

ellensp, 2022-01-21:
And I believe that https://github.com/MarlinFirmware/New-LiquidCrystal also works.

jaisonvj, 2022-01-21:
[image: Screenshot (32).png]
can you give solution for this please

thisiskeithb, 2022-01-21:
@jaisonvj: This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).