MarlinFirmware/Marlin #7123: UBL: G29 P2 B seems to skip shim measurement, gives high values.

Knifa, 2017-06-23:
Running 1.1.3, having some problems with UBL.

After running `P1` successfully and moving onto `P2`, the shim measurement as described in the [user guide](http://marlinfw.org/docs/features/unified_bed_leveling.html) appears to be skipped for me, plus some other weirdness.

Steps:
- `G29 P1 T` (works as expected)
- `G29 P2 B T`

What happens:
- Extruder moves to max X (maybe a little over? --- seems to knock against the side of the frame) and Y at high speed.
- Extruder moves to first unfilled point, terminal gives the `Place shim and measure` message.
- After using the encoder to lower onto the shim and pressing the button, extruder moves to next point. Doesn't ask to lower to the bed as mentioned in the guide.
- Mesh values are inflated with shim size (e.g., a solid +0.3mm above all other values).
- Running a print after this causes the Z to move up obviously way more than it should.

Possible useful config values (let me know if anything else is needed):
```
#define X_PROBE_OFFSET_FROM_EXTRUDER 40  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -10  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.6   // Z offset: -below +above  [the nozzle]

#define XY_PROBE_SPEED 6000

#define X_MIN_POS 0
#define Y_MIN_POS -23
#define Z_MIN_POS 0
#define X_MAX_POS 193
#define Y_MAX_POS 217
#define Z_MAX_POS 200

#define GRID_MAX_POINTS_X 7      // Don't use more than 15 points per axis, implementation limited.
#define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
#define UBL_MESH_EDIT_MOVES_Z     // Sophisticated users prefer no movement of nozzle

// (to max out mesh area without colliding with clips, etc)
#define UBL_MESH_MIN_X (0)
#define UBL_MESH_MAX_X (170)
#define UBL_MESH_MIN_Y (20)
#define UBL_MESH_MAX_Y (170)
```

Everything else seems to work as expected (probing, printing after leveling using `G29 P3` to fill in with sameish values, etc). Just not `P2 B`. Am I just misunderstanding the guide?

Roxy-3D, 2017-06-23:
I'll have to check this out in the morning...   And other people are going to disagree...  But there really isn't much need for the P2 Phase any more.   Just move on to G29 P3 T  (do that until the mesh is fully populated)   And then do a G26.   With the G26 pattern on the bed, start editing any areas that need adjustment with the G29 P4 R command.

If you give that a try and post any questions are problems you have....  I'll respond first thing in the morning....    (And please post the Topology map from a G29 T) so I can see what you mesh looks like.

Bob-the-Kuhn, 2017-06-23:
@Roxy-3D - could this be an extreme case of bounce on the encoder switch?  Maybe there should be a debounce on exiting `measure_point_with_encoder()`.

@Knifa - by chance are you double clicking the encoder?  That would explain how the second measurement is being skipped.

Roxy-3D, 2017-06-23:
> could this be an extreme case of bounce on the encoder switch? Maybe there should be a debounce on exiting measure_point_with_encoder().

Yes!  It could be!   I just looked at the  float unified_bed_leveling::measure_business_card_thickness(float &in_height) function and it doesn't debounce the Encoder Wheel click.    

That needs need to be changed.

Knifa, 2017-06-23:
@Bob-the-Kuhn: just a single click so if it's double clicking it's not on purpose. :smile:

agambini, 2017-07-24:
The question about the manual probe or auto probe is : Why one works and another not, basically , a mesh is created in manual probe and the printer obey the distances but not in auto mode. I'm not a programmer , and my opinion is very superficial , but looks here that mesh created in auto mode is wrong stored or ignored in print process
Is this possible ?

Knifa, 2017-07-24:
Sorry, I totally forgot about this. @Bob-the-Kuhn: do you still need a topology map to continue looking into this?

Roxy-3D, 2017-08-03:
>  but looks here that mesh created in auto mode is wrong stored or ignored in print process

It doesn't matter how a value gets put into the Topographical mesh.  It can get there from G29 P1 (which is automated).  It can get there from G29 P2 (which is manual probing...)   It can get there from G29 P3 (which fills in unprobed mesh points) or it can get there from G29 P4 or M421 (which edit a mesh point).

The value is used as is...   It doesn't matter how it gets into your mesh.  It is respected and used to calculate the Z-Height correction.

Roxy-3D, 2017-08-03:
Are people having problems getting G29 P2 to work reliably?

Do we still need to add the Encoder Wheel debounce for the G29 P2 ?

closed 2018-02-09 by thinkyhead