MarlinFirmware/Marlin #7126: Auto bed leveling bilinear not working

AcroFlax, 2017-06-23:
First of all, Marlin version is 1.1.3
The thing is that G28 command homes to center correctly and without a problem. But when it comes to G29 command, it does not work for me. The problem is that G29 does not probe, it goes to all 9 points on the bed, but instead of going down to the bed and probe, it goes up after all locations and thats it. I am using AUTO_BED_LEVELING_BILINEAR. Probe is J18A3-8-Z/AX inductive sensor. If trigger is inverted, G28 wouldn't be able to home, so as G29. I tried disabling double probe, inverting Z axis min stop, nothing helped.

Bob-the-Kuhn, 2017-06-23:
Sounds like the system is seeing the probe as always triggered.

Use the M119 command to see what Marlin thinks the state of the probe is.  I expect that you'll see it as always triggered.

Bob-the-Kuhn, 2017-06-23:
Try turning off the internal pullup on the probe.  That's caused the probe to appear as in one state all the time.

If you have `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN `enabled then use the following .
```
//#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

#define Z_MIN_ENDSTOP_INVERTING false
#define Z_MIN_PROBE_ENDSTOP_INVERTING false
```

If you have `Z_MIN_PROBE_ENDSTOP `enabled then use the following .  
```
//#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  #define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

#define Z_MIN_PROBE_ENDSTOP_INVERTING false
```

AcroFlax, 2017-06-28:
Solved my problem, with your given advice. Still weird, because when I was using old firmware, which was 1.0.2. I didnt't need to specify or change such parameters :)

Bob-the-Kuhn, 2017-06-28:
Glad to see you up & running.

I just looked at 1.0.2.1 & 1.0.2.2 and they both also default to all the endstop pullups being enabled.  

Agreed - weird.

davetech, 2017-07-17:
I had the same problem with 1.1.4 but I have no problem with 1.1.3. I am using a simple normally closed microswitch.

closed 2018-02-09 by thinkyhead