MarlinFirmware/Marlin #7130: Dual x carriage,  duplication mode - MIRRORED

pbrgle1, 2017-06-23:
Hi. I had projects where i needed multiple mirrored parts and i was thinking it could be possible to have a mirrored duplication mode, since one coordinate stays the same and the second one (x) is only in the other direction. 

However, the x direction would have to be smaller to avoid carriages bumping into each other. Like a dead zone in the middle that has the width of the distance between the nozzles when the carriages are together. 

Would it be possible to implement this option?

BjoernC, 2017-08-27:
Hi. I was thinking the same thing, and often need "mirror" parts.
i've been navigating through marlin, trying to implement this feature, but it is not as simple as i thought.
A "quick fix" could be to reverse the DIR on X2 , make x-offset = X2_MAX_POS  and disable software endstops, but,this scenario could generate some serious damage and is really more of a "workaround" than an actual solution. with this you have to "know" that the carriages do not crash.
as i see it, the X2 is really just an "slave" of the X, with an offset, when running in "duplication mode",
To have it running in "mirror" mode there is need for for some way to "know" where the X2 axes is, so we don't trigger software endstop, and maybe have some kind of min X2 pos and max X pos in "mirror" mode so the carriages do not crash against each other. 
but i have not been able to figure out how, yet.
Anybody have some thought on this matter ? or maybe a pointer in the right direction ?

thinkyhead, 2018-02-09:
One would have to ensure the input G-code is well within one half of the bed. Mirroring could then be done relative to the center position. Due to the width of the X carriages, the center region would not be usable.

InsanityAutomation, 2019-10-15:
@boelle @shitcreek this is done as well.

closed 2019-10-15 by boelle