MarlinFirmware/Marlin #7135: Controller fan is not activated while pid autotune is running

DaviVolcan, 2017-06-24:
Controller fan is not activated while pid autotune is running, this can cause mosfet overheating in ramps and other boards. i was running pid autotune when i realized that the controller fan was off an the mosfet got hot. but while im printing the fan speeds up.

thinkyhead, 2018-02-09:
Hmm… Current code for the controller fan only enables it when steppers are enabled. It doesn't care about heating mosfets, because they don't have a failure mode. If MOSFETs are getting dangerously hot, consider attaching a larger heatsink.

boelle, 2019-02-19:
i would agree with @DaviVolcan here

i have a 80mm fan cooling off the stepper drivers and the mosfets, its a Ramps 1.4SB and that does not allow space for heat sinks, 

but it can/should be added at some point

ellensp, 2020-08-24:
Current code already does this for the heated bed mosfet
```
 // If any of the drivers or the heated bed are enabled...
    if (motor_on || TERN0(HAS_HEATED_BED, thermalManager.temp_bed.soft_pwm_amount > 0))
      lastMotorOn = ms; //... set time to NOW so the fan will turn on
```
This shouldn't be needed for any other mosfet as they shouldn't get hot.  (nether should bed mosfet for that matter, some have bad design choices)

closed 2020-08-24 by ellensp