MarlinFirmware/Marlin #7137: [Question] G-Code to echo to serial out?

benlye, 2017-06-25:
I have my printer connected to an OctoPrint host, and a TP-Link HS100 smart switch.  I normally control the power to the printer using OctoPrint to control the smart switch.  Now that we have custom menu commands, it would be nifty if one of those could turn my printer off.  

I've found the [ActionTrigger](https://github.com/Booli/OctoPrint-ActionTriggerPlugin) plugin for OctoPrint, which takes action based on serial output from the printer.  My question is, is there already a way to get Marlin to write custom text to the serial output from a user menu command?  

If not, it seems like implementing a g-code command which just echoes to serial wouldn't be too hard - I just thought I'd ask before I coded it myself.

fiveangle, 2017-06-25:
MSG_KILLED string is sent to serial indicating last gasp of breath for any condition where Marlin gives up the ghost, including accepting M112 command but also just toggling SUICIDE_PIN.  No custom menu needed, just hit the reset button on the printer, set Actionscript to key off the MSG_KILLED string that Marlin sends when it commits hari kari and when Actionscript detects, kill life support.

This is actual a better idea than your custom serial echo option, since if Marlin hits any issue where it has to halt, killing power to the heaters prevents any possible catastrophe 👍

benlye, 2017-06-25:
I really like your idea, but in my case it won't help my primary use case as my display doesn't have a kill button, it only has a reset button which is hard wired to the board reset pin (thanks, Anet).  

I'll do both :-)

benlye, 2017-06-25:
Oh, wait, now I get it. Put M112 in as the custom menu action, and configure ActionTrigger to handle MSG_KILLED. No coding needed and it handles both scenarios. Neat.

fiveangle, 2017-06-25:
You got it.  I took a look at ActionTrigger and it doesn't have any of this functionality so assuming you add it, please submit a PR to the author to add it to the main plugin.  With this, I think I might finally be intersted in power control for my printers :)

benlye, 2017-06-25:
So it turns out this is and isn't so easy.

G-code specifies a way for the robot to send text, including commands, back to the host:
http://reprap.org/wiki/Gcode#Replies_from_the_RepRap_machine_to_the_host_computer
> The RepRap machine may also send lines that look like this:
// This is some debugging or other information on a line on its own. It may be sent at any time.
Such lines will always be preceded by //.
On the latest version of Pronterface and Octoprint (1.2.0+) a special comment of the form:
// action:command
is allowed to be sent from the firmware, the command can currently be pause, resume or disconnect which will execute those commands on the host. As this is also a comment other hosts will just ignore these commands.
The most common response is simply:
ok 

Octoprint includes funtionality to handle a certain set of commands natively:
http://docs.octoprint.org/en/master/features/action_commands.html
>**pause**
    When this command is received from the printer, OctoPrint will pause streaming of a current print job just like if the “Pause” button had been clicked.
**resume**
    When this command is received from the printer, OctoPrint will resume streaming of a current print job just like if the “Resume” button had been clicked.
**disconnect**
    When this command is Received from the printer, OctoPrint will immediately disconnect from it. 

Nothing in OctoPrint (today) seems to handle parsing of the raw serial output looking for a specific string (such as the full MSG_KILLED message).  ActionTrigger comes the closest.

To that end, I've implemented a super simple function, M217, which sends whatever string follows the command out to serial:
```
/**
 * M217: Send text to serial output
 */
inline void gcode_M217() { SERIAL_ECHOLN(parser.string_arg); }
```
And now I can send action commands to OctoPrint using syntax like `M217 // action:disconnect` (which works).  Next step is to get a "kill" action coded into ActionTrigger, then I'll look at a configuration option to send something like `// action:kill` to serial when `kill()` runs.

fiveangle, 2017-06-25:
Seems the "correct" approach here is to file an ER against Octoprint to request to add ability to trap on receipt of MSG_KILLED string, but MSG_KILLED is language-dependent, so that may add complication).  Alternatively `kill()` in , Marlin could be augmented to output the required `// action:kill` format upon SUICIDE.

Thinking about this, the latter seems cleaner and still gets the desired results, but probably would be good to hear from @foosel and @thinkyhead

fiveangle, 2017-06-25:
> Next step is to get a "kill" action coded into ActionTrigger, then I'll look at a configuration option to send something like // action:kill to serial when kill() runs.

Missed your last line there.  Yup, I concur :)

benlye, 2017-06-25:
> Next step is to get a "kill" action coded into ActionTrigger, then I'll look at a configuration option to send something like // action:kill to serial when kill() runs.

I'm _so_ close - just need to figure out how to call a system command from the plugin, if it's possible.  Then I'd like to extend ActionTrigger to provide a UI for adding and configuring the triggers.

To your point about the 'correct' approach - I still like the idea of having an M code to generate arbitrary serial output as well as automatically sending `// action: kill` on `kill()`, as it could be potentially useful for other reasons (not that I can think of many others right now), and it seems a cheap (in terms of PROGMEM and risk) function to implement.  Another way would be a code which just sent action output, e.g. `M217 dosomething` causes `// action:dosomething` to be sent to serial.  Certainly interested in other's thoughts.

I'l plug away at it for a bit and circle back as/when I make progress.

fiveangle, 2017-06-25:
Might be easiest to [add a system command](https://github.com/foosel/OctoPrint/wiki/Cookbook:-System-Commands) in the OP config then have ActionScript trigger that action ?

foosel, 2017-06-26:
Not sure what kind of input from my side would be required here. Action commands were derived (not by me btw) for stuff like that, and are definitely way less cumbersome to support than a myriad of (potentially incompatible) kill-strings across not only this version of Marlin but every other Marlin/Repetier/Sprinter/Smoothie/RepRapFirmware/etc-version under the sun ;)

fiveangle, 2017-06-26:
Yes @foosel - my comment was prior to seeing the better direction :)

benlye, 2017-06-26:
I got it to work :-)

Either on M112, or using `M217 // action:kill`, the printer sends `// action:kill` to serial and the ActionTrigger plugin for OctoPrint responds by running a new system command called `systemKillCommand`, which sends a command to turn off the smart plug.

There are some interesting dynamics in the way OctoPrint behaves if you use the OctoPrint terminal to send `M112` - it disconnects from serial immediately, before the printer has time to send the action command - but if I put `M112` in the CUSTOM_USER_MENUS and call it, it behaves perfectly - power goes off nearly instantly after the command is triggered.

I'll post more details shortly, once I've cleaned it up a bit.

benlye, 2017-06-29:
Quick update...

Now that we have M118 to send output to serial, this is coming along.  I'm working on completely re-writing the ActionTrigger plugin to make it more flexible, and easier to configure - I've made the settings page a user-editable list of actions and associated commands.  I'm still working on getting commands parsed and executed, but it's coming together.  I should be able to post it in a few more days.

closed 2017-08-03 by benlye

jneilliii, 2019-07-20:
@benlye, et. al., I know you got this poweroff option working with M112 in CUSTOM_USER_MENUS, but can you provide an example of using M118?  I have updated my Configuration_adv.h like below and for some reason the command being sent back to OctoPrint is getting truncated.

```
#define CUSTOM_USER_MENUS
#if ENABLED(CUSTOM_USER_MENUS)
  //#define USER_SCRIPT_DONE "M117 User Script Done"
  //#define USER_SCRIPT_AUDIBLE_FEEDBACK
  #define USER_SCRIPT_RETURN  // Return to status screen after a script

  #define USER_DESC_1 "MQTTPublish"
  #define USER_GCODE_1 "M118 //action:MQTTPublish topic message"
#endif
```

This is what is being received by OctoPrint.

```
Recv: echo:enqueueing "M118 //action:MQTTPublish top"
Recv: //action:MQTTPublish top
```