MarlinFirmware/Marlin #7138: Using LCD for Melzi with Melzi board

webliya, 2017-06-25:
Have been struggling to use last Marlin with Melzi and LCD for Melzi board. After hours I managed to make it work. The additions to the code are as below.

I hope you add this code so everybody can use it since a lot of people suffer (It is the best and easiest way to use Marlin with Wanhao i3 duplicator v1 to 2.1.).

Configuration.h
-----------------
Just before;
//=============================================================================
//=============================== Extra Features ==============================
//=============================================================================
Place the code below. Delays are must to be used as below. Delay setup in ultralcd_st7920_u8glib_rrd.h is not enough for Melzi and full graphics board.
 
// If you want you can define your own set of delays in Configuration.h (If you see garbled graphics on the screen uncomment all three)
#define ST7920_DELAY_1 DELAY_0_NOP
#define ST7920_DELAY_2 DELAY_3_NOP
#define ST7920_DELAY_3 DELAY_0_NOP

//LCD for MELZI (If you have a Melzi board with LCD for Melzi card uncomment this)
#define LCD_MELZI true

And place the code below in;
pins_SANGUINOLOLU_11.h
------------------------------
Just before   #define SD_DETECT_PIN         -1

  #if LCD_MELZI
	#define LCD_PINS_RS       17
	#define LCD_PINS_ENABLE   16
	#define LCD_PINS_D4       11
	#define BTN_ENC           28
	#define BTN_EN1           29
	#define BTN_EN2           30
  #endif // LCD MELZI

A lot of people suffer from this. Please add to the next development version.

Can

Tannoo, 2017-06-27:
Fork this repo, then clone your fork to your computer, create a new branch on your fork. Make your changes in that new branch. Then submit that as a PR.

Tannoo, 2017-06-27:
Nevermind... I see that you've done that already.

thinkyhead, 2017-07-02:
A PR is pending…

closed 2017-07-02 by thinkyhead