MarlinFirmware/Marlin #7142: Stopped

jw10240, 2017-06-25:
So I have been trying to get my printer set up with dual extruders with Geeetch auto bed leveling. I have tried different peoples Firmware and creating my own. Every time I re flash the printer I now get this error STOPPED message on the LCD screen. Extruder E0 will heat up, but E1 will not. I have changed out the thermistors and heating elements and I still have the same issue. I also end up having three extruders show up after upload, when I have only set it to two. Has any had this issue before? Thanks for the help.
[image]
[image]
[image]

Bob-the-Kuhn, 2017-06-25:
We'll be glad to help.

What version(s) of Marlin have you tried?

The easiest way to help is:

- Please use the latest Marlin - [bugfix1.1.x ](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x)
- You'll need to manually move your machine specific items over to the new config files.
- ZIP up your config files and drop the ZIP onto your reply.
- What controller are you using?
- What type of printer is this? Cartesian, delta, coreXY, ...

jw10240, 2017-06-26:
Please see attached.I hope this is what you need. I have tried all three
versions available from the Marlin website, with the same results. My
printer is a 2016 Tevo Tarantula small bed. Everything is stock except, I
have added dual E3D extruders and Geetech 3D Touch sensor. So my printer is
a cartesian. The board on the printer is MKS Base V1.4. If you need any
additional information, please let me know. Thanks for the help, it is
greatly appricated.

Bob-the-Kuhn, 2017-06-27:
The attachment didn't come through.  Try dropping it again.

Did switching to bugfix1.1.x make any difference?

Have you tried to manually heat up the second extruder via the host software?

Does the second extruder's temperature rise if you bring a heat source near by?

jw10240, 2017-06-27:
Okay, hopefully it will come thru this time. I have that version loaded on
the printer now. I used a light, indirect heat, to try and heat up the
thermistor. E0 thermistor value went up, but E1 did not. I know this is a
working thermistor, because it came out of the hot end I'm upgrading from.
When I try to heat with the LCD screen, the thermistor doesn't change and I
get the error, Heating Failed, printer halted, please reset. With
repetier-Host I get a M999 code if I remember correctly. Please let me know
if I can provide any more information, I'll be glad to help. Thanks for the
help.

Bob-the-Kuhn, 2017-06-27:
The attachment didn't come through.

Did E1 heat up when commanded?  

In configuration_adv.h enable PINS_DEBUGGING and then 
- Issue a `M43 I1 `command.  That'll display all the pins and what signal(s) are attached to them.  See if TEMP_1_PIN is attached to analog input A15 (pin 69) and if it is an input or an output.  
- Heat up that thermistor and then issue `M43 I1 `again.  Look for differences in the analog inputs.  Maybe the thermistor is attached to the wrong pin.

jw10240, 2017-06-27:
So this is what comes back: < 12:01:43 AM: PIN:  69   Port: K7 (A15)
 TEMP_1_PIN                  Analog in =   994. I have also learned that It
will only heat E0, no matter if you choose E1 or even E2 (just to test). So
is the E1 pin not activated? Thanks for the help.

Here is the entire log below, if it helps:
> 12:16:44 AM: N23 M104 S200 *118
> 12:16:47 AM: N25 T1 *44
< 12:16:47 AM: echo:Active Extruder: 1
> 12:16:48 AM: N27 M104 S200 *114
> 12:16:50 AM: N29 T2 *35
< 12:16:50 AM: echo:Active Extruder: 2
> 12:17:06 AM: N45 M43 I1 *93
< 12:17:06 AM: PIN:   0   Port: E0        <unused/unknown>            Input
 = 1
< 12:17:06 AM: PIN:   1   Port: E1        <unused/unknown>            Input
 = 0
< 12:17:06 AM: PIN:   2   Port: E4        <unused/unknown>            Input
 = 1    TIMER3B   PWM:     0    WGM: 0    COM3B: 0    CS: 2    TCCR3A: 0
 TCCR3B: 2    TIMSK3: 2   non-standard PWM mode
< 12:17:06 AM: PIN:   3   Port: E5        X_MIN_PIN                   Input
 = 1    TIMER3C   PWM:     0    WGM: 0    COM3C: 0    CS: 2    TCCR3A: 0
 TCCR3B: 2    TIMSK3: 2   non-standard PWM mode
< 12:17:06 AM: PIN:   4   Port: G5        FAN_PIN
Output = 0    TIMER0B   PWM:   128    WGM: 3    COM0B: 0    CS: 3
 TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   compare interrupt enabled   overflow
interrupt enabled
< 12:17:06 AM: .                         FIL_RUNOUT_PIN              Output
= 0
< 12:17:06 AM: .                         SERVO3_PIN                  Output
= 0
< 12:17:06 AM: PIN:   5   Port: E3        SERVO2_PIN                  Input
 = 1    TIMER3A   PWM:  -25536    WGM: 0    COM3A: 0    CS: 2    TCCR3A: 0
   TCCR3B: 2    TIMSK3: 2   non-standard PWM mode   compare interrupt
enabled
< 12:17:06 AM: PIN:   6   Port: H3        SERVO1_PIN                  Input
 = 1    TIMER4A   PWM:     0    WGM: 1    COM4A: 0    CS: 3    TCCR4A: 1
 TCCR4B: 3    TIMSK4: 0
< 12:17:06 AM: PIN:   7   Port: H4        <unused/unknown>            Input
 = 0    TIMER4B   PWM:     0    WGM: 1    COM4B: 0    CS: 3    TCCR4A: 1
 TCCR4B: 3    TIMSK4: 0
< 12:17:06 AM: PIN:   8   Port: H5        HEATER_BED_PIN
 Output = 0    TIMER4C   PWM:     0    WGM: 1    COM4C: 0    CS: 3
 TCCR4A: 1    TCCR4B: 3    TIMSK4: 0
< 12:17:06 AM: .                         RAMPS_D8_PIN                Output
= 0
< 12:17:06 AM: PIN:   9   Port: H6        HEATER_1_PIN
 Output = 0    TIMER2B   PWM:     0    WGM: 1    COM2B: 0    CS: 4
 TCCR2A: 1    TCCR2B: 4    TIMSK2: 0
< 12:17:06 AM: .                         RAMPS_D9_PIN                Output
= 0
< 12:17:06 AM: PIN:  10   Port: B4        HEATER_0_PIN
 Output = 1    TIMER2A   PWM:     0    WGM: 1    COM2A: 0    CS: 4
 TCCR2A: 1    TCCR2B: 4    TIMSK2: 0
< 12:17:06 AM: .                         RAMPS_D10_PIN               Output
= 1
< 12:17:06 AM: PIN:  11   Port: B5        SERVO0_PIN
 Output = 0    TIMER1A   PWM:  2000    WGM: 4    COM1A: 0    CS: 2
 TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode   compare
interrupt enabled
< 12:17:06 AM: PIN:  12   Port: B6        PS_ON_PIN
Output = 0    TIMER1B   PWM:     0    WGM: 4    COM1B: 0    CS: 2
 TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
< 12:17:06 AM: PIN:  13   Port: B7        LED_PIN                     Input
 = 0    TIMER0A   PWM:     0    WGM: 3    COM0A: 0    CS: 3    TCCR0A: 3
 TCCR0B: 3    TIMSK0: 5   overflow interrupt enabled
< 12:17:06 AM:  .                  TIMER1C is also tied to this pin
        TIMER1C   PWM:     0    WGM: 4    COM1C: 0    CS: 2    TCCR1A: 0
 TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
< 12:17:06 AM: PIN:  14   Port: J1        Y_MIN_PIN                   Input
[... 111 lines cut ...]
Output = 1
< 12:17:06 AM: PIN:  62   Port: K0 (A 8)  Z_ENABLE_PIN
 Output = 1
< 12:17:06 AM: PIN:  63   Port: K1 (A 9)  <unused/unknown>   Analog in =
929   Input  = 1
< 12:17:06 AM: PIN:  64   Port: K2 (A10)  <unused/unknown>   Analog in =
778   Input  = 1
< 12:17:06 AM: PIN:  65   Port: K3 (A11)  <unused/unknown>   Analog in =
710   Input  = 1
< 12:17:06 AM: PIN:  66   Port: K4 (A12)  MAX6675_SS                  Input
 = 1
< 12:17:06 AM: PIN:  67   Port: K5 (A13)  TEMP_0_PIN
 Analog in =   513
< 12:17:06 AM: PIN:  68   Port: K6 (A14)  TEMP_BED_PIN
 Analog in =   990
< 12:17:06 AM: PIN:  69   Port: K7 (A15)  TEMP_1_PIN
 Analog in =   974
> 12:17:11 AM: N50 T0 *47
< 12:17:11 AM: echo:Active Extruder: 0
> 12:17:12 AM: N52 M104 S0 *114
> 12:17:14 AM: N54 T1 *42
< 12:17:14 AM: echo:Active Extruder: 1
> 12:17:15 AM: N56 M104 S0 *118
> 12:17:16 AM: N58 M104 S200 *122
> 12:17:20 AM: N63 M104 S0 *112

jw10240, 2017-06-27:
Im sorry been along day, I forgot I tried to see if one extruder re flash
would let me use the auto level sensor. So I have re uploaded the dual
extruder set up, and yes the E1 thermistor heats up when I healed the light
to it. Here is the M43 I1 Code

> 12:22:48 AM: N97 M104 S200 *121
> 12:22:54 AM: N103 T0 *24
< 12:22:54 AM: echo:Active Extruder: 0
> 12:22:55 AM: N105 M104 S200 *67
> 12:23:27 AM: N137 M43 I1 *105
< 12:23:27 AM: PIN:   0   Port: E0        <unused/unknown>            Input
 = 1
< 12:23:27 AM: PIN:   1   Port: E1        <unused/unknown>            Input
 = 0
< 12:23:27 AM: PIN:   2   Port: E4        <unused/unknown>            Input
 = 1    TIMER3B   PWM:     0    WGM: 0    COM3B: 0    CS: 2    TCCR3A: 0
 TCCR3B: 2    TIMSK3: 2   non-standard PWM mode
< 12:23:27 AM: PIN:   3   Port: E5        X_MIN_PIN                   Input
 = 1    TIMER3C   PWM:     0    WGM: 0    COM3C: 0    CS: 2    TCCR3A: 0
 TCCR3B: 2    TIMSK3: 2   non-standard PWM mode
< 12:23:27 AM: PIN:   4   Port: G5        FAN_PIN
Output = 0    TIMER0B   PWM:   128    WGM: 3    COM0B: 0    CS: 3
 TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   compare interrupt enabled   overflow
interrupt enabled
< 12:23:27 AM: .                         FIL_RUNOUT_PIN              Output
= 0
< 12:23:27 AM: .                         SERVO3_PIN                  Output
= 0
< 12:23:27 AM: PIN:   5   Port: E3        SERVO2_PIN                  Input
 = 1    TIMER3A   PWM:  -25536    WGM: 0    COM3A: 0    CS: 2    TCCR3A: 0
   TCCR3B: 2    TIMSK3: 2   non-standard PWM mode   compare interrupt
enabled
< 12:23:27 AM: PIN:   6   Port: H3        SERVO1_PIN                  Input
 = 1    TIMER4A   PWM:     0    WGM: 1    COM4A: 0    CS: 3    TCCR4A: 1
 TCCR4B: 3    TIMSK4: 0
< 12:23:27 AM: PIN:   7   Port: H4        <unused/unknown>            Input
 = 0    TIMER4B   PWM:     0    WGM: 1    COM4B: 0    CS: 3    TCCR4A: 1
 TCCR4B: 3    TIMSK4: 0
< 12:23:27 AM: PIN:   8   Port: H5        HEATER_BED_PIN
 Output = 0    TIMER4C   PWM:     0    WGM: 1    COM4C: 0    CS: 3
 TCCR4A: 1    TCCR4B: 3    TIMSK4: 0
< 12:23:27 AM: .                         RAMPS_D8_PIN                Output
= 0
< 12:23:27 AM: PIN:   9   Port: H6        HEATER_1_PIN
 Output = 0    TIMER2B   PWM:     0    WGM: 1    COM2B: 0    CS: 4
 TCCR2A: 1    TCCR2B: 4    TIMSK2: 0
< 12:23:27 AM: .                         RAMPS_D9_PIN                Output
= 0
< 12:23:27 AM: PIN:  10   Port: B4        HEATER_0_PIN
 Output = 1    TIMER2A   PWM:     0    WGM: 1    COM2A: 0    CS: 4
 TCCR2A: 1    TCCR2B: 4    TIMSK2: 0
< 12:23:27 AM: .                         RAMPS_D10_PIN               Output
= 1
< 12:23:27 AM: PIN:  11   Port: B5        SERVO0_PIN
 Output = 0    TIMER1A   PWM:  2000    WGM: 4    COM1A: 0    CS: 2
 TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode   compare
interrupt enabled
< 12:23:27 AM: PIN:  12   Port: B6        PS_ON_PIN
Output = 0    TIMER1B   PWM:     0    WGM: 4    COM1B: 0    CS: 2
 TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
< 12:23:28 AM: PIN:  13   Port: B7        LED_PIN                     Input
 = 0    TIMER0A   PWM:     0    WGM: 3    COM0A: 0    CS: 3    TCCR0A: 3
 TCCR0B: 3    TIMSK0: 5   overflow interrupt enabled
< 12:23:28 AM:  .                  TIMER1C is also tied to this pin
        TIMER1C   PWM:     0    WGM: 4    COM1C: 0    CS: 2    TCCR1A: 0
 TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
< 12:23:28 AM: PIN:  14   Port: J1        Y_MIN_PIN                   Input
 = 0
< 12:23:28 AM: PIN:  15   Port: J0        <unused/unknown>            Input
 = 1
[... 100 lines cut ...]
920   Input  = 1
< 12:23:28 AM: PIN:  58   Port: F4 (A 4)  <unused/unknown>   Analog in =
779   Input  = 1
< 12:23:28 AM: PIN:  59   Port: F5 (A 5)  FILWIDTH_PIN
 Analog in =   919
< 12:23:28 AM: PIN:  60   Port: F6 (A 6)  Y_STEP_PIN
 Output = 0
< 12:23:28 AM: PIN:  61   Port: F7 (A 7)  Y_DIR_PIN
Output = 1
< 12:23:28 AM: PIN:  62   Port: K0 (A 8)  Z_ENABLE_PIN
 Output = 1
< 12:23:28 AM: PIN:  63   Port: K1 (A 9)  <unused/unknown>   Analog in =
920   Input  = 1
< 12:23:28 AM: PIN:  64   Port: K2 (A10)  <unused/unknown>   Analog in =
772   Input  = 1
< 12:23:28 AM: PIN:  65   Port: K3 (A11)  <unused/unknown>   Analog in =
817   Input  = 1
< 12:23:28 AM: PIN:  66   Port: K4 (A12)  MAX6675_SS                  Input
 = 1
< 12:23:28 AM: PIN:  67   Port: K5 (A13)  TEMP_0_PIN
 Analog in =   355
< 12:23:28 AM: PIN:  68   Port: K6 (A14)  TEMP_BED_PIN
 Analog in =   987
< 12:23:28 AM: PIN:  69   Port: K7 (A15)  TEMP_1_PIN
 Analog in =   963

Bob-the-Kuhn, 2017-06-27:
Did you heat up the thermistor for E0 or E1?  The E0 one moved a lot while the E1 didn't move much.
```
< 12:17:06 AM: PIN:  67   (A13)  TEMP_0_PIN  Analog in =   513
< 12:23:28 AM: PIN:  67   (A13)  TEMP_0_PIN  Analog in =   355
 
< 12:17:06 AM: PIN:  68   (A14)  TEMP_BED_PIN  Analog in =   990
< 12:23:28 AM: PIN:  68   (A14)  TEMP_BED_PIN  Analog in =   987

< 12:17:06 AM: PIN:  69   (A15)  TEMP_1_PIN  Analog in =   974
< 12:23:28 AM: PIN:  69   (A15)  TEMP_1_PIN  Analog in =   963
```

----

I really need your config files.  Please ZIP them up & attach to your reply.

Bob-the-Kuhn, 2017-06-27:
I just looked at the pictures again and they show that both extruders are increasing in temperature.

I'm wondering if the problem is the error checking is set too tight.

Try these settings:
```
#define TEMP_HYSTERESIS 10      
#define TEMP_WINDOW     5 
```

Try these commands:
```
T0 ; make extruder 1 active
M109 S200  ;  heat E1 to 200C
T1: make extruder 2 active
M109 S200  ;  heat E2 to 200C
```

Your host software should have a means of graphing the temperature vs. time.

jw10240, 2017-06-28:
[Marlin-170627a.zip](https://github.com/MarlinFirmware/Marlin/files/1107524/Marlin-170627a.zip)
Well my power supply took a dump, so I will have to test heating the hot ends tomorrow. I did change the parameters and uploaded the firmware. I still get "STOPPED" on the screen. I don't if this would go away once I heat the extruders or not. Also I can't get the 3D Touch to respond to any commands. It runs a self test when power is applied, but the will not respond after that. I have attached the config file, so hopefully that will help. Thanks for the help.

Bob-the-Kuhn, 2017-06-28:
I just compiled your ZIP (with no changes), connected the orange pin to SERVOo (logical pin D11) and the white pin to the Z_MIN endstop on my RAMPS board and it worked perfectly.

Since the BLTouch is doing it's self test on power up then the brown & red wires are hooked up correctly.  That means the orange wire is either not attached to the correct pin or there's a problem with the wire/connectors.

Here's the graphic I used to hook it up.

[image]

jw10240, 2017-07-04:
Sorry it has taken awhile to get back to you. PS took a dump randomly. In the end I just left off the brown wire and it worked perfect for the touch sensor. So I'm not getting 12v when sending the code of from the LCD control to the second hot end, E1. Could that part of board gone bad? I have tinned all the connections to make sure they are getting contact. I even swapped the heating elements just to double check if it was bad, heated right up. Thanks for the help

closed 2018-02-09 by thinkyhead