MarlinFirmware/Marlin #7143: G29 not leveling properly

copystring, 2017-06-25:
Whenever my printer probes my bed the results are near useless. While the Z-axis works to level the bed, it levels it completly wrong.
I start with G28 followed by G29 with a grid of 7. I have a fairly large and most likely uneven bed. Not really visible the eye hower when I lay a straight piece of wood or steel on it, I can see my torch undernearth from the other side.
Also confirmed this by uploading 3 different times on maui-3d.com. For example: http://www.maui-3d.com/cgi-bin/plotG29?plotG29ID=12194
The other 2 I have plotted look nearly the same.

To me it seems very likely to be a issue with the firmware. When I manually level the X-axis to the bed and manually move the axis left and right then it is mostly even. Then I do G28 and G29 and doing the same move I did by hand via the interface of Octoprint. Looking at the distance between the nozzle and bed, it is clear that is correcting totally wrong.

Here is my configcuration.h: https://pastebin.com/BkSdJ2BH

For the past several days I tried to figure out what the issue is. I am very sure this is an issue with the firmware or maybe my config.
I am totally lost here.

benlye, 2017-06-25:
Is it possible that something in your printer's mechanics alters the distance between the probe and the bed as it traverses across the X and Y axis?  Cable tension, bent rods, ...?

I would try determining the Z offset from the probe to the nozzle near to 0,0, the center, and near the extremes of the X and Y and comparing the results - you would expect them to be the same.

copystring, 2017-06-25:
At one point I thought this as well. I then went on designed and printed new parts with 50% infill and replaced the old ones.  Bought new belts and replaced the whole y carriage apart from the bed.
The rods I have not replaced yet. Everything else looks good to me. I spent the past 5 or so days checking everything I could think of.

Tested the Z offset in the corners and in the center of my bed with the probe. Here is what I got:

X0Y0		: -0.4
X360Y0		: -0.80
X360Y210	: -0.70
X0Y210		: -0.70
X180Y105	: -0.50

benlye, 2017-06-25:
Hmm, they look fishy to me - especially when you define the Z offset as -1 in your configuration.h.  I'd expect variation in the Z offset to be near 0 for good results, a 0.4mm variation along the X axis doesn't explain the large variation in your plot, but it's still significant.

BTW, I offer help as someone else who went through a similar exercise on my own printer, only to find all my issues were caused by `#define Y_PROBE_OFFSET_FROM_EXTRUDER` being out by 2mm.

copystring, 2017-06-25:
Well the -1 is changed in the EEPROM. It is just an older value from testing.
The Y offset I measured from my own design in Inventor: 
[image]

Keep in mind that I did a G28 Z for every offset. That does not represent a plane or mesh of the bed.

copystring, 2017-06-27:
Today I replaced my rods for hardened steel ones. They definitely do not bend while moving the bed around.
Any other ideas what it could be?

copystring, 2017-06-28:
After digging through a lot of forums, topic, tickets, etc I think adding M420 S1 might have solved the problem mostly. At least it is level apart from the last 3-4 cm from the edge of my bed.

closed 2018-02-09 by thinkyhead