MarlinFirmware/Marlin #7146: Z Homing ignores Z-Axis Proximity Sendor

keremers, 2017-06-26:
I have a Prusa i3 printer. I was previously using 1.1.0-rc7 with my printer. Then homing was fine. Recently upgraded to 1.1.3. I needed to change Z homing pin to 12. After the modification printer start to detect Z proximity sensor. When the extruder is close to the bed I see the Z_MIN_probe triggers to TRIGGERD staus from Open. But When I want to Home the extruder X and Y homes just fine but the software ignores the Z probe and servo keeps running until the extruder is stuck to bed. How can I fix this?

Bob-the-Kuhn, 2017-06-26:
If you are using the probe to home then `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` must be enabled.

If `Z_MIN_PROBE_ENDSTOP` is enabled then the Z_MIN_PROBE_PIN is ignored during homing.

keremers, 2017-06-26:
https://guides.github.com/features/mastering-markdown/

keremers, 2017-06-26:
The thing is I am using PS_ON pin for Z Min probe. When I enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN it does not check for the Z_MIN_PROBE_PIN either. Besides I was using this setup with previous versinos on Marlin FW like this.

Bob-the-Kuhn, 2017-06-27:
Is PS_ON_PIN defined?  If it is then comment it out.  

If it's defined then that pin is probably set as an output which probably overpowers the probe's signal.

keremers, 2017-06-28:
I don't think this is the case either. Because when I defined Z_MIN_PROBE_PIN to 12 adn issue M119 I see tthe Z is triggered when close to the baseplate. Still Marlin ignores it. So this is not about the pin was defined as output. 

Your initial comment is confusing for me. Because when I enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN the firmware starts the ignore any setting about Z_MIN_PROBE including the Z_MIN_PROBE_PIN. So I'd expect when I disable it and enable Z_MIN_ENDSTOP I'll expect the firmware to use the Z_MIN_PROBE_PIN. According to your commant the setting is ignored in both cases. So what is the use of hving such a setting ? Besides It was working with previous version of the firmware. 
To fix this I've used a workaround. I've moved my proximity sensor to Z-MIN_ENDSTOP pin instead of PS_ON

Bob-the-Kuhn, 2017-06-28:
> If Z_MIN_PROBE_ENDSTOP is enabled then the Z_MIN_PROBE_PIN is ignored during homing.

That statement isn't the complete story.  If Z_MIN_PROBE_ENDSTOP is enabled then the Z_MIN_PIN is checked during homing and the Z_MIN_PROBE_PIN is checked when probing.

I'm glad you found a combination that works.

keremers, 2017-06-30:
Thank you. I'll try them. The curious thing is the pins are what they were previously (in pins_RAMPS.h). But older versions seem to ignore these settings and only now obeying them.

closed 2018-02-09 by thinkyhead