MarlinFirmware/Marlin #7147: Auto Bed Leveling - Bilinear

AIALLEN, 2017-06-26:
@FHeilmann 
Hi， I'm using the anet A8 printer, and I've got some problems to ask you. I used the bilinear auto bed leveling, but when I printed it, I found that the Z axis didn't compensate for the unevenness of the bed,    
Do I need to change anything on the SkyNet 3D firmware?Thank you somuch.

Best wishes

FHeilmann, 2017-06-26:
Hi,
sorry I am not familier with the Skynet3D firmware. Maybe you have to enable the bed leveling after you perform a G28 (in Marlin this is done via M420 S1) but I am not sure how this is implemented in Skynet3D

benlye, 2017-06-26:
@FHeilmann, SkyNet3D is a user-distributed fork of Marlin, pre-configured for Anet printers.

@AIALLEN. try the troubleshooting steps here:
https://3dprint.wiki/reprap/anet/a8/improvement/autobedleveling#troubleshooting

Most issues with with autolevelling seem to come down to printer setup/calibration issues.

AIALLEN, 2017-06-26:
But in this video, the obvious hot bed is tilted, and the Z axis also compensates. How do you implement the automatic leveling of the marlin firmware? Very much hope for your help and guidance.
https://www.youtube.com/watch?v=x8eqSQNAyro&t=46s

fiveangle, 2017-06-26:
If the troubleshooting @benlye suggested doesn't help, erhaps check with [SkyNet3D](https://github.com/SkyNet3D/Marlin/tree/SkyNet3D-Oderwat) project first ?  Or are you running straight Marlin from here ?

Lord-Quake, 2017-06-26:
I'm using SkyNet3D Version 2.4.5 on an Anet A8 with activated "#define AUTO_BED_LEVELING_BILINEAR" in "Configuration.h"
Maybe I can help?

closed 2018-02-07 by thinkyhead