MarlinFirmware/Marlin #7149: unexpected behavior of DEFAULT_AXIS_STEPS_PER_UNIT when porting from Marlin 1.0.1 to 1.1.x

svolpe, 2017-06-26:
Not sure if this is a bug or there is something I'm missing.  I'm working on porting the Geeetech PRUSA i3 REV A
https://www.geeetech.com/geeetech-prusa-i3-a-pro-3d-printer-diy-kit-p-997.html

from Marlin 1.0.x to the new Marlin 1.1.x.  Everything is working great but the Z axis step per unit seems to be too large.  If I have it move 1 mm it move about 3 mm. I can compile the Marlin 1.0.x and the movement is correct but with the Marlin 1.1.x it is not.  
I'm using the fallowing steps per unit for both the 1.0.x and the 1.1.x
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,2560,93}

Is there something else that could be impacting the step size?  I've searched the documents and have not found a reference to any other defines that could impact this.
Any help would be appreciated.

Bob-the-Kuhn, 2017-06-27:
There's a couple of things you can try:
- In configuration_adv.h set MINIMUM_STEPPER_PULSE to 1
- Try increasing the current limit on the stepper.

svolpe, 2017-06-27:
Bob-the-Kuhn,
I will give your recommendations a try.  It is not 100% clear to me how MINIMUM_STEPPER_PULSE would effect it.  I would think if the pulse was too narrow it might miss triggers but not add more.  Also the Y and X axis work perfect and they are the same type of servo and servo controller as the Z.  

I did increase my Z servo current to 1.1V, it was originally about 0.8V.  This made no difference.  

I also changed the DEFAULT_AXIS_STEPS_PER_UNIT {80,80,2560,93} to DEFAULT_AXIS_STEPS_PER_UNIT {80,80,400,93} and now it works perfect but I'm not ok leaving that as the final answer since the original DEFAULT_AXIS_STEPS_PER_UNIT worked for Marlin 1.0.1 so it makes no sense why I would have to change it for Marlin 1.1.x

Bob-the-Kuhn, 2017-06-27:
We've seen strange things on the Z axis that have been cured by the two items I mentioned, mostly by increasing the current.  The minimum step pulse is one of the things that changed from 1.0.1 to the current Marlin.

I don't know why the steps per unit was affected.  All I can say is we've had lots of people convert and I don't remember another where the steps per unit had to change in order to get the right amount of travel.  If you had been seeing printer freezes/resets then I could tell you why (the newer options take up enough CPU cycles that the watch dog timer triggers if the step rate is too high).

thinkyhead, 2017-07-02:
One thing I know is you should start by cutting all your old jerk values in half, then tune upwards from there. Do an `M502` / `M500` after flashing, to make sure defaults take effect.

boelle, 2019-02-19:
@svolpe do you still have problems with the latest bugfix 2.0?

svolpe, 2019-02-19:
Bo,
I'll test it in the next few days and let you know.

closed 2019-02-20 by thinkyhead