MarlinFirmware/Marlin #7152: G33  runs but the result are only zeroes

novio8, 2017-06-26:
Hi yesterday I downloaded and installed
https://github.com/LVD-AC/Marlin-AC is Marlin 1.1.3 with #6850 included
I'v run G33 or G33 P4 C0.02, bed leveling starts, one round
and stops. Repetier report:
19:37:27.093 : Calibration OK std dev:-2147483.750
19:37:27.096 : .Height:265.64 Ex:+0.00 Ey:+0.00 Ez:+0.00 Radius:163.00
19:37:27.097 : .Tower angle : Tx:+0.00 Ty:+0.00 Tz:+0.00
19:37:27.100 : Save with M500 and/or copy to Configuration.h

So everything is 0.
What am I doing wrong again ?
Before upgrading I manually leveled the bed and I'v got some good prints.
Any suggesyions LVD-AC ?
thank you

fiveangle, 2017-06-26:
700 issues #ohmy !

LVD-AC, 2017-06-28:
Is this a repeat issue from #6997 ?

novio8, 2017-06-28:
Well I just  upgraded Marlin as you suggested, manually leveled the bed
and got some decent prints.
Then I enabled UBL as you suggested,  

//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING

cleared EEPROM and I'v run G33 several times but I'v always got all Ex,Ey,Ez, Tx, Ty, Tz values Zero.
Printer probes the points respectively to P value, but only once and
returns to home, regardles if I run G33,  G33 P3, G33 P7 C0.02 ... etc
Printer works OK, I can print, but I would like to have ABL and help
improving Marlin if needed
With previous version 1.1.2, G33 behaved as described in #6997

LVD-AC, 2017-06-28:
I do not know what happens, but give 1.1.x-AC at https://github.com/LVD-AC/Marlin-AC a try; I made some changes for you, see if they work?

novio8, 2017-06-28:
installed your "new" version
with myconfig and config.adv files
run G33 and I get:

19:28:55.212 : echo:busy: processing
19:28:55.965 : Iteration : 01                                    std dev:999.000
19:28:55.966 : .Height:0.00    Ex:0.00  Ey:0.00  Ez:0.00    Radius:0.00
19:28:55.969 : .Tower angle :    Tx:0.00  Ty:0.00  Tz:+0.00
19:28:57.229 : echo:busy: processing
19:28:59.244 : echo:busy: processing
19:29:01.261 : echo:busy: processing
19:29:03.277 : echo:busy: processing
19:29:05.294 : echo:busy: processing
19:29:05.768 : Calibration OK                                    rolling back.
19:29:05.772 : .Height:266.68    Ex:-0.10  Ey:-0.10  Ez:+0.00    Radius:162.80
19:29:05.774 : .Tower angle :    Tx:+0.00  Ty:+0.00  Tz:+0.00
19:29:05.776 : Save with M500 and/or copy to Configuration.h
19:29:06.241 : ok

and this time it rolls back

LVD-AC, 2017-06-28:
something is wrong with your probe hardware or configuration

novio8, 2017-06-29:
if I run M48 P12 I get 0.006 deviation
so probe is working

novio8, 2017-06-29:
I reinstalled 1.1.2 and G33 runs as usuall

LVD-AC, 2017-06-29:
Something is wrong then in your configuration. Check if the M665 B(ed radius for probing) is not set to high for your eccentric probe...

novio8, 2017-06-29:
yes you are right I reduced delta probing radius and G33 is alive again.
Ver. 1.1.2 gave an error if radius was too big, but ver 1.1.3 gives no error

LVD-AC, 2017-06-29:
I've made a new version 1.1.3b-AC with improved testing for eccentric probes (you have to define DELTA_PHYSICAL_RADIUS in configuration.h). See also #7184

MagoKimbra, 2017-06-29:
The problem is that in the new version there is the Force Iteration sub command.
`
Fn  Force to run at least n iterations and takes the best result

const int8_t force_iterations = parser.intval('F', 1);

`
If you omit the command, however, it only makes one of the iteration and therefore does not adjust anything.

Try it with G33 F5 and you'll see it go ...

novio8, 2017-07-02:
LVD-AC installed lastest AC version,
G33 and UBL leveling both work for me
thank you

MagoKimbra, 2017-07-02:
Yes, but I insist if I omit F, force_iterations is equal to 1, so it does one iteration because in the do while cycle it just comes out iteration is greater than force_iterations.
In my opinion if you omit F you should do normal cycles until you find the right values, while with F then you must make the number put and get the best..

LVD-AC, 2017-07-02:
That is exactly what it does: when you omit F it iterates until no more improvement is found, but if your printer is already calibrated it just checks with 1 iteration and doesn't change anything. With F set it forces at least the set number of iterations and takes the best result.

MagoKimbra, 2017-07-02:
No, if omitted F finishes always after 1.

LVD-AC, 2017-07-03:
Just did G33 without F on an uncalibrated printer; it did 7 iterations...

LVD-AC, 2017-07-03:
Close issue

closed 2018-02-06 by thinkyhead