MarlinFirmware/Marlin #7155: SD is not init on startup ?

fred2088, 2017-06-27:
Dear Sirs,

SD is not init on startup while RC1.0.8 work fine, is there default #define is changed ?

I have try to #define sd_detect_pin -1
also try to disable sd_detect_inverted, but no lucik

TIA

Bob-the-Kuhn, 2017-06-27:
Check SDSUPPORT in configuration .h.  It's disabled by default.

fred2088, 2017-06-27:
#define SDSUPPORT already, but need to select "init sd card" manually every startup the printer

Bob-the-Kuhn, 2017-06-27:
What host software are you using?

What happens (or doesn't happen) when you do not issue the  "init sd card" ?

Please post the log of the following sequence:
1. power up/reset the controller
2. issue `M20` command to the controller

fred2088, 2017-06-28:
I am not connecting any pc, power on the printer, and the sd card is not init by default(enabled from firmware) after I press the init sd card, it can list and print the file from sd card, 

Regards

fred2088, 2017-06-28:
M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
Begin file list    <<<<<<<<<<<<<<<< M20, the SD card is always there
End file list
ok
echo:enqueueing "M21"  <<<< "INIT SD CARD from LCD Menu"
echo:SD card ok
Begin file list   <<< M20 again the SD card is always there
YODA.GCO 18065447
BODY.GCO 17498505

Bob-the-Kuhn, 2017-06-28:
Please ZIP up your entire Marlin directory and attach it to your reply.  I need to know what version of Marlin you are working with.

As best I can tell the SD_DETECT_PIN is disabled or doesn't exist on your system.  When I do that the LCD now shows two LCD menu lines:
```
Print from SD ->
Change SD card
```

When I click `Print from SD ->` I get the file list plus a `Refresh` menu item. 

When I click the `Change SD card` I see `echo:enqueueing "M21"` in the console log.

----

I'm not seeing "init sd card" on the LCD, In Repetier Host or Pronterface.  You'll need to take a picture or a screen shot so I understand what you mean by "press the init sd card".

fred2088, 2017-06-28:
[image]

Bob-the-Kuhn, 2017-06-28:
The picture gave me enough info to track down the code.

I need more info.  Please do the following:
- hook up the USB cable to the PC
- start the host software
- clear the host software's log
- connect to the printer
- wait for the startup messages to finish
- use the LCD to issue the init SD card command
- copy the entire log into a .TXT file
- attach the .TXT file to your reply.

I'd still like to see your configuration.h file.  Make a copy of it and then add .TXT to the file name after the .h.  Now you can attach the copy to your reply.

fred2088, 2017-06-29:
it's seem don't have init sd card from marlin when boot up
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:enqueueing "M21"  < ------ Manually init sd card from LCD Menu
echo:SD card ok

[github.txt](https://github.com/MarlinFirmware/Marlin/files/1110741/github.txt)

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1110742/Configuration.txt)

Bob-the-Kuhn, 2017-06-29:
You're not running Marlin bugfix-1.1.x.  You're running a modified version of it.

I can't find the software for a Tinyboy E8.

I can't find a schematic or a pinout for the TINYOLED.

At a minimum I need the software that you're running.  Please try to attach the .ZIP file again.  If that doesn't work then use a service like [DropBox](https://www.dropbox.com/downloading) and include the link.  If you do use DropBox get the Basic version - that one is free.

Can you point me to any technical documentation on the TINYOLED?

fred2088, 2017-06-29:
find the problem that when enable this define, it will not init sd on starting the printer
#define SHOW_CUSTOM_BOOTSCREEN

otherwise it works fine ...

Bob-the-Kuhn, 2017-06-29:
I'd never have guessed that they would interact.

Glad you found a solution.

thinkyhead, 2017-07-02:
I believe the problem is that `safe_delay` is being called from `lcd_implementation_init`, preventing initialization from completing.

You can try this…. In `ultralcd.cpp` find the `lcd_init` function (around line 4050). Move these lines near the end of that function…
```cpp
  #if ENABLED(SDSUPPORT) && PIN_EXISTS(SD_DETECT)
    SET_INPUT_PULLUP(SD_DETECT_PIN);
    lcd_sd_status = 2; // UNKNOWN
  #endif
```
…before the call to `lcd_implementation_init`.

If that doesn't fix the issue, a small re-work of the lcd init routines will be done to address it.

thinkyhead, 2017-07-02:
See if #7211 has any effect. It's a minor cleanup of the boot screen code.

Uinden, 2018-08-24:
Hi. 
I have same problem.
My printer is `Anet A6` and boad is `Anet V1.0`
Host is `Repeiter-Host V2.1.3`
Tried `Marlin 1.1.9` release and `bugfix-1.1.x` branch, both have same problem.
Modified configs from `example_configurations/Anet/A6/`
After boot I have menu items `No SD card` and `Init. SD card`
After I press `Init. SD card` it start to work.
`M20` shows this:
```code
00:50:20.548 : N15 M20*37
00:50:20.552 : Begin file list
00:50:20.552 : End file list
00:50:20.553 : ok
```
Clear log, power on, wait, press `Init. SD card` produces this log:
[bugfix-boot-log.txt](https://github.com/MarlinFirmware/Marlin/files/2320001/bugfix-boot-log.txt)
Enabling `#define SHOW_CUSTOM_BOOTSCREEN`, downloading random `_Bootscreen.h` and adding `#define CUSTOM_BOOTSCREEN_TIMEOUT 1900` did not help.
Attaching my configs.
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/2319989/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/2319990/Configuration_adv.h.txt)

Any ideas?

LemaruX, 2018-08-25:
Same issue here on an Ender 3 since updating to 1,1.9

gloomyandy, 2018-08-26:
A similar problem was reported on the Marlin facebook group:
https://www.facebook.com/groups/1049718498464482/permalink/1331365776966418/?__xts__[0]=68.ARC-du0IBlVP7E8a5AnAsNGzotyR6sWgSnenHfwdTBuGYho-6wMizdNqth-O3txP8CLkyBlcqMsea8k7bQ2i2-wG1OoSNWIx_Mf3tESxbNiKo-ClvuvuzylgO-V3I71tOZNRk6E5SND3naS380DwI8jiguLZ6jgysby8xgUlcVQKdxEQX1lR4w&__tn__=-R

There is a possible solution in that thread.

Uinden, 2018-08-26:
@gloomyandy 
Adding 
```code
  /* power on init. sd card */
  #if ENABLED(SDSUPPORT)
    if (!card.cardOK) card.initsd();
  #endif
```
In `void setup()` of `Marlin_main.cpp` solved the problem. 
Thanks!

LemaruX, 2018-08-26:
I'll give it a go, thanks. Though not sure that the cause is no SD detect pin as the SD card used to initialise fine before upgrading Marlin. 

If this sorts it though then all good

closed 2018-08-30 by thinkyhead

fred2088, 2018-11-02:
2.0.x got the same problem that can't init sd on startup,

 while 1.1.x can init sd correctly on startup

thinkyhead, 2018-11-02:
@fred2088 — Try adding this to the end of the `setup()` function in `Marlin.cpp`:
```cpp
 #if ENABLED(SDSUPPORT) 
  if (!card.cardOK) card.initsd();
 #endif
```

billsierer, 2018-11-06:
I was running Marlin 1.1.8 and everything work fine I uploaded my firmware to Marlin 1.1 .9 I set my defines and I too am having the exact same issue my SD card is installed I turn on the printer but it says no SD card found but if I go into menu and click initialize SD card it will load just fine why is it not Auto initializing I have all the Define set to what they need to be but it's still not working.  I did not have this issue with Marlin 1.1.8 and I also compared my configuration files and configure advance and everything matches perfectly

thinkyhead, 2018-11-07:
Try the workaround I posted above and see if it helps.

amedranogil, 2018-12-28:
Same issue in 1.1.9.
Proposed solution works.
Strange there is no pull request for this...

momo42, 2019-01-06:
I'm also affected with unmodified Marlin 1.1.9 on Creality Ender 2.
The proposed solution also worked for me.

momo42, 2019-01-06:
I tested with 1.1.x bugfix release and this bug is persistent!

leolimajr, 2019-01-12:
Proposed solution also worked for me! I'll create a PR for this.

robbycandra, 2019-02-03:
My solution is to change the code at the end of setup() at Marlin_main.cpp:

From 

```
#if ENABLED(SDSUPPORT) && DISABLED(ULTRA_LCD)
    card.beginautostart();
  #endif

```
To :

```
#if ENABLED(SDSUPPORT) 
    card.beginautostart();
  #endif
```

thinkyhead, 2019-02-05:
That helpfully causes the `card.beginautostart()` code to be invoked twice. Once when the LCD is done initializing, and again at the end of `setup()`.

merdok, 2019-08-12:
Hello,

why this is still not fixed? I had to add the following code on my Ender 3 with the 1.1.5 Mainboard with the latest 1.1.x bugfix:

```
  /* power on init. sd card */
  #if ENABLED(SDSUPPORT)
    if (!card.cardOK) card.initsd();
  #endif
```

InsanityAutomation, 2019-08-12:
> 
> 
> Hello,
> 
> why this is still not fixed? I had to add the following code on my Ender 3 with the 1.1.5 Mainboard with the latest 1.1.x bugfix to make the sd card init on start:
> 
> ```
>   /* power on init. sd card */
>   #if ENABLED(SDSUPPORT)
>     if (!card.cardOK) card.initsd();
>   #endif
> ```

1.1.x is EOL, try 2.0

merdok, 2019-08-12:
I guess a lot of people still using the 1.1.x bugfix on the Creality Melzi board, so shouldn't at least this be fixed? I mean there are still fixes coming in to this branch so why not this?

nyxaria, 2019-10-19:
bump, can this be integrated upstream into 1.1.x? Just because it's EOL doesn't mean it shouldn't be supported for a few years, at least.

leolimajr, 2019-10-19:
I'm using a fix since long ago on my fork so I create a PR for this so everyone can profit from it. Credits to @robbycandra.