MarlinFirmware/Marlin #7158: Connecting 2 Controllers with I2C Protocol

AllieGrey, 2017-06-27:
Hi, 

I need to figure out how to use I2C Protocol to connect 2 controller boards. Does anyone have a startup guide or tips/instructions as to how to do so?

I am trying to run the print bed with 3 motors and I am also thinking about using multiple extruders. I want to add an extra controller board and I'm trying to understand how to control it.

Thank you for helping!

Bob-the-Kuhn, 2017-06-27:
What is the problem that you are trying to fix?

If you want to run more steppers than is supported by your current controller then you have only a few options:
- Replace your current controller with one that supports more steppers.
- Use external stepper drivers. 

External stepper drivers require three logic signals each from the controller.  They range from $5 boards (that have a socket to plug in a stepper stick) to large self contained units.

thinkyhead, 2018-02-09:
@AllieGrey See https://github.com/MarlinFirmware/Marlin/issues/4776#issuecomment-246262879

closed 2018-02-09 by thinkyhead