MarlinFirmware/Marlin #7161: BED Temperature in LCD Full Graphics

igortomiatti, 2017-06-27:
My heated bed takes a while to heat up, so I have to leave the Thermal Runway turned off.

But the option to control the temperature of the BED by the LCD does not appear.

If the Thermal Runway is enabled, the bed temperature control menu appears normally.

This only happened after I upgraded to 1.1.3, in RC8 version works normally.

Bob-the-Kuhn, 2017-06-28:
Confirmed - 1.1.3 has this "characteristic".

bugfix-1.1.x doesn't have this issue.  

The quickest & best way to resolve this is to download [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x).  I don't know if 1.1.3 is going to have any more updates as 2.0 is a week or so away.

The fix is to change line 2754 in ultralcd.cpp from `#if WATCH_THE_BED` to `#if HAS_TEMP_BED`.

----

Are you getting "Thermal Runaway" or "Heating Failed" errors?

False "Thermal Runaway" errors are the result of `THERMAL_PROTECTION_BED_PERIOD`  being set too tight for your printer.  Try doubling the current value and see if the error message goes away.

False "Heating Failed" errors are the result of `WATCH_BED_TEMP_PERIOD`  being set too tight for your printer.  Try doubling the current value and see if the error message goes away.

These items are in configuration_adv.h

fiveangle, 2017-07-01:
After fixing your LCD, you really should resolve your temperature settings so you can enable thermal protection.  Doubling the period is a good start and should resolve so you can't accidentially burn your house down.  Disabling thermal protection should really only be used temporarily while performing manual PID tuning.

thinkyhead, 2017-07-02:
I remember we did fix this — it was the wrong condition in the lcd code. Much better now. Version 1.1.4 will be out shortly…

closed 2017-07-02 by thinkyhead