MarlinFirmware/Marlin #7166: f4246dc Broke my Z axis

jimovonz, 2017-06-28:
Downloaded the latest bugfix (after commit f4246dc) and my z axis homes OK but completely ignores G2 commands (X & Y fine). Reverted to the previous days code and all is fine again (only preserved Configuration.h/Configuration_adv.h)

tcm0116, 2017-06-28:
I'm having the same issue. Glad it's not just me...

tcm0116, 2017-06-28:
Looks likes disabling **FASTER_GCODE_PARSER** in _Configuration_adv.h_ works as a temporary fix for this issue.

Justinjhcs1, 2017-06-29:
Mine is still broken so I went back to Sprinter No remote but at least it works.

fiveangle, 2017-06-29:
#7182 ?

Tannoo, 2017-06-29:
I'm not sure why, but it's not parsing Z:
```
Got letter G at index 5 (has_num)
Set bit 7 of index 0 | param = 0x0006
Got letter Z at index 8 (has_num)
Command: G800 G0 Z10 (G800)
 args: "G "

Code 'G':
    float: 0.00
     long: 0
    ulong: 0
   millis: 0
   sec-ms: 0
      int: 0
   ushort: 0
     byte: 0
     bool: 0
   linear: 0.00
  celsius: 0.00
```

Tannoo, 2017-06-29:
I found in `gcode.h`, two changes gets Z working:
```
      const uint8_t ind = LETTER_OFF(c);
      if (ind > COUNT(param)) return;           // Only A-Z     ** CHANGED FROM ">=" to ">"
      SBI(codebits[PARAM_IND(ind)], PARAM_BIT(ind));        // parameter exists
      param[ind] = ptr ? ptr - command_ptr : 0;  // parameter offset or 0
      #if ENABLED(DEBUG_GCODE_PARSER)
        if (debug) {
          SERIAL_ECHOPAIR("Set bit ", (int)PARAM_BIT(ind));
          SERIAL_ECHOPAIR(" of index ", (int)PARAM_IND(ind));
          SERIAL_ECHOLNPAIR(" | param = ", hex_address((void*)param[ind]));
        }
      #endif
    }

    // Code seen bit was set. If not found, value_ptr is unchanged.
    // This allows "if (seen('A')||seen('B'))" to use the last-found value.
    // This is volatile because its side-effects are important
    static volatile bool seen(const char c) {
      const uint8_t ind = LETTER_OFF(c);
      if (ind > COUNT(param)) return false; // Only A-Z     ** CHANGED FROM ">=" to ">"
      const bool b = TEST(codebits[PARAM_IND(ind)], PARAM_BIT(ind));
      if (b) value_ptr = command_ptr + param[ind];
      return b;
    }
```

Tannoo, 2017-06-29:
In `gcode.h`,
I changed LN#117
`      if (ind >= COUNT(param)) return;           // Only A-Z`
to
`      if (ind > COUNT(param)) return;           // Only A-Z`

and LN#134
`      if (ind >= COUNT(param)) return false; // Only A-Z`
to
`      if (ind > COUNT(param)) return false; // Only A-Z`

Seems to fix it for me.

Tannoo, 2017-06-29:
DEBUG now report the wrong letter..
```
Got letter G at index 5 (has_num)
Set bit 7 of index 0 | param = 0x0006
Got letter Z at index 8 (has_num)
Set bit 2 of index 3 | param = 0x0009
Command: G800 G1 Z10 (G800)
 args: "C Z "

Code 'C':
    float: 0.00
     long: 0
    ulong: 0
   millis: 0
   sec-ms: 0
      int: 0
   ushort: 0
     byte: 0
     bool: 0
   linear: 0.00
  celsius: 0.00

Code 'Z':
    float: 10.00
     long: 10
    ulong: 10
   millis: 10
   sec-ms: 10000
      int: 10
   ushort: 10
     byte: 10
     bool: 1
   linear: 10.00
  celsius: 10.00
```

Tannoo, 2017-06-29:
>Looks likes disabling `FASTER_GCODE_PARSER` in Configuration_adv.h works as a temporary fix for this issue.

Although, I had to do this also to get Repetier to work.

I can send `G1 Z10` and it work fine.. even through Repetier... But, I click the button in Repetier to do the same thing, nothing moves.

Bob-the-Kuhn, 2017-06-29:
@Tannoo  - see if changing only line 107 in gcode.h gets Z movement.
Is: `#define LETTER_OFF(N) ((N) - 'A' + 1)`
To: `#define LETTER_OFF(N) ((N) - 'A')`

I believe the root cause is the array **param** has indices 0-25 but LETTER_OFF(N) returns 1-26.

Tannoo, 2017-06-29:
That works, thank you.

Tannoo, 2017-06-29:
It now works as expected with and without `FASTER_GCODE_PARSER` and with or without Repetier.

Tannoo, 2017-06-29:
And the debug is okay again.. lol

Bob-the-Kuhn, 2017-06-29:
OK - I'll push a PR immediately.

Bob-the-Kuhn, 2017-06-29:
I've just merged PR #7183 to fix this.

Andrei-Pozolotin, 2017-06-30:
seems to work now

boelle, 2019-02-19:
@jimovonz does it also work for you now?

if so maybe consider closing the issue?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-03-16:
I can't reproduce, G1 moves are OK

@Bob-the-Kuhn can you close it or has @thinkyhead  to do that?

closed 2019-03-17 by p3p