MarlinFirmware/Marlin #7172: Bilinear bed leveling with manual probing

kozolika, 2017-06-28:
Hi, 
I am struggling to get the ABL working with manual probing.
Using Marlin 1.1.3 stable with Anycubic Kossel Linear Plus.

The issue is that when using G29 or the LCD, on the first movement the nozzle will go down and hit the bed and the calibration cannot be done. I already calibrated the height of the print area.

Attached are the config files. What am I missing? How can I debug to see where the nozzle wants to go?

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1109207/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1109208/Configuration_adv.txt)

dagobertch, 2017-06-28:
I tried yesterday eveneing and found the same issue.
I am using Marlin 1.1.3 on a 101Superhero (modified 101Hero) and got the same result.
G29 (or Bed Leveling on LCD) starts to move the head down and hit the buliding plate...
my config looks similar: [https://www.dropbox.com/s/7i97h8tr51928pf/Configuration.h?dl=0](url)

Bob-the-Kuhn, 2017-06-28:
Z_MAX position is known because homing is done in the + direction on deltas.  Z0 is inferred by setting the logical Z_MAX value to `DELTA_HEIGHT`.  If `DELTA_HEIGHT` is too large then Z0 will be below the bed.

Try slowly lowering the nozzle until it's near the bed.  Then use the paper/shim method to get the nozzle within 0.2mm of the bed.  If you then issue a M114 command the Z value should be less then 0.2mm.  If it's bigger than that you'll need to adjust the `DELTA_HEIGHT` 

Example: if M114 says Z = 13mm after the paper test, then subtract 13 from `DELTA_HEIGHT`.  Since you're doing manual probing I'd suggest subtracting 15 so that the nozzle is always above the bed when you start the manual probing.

----

@dagobertch - the link just gives me a "page not found" error.

I just ran across the 101hero & the super upgrades.  Curiosiity - what did it cost to do the upgrades?

dagobertch, 2017-06-28:
Now the link should work.
I can print with the printer, so the delta height is Ok. I removed also the Glas plate and it also does not stop a few cms below...

The upgrades (Motors, hotend, extruder, Electronic and some alu for stiffing the frame) was about 100€. More than the whole printer :) but I have learned a lot about 3D printing through the upgrades...
A Heatbed will be installed as las upgrade, next Project is a the Hypercube Evo, pront d on the 101Superhero :) but for that I need full calibration, because I have seen, that some XY angles are not correct...

kozolika, 2017-06-30:
I was able to get the manual probing working with bilinear bed leveling.
It looks like this feature is not implemented correctly for delta printers in 1.1.3 stable. The motion commands during ABL are not translated to delta coordinates, but are sent as cartesian to the steppers. I fixed this by creating a new function and calling that during manual probing.
Here is what I did:
1. add function to `Marlin_main.cpp`:
`inline void line_to_current_position_kinematic() {
	planner.buffer_line_kinematic(current_position, feedrate_mm_s, active_extruder);
}`
2. replace calls to `line_to_current_position()` with calls to `line_to_current_position_kinematic()` in `_manual_goto_xy()`

I attached my `Marlin_main.cpp` for reference.
[Marlin_main.cpp.txt](https://github.com/MarlinFirmware/Marlin/files/1114703/Marlin_main.cpp.txt)

Anycubic, 2017-08-26:
@kozolika thanks a lot for you post! Actually the bug is still there in 1.1.4 , I wonder if this feature was ever tested before being released and the outcome can be pretty dangerous (glass or hotend broken). I fixed the code the way you suggested and it worked great, thanks again

Maesto, 2017-10-18:
I can confirm this behavior for Marlin Version 1.1.5
But I can't say if  kozolika's "patch" is working for me. It now finds the first point without a problem. But each later point seems to be having the same problem as the first point since only single machine axis are moved.

Would it break other functions, if you completely replace the function line_to_current_position() with line_to_current_position_kinematic()?

thinkyhead, 2017-11-09:
See if #8342 clears up manual probing now. Moves were incorrect for kinematic machines, and as correctly pointed out, this functionality also wasn't tested on non-kinematic machines so the code was never updated for them. Here's the proposed fix…
```cpp
  inline void _manual_goto_xy(const float &rx, const float &ry) {

    #if MANUAL_PROBE_HEIGHT > 0
      const float prev_z = current_position[Z_AXIS];
      do_blocking_move_to_z(MANUAL_PROBE_HEIGHT, homing_feedrate(Z_AXIS))
    #endif

    do_blocking_move_to_xy(rx, ry, MMM_TO_MMS(XY_PROBE_SPEED));

    #if MANUAL_PROBE_HEIGHT > 0
      do_blocking_move_to_z(prev_z, homing_feedrate(Z_AXIS))
    #endif

    stepper.synchronize();

    current_position[X_AXIS] = rx;
    current_position[Y_AXIS] = ry;

    #if ENABLED(PROBE_MANUALLY) && ENABLED(LCD_BED_LEVELING)
      lcd_wait_for_move = false;
    #endif
  }
```

closed 2017-11-16 by thinkyhead