MarlinFirmware/Marlin #7174: Bed Leveling Option on LCD

devekrehbiel, 2017-06-28:
Thanks for doing such amazing work. I noticed there used to be a "Bed Level" feature under Prepare just below the Homing feature that is gone. I do not want to enable LCD leveling because I just want the feature to go through the leveling process on command, not just prior to printing. Does this make sense?

Nickduino, 2017-06-28:
I don't get it. You don't want to activate LCD leveling but you want the leveling option on your LCD?

devekrehbiel, 2017-06-28:
Well, I want it the way it used to be. Like Auto Home feature. Its missing completely. If I enable LCD leveling, I have to use Mesh Leveling and change a whole bunch of parameters. The program wont compile unless you choose Mesh.

Nickduino, 2017-06-28:
Agreed (https://github.com/MarlinFirmware/Marlin/issues/7176). We should be able to turn the LCD leveling on without turning the mesh leveling on.

benlye, 2017-06-28:
Enable `#define CUSTOM_USER_MENUS` and add a user menu option which just runs G29?

Tannoo, 2017-06-29:
`LCD_BED_LEVELING` gives the option of leveling manually on the LCD and requires `MESH_BED_LEVELING` or `PROBE_MANUALLY`.

`PROBE_MANUALLY` is leveling without a probe (with or without `LCD_BED_LEVELING`) and requires one of the following:
```
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING
```

Tannoo, 2017-06-29:
>Agreed (#7176). We should be able to turn the LCD leveling on without turning the mesh leveling on.

You can... just enable something other than mesh leveling.
How can you tell the printer to manually level without a leveling plan of what to do?

Tannoo, 2017-06-29:
If you enable something other than MESH_BED_LEVELING, there's no need to have LCD_BED_LEVELING enabled.

Tannoo, 2017-06-29:
I hope I helped clear this up.

devekrehbiel, 2017-06-29:
Tannoo, no you can't. If you choose anything other than mesh, you get a compile error. Try it and see. It may be stupid to have it there (without mesh) for some, but its the way I test to make sure my auto level sensor is working from time to time. Its there just for test purposes and peace of mind! :) I know Auto Home tests it too, but not in all 9 places on the bed. When you build your own 3D Printers from a DIY kit (many of them), its a good test to have in there for redundancy too make sure its working. I will add it per Bob for now, but it IS missed by others too.

Tannoo, 2017-06-29:
LCD_BED_LEVELING can only be used with MESH_BED_LEVELING or PROBE_MANUALLY.

There is no need for LCD_BED_LEVELING with any ABL modes unless PROBE_MANUALLY is enabled.

Tannoo, 2017-06-29:
LCD_BED_LEVELING is not a leveling scheme.

Lutunen, 2017-06-29:
I have same issue; For manual testing (with paper after G28, G29) how the leveling functions with LCD is handy. This was possible still in 1.1.0

devekrehbiel, 2017-06-29:
Sometimes logic doesn't have anything to do with it.  We found ways to use that feature you didnt expect. Here is the problem. If I choose Bi-Linear for example. The ONLY time I know that it is working is during a print. I have to wait until the heat bed, extruder heat, everything is at temp, then I have to watch it auto home, then bed leveling. At no time before that am I allowed to see if just the damn bed leveling routine is working! I like to check these routines before wasting my time. That is the same theory behind your Auto Home function in the LCD. I just want to check to see if it is going to auto level is all.

Don't put it in LCD Bed Leveling. Put it in the same place you allowed us to choose Auto Home. We loved that feature and then it was gone. I am just begging you to PLEASE put it back! These sensors are not 100% reliable and there is a time every now and then when the sensor chokes and it tries to put a hole in my glass beds. If I can test that in all 9 places prior to a long print job, I am HAPPIER! Just make us happy. :)

benlye, 2017-06-29:
Seriously, use custom user menus to define a menu item which runs G29.  Job done.

Tannoo, 2017-06-29:
I see that the bed leveling menu option is not there with `AUTO_BED_LEVELING_BILINEAR` and `FIXED_MOUNTED_PROBE` are enabled and `LCD_BED_LEVELING` is disabled.

That's not right. I have to look into this later.

Tannoo, 2017-06-30:
I got the leveling option again on the LCD by inserting these lines:
```
    #elif PLANNER_LEVELING
      MENU_ITEM(gcode, MSG_BED_LEVELING, PSTR("G28\nG29"));
```
into `ultralcd.cpp` section like so (starting at approx. LN#2430):
```
    //
    // Level Bed
    //
    #if ENABLED(AUTO_BED_LEVELING_UBL)
      MENU_ITEM(submenu, MSG_UBL_LEVEL_BED, _lcd_ubl_level_bed);
    #elif ENABLED(LCD_BED_LEVELING)
      #if ENABLED(PROBE_MANUALLY)
        if (!g29_in_progress)
      #endif
      MENU_ITEM(submenu, MSG_BED_LEVELING, lcd_bed_leveling);
    #elif PLANNER_LEVELING
      MENU_ITEM(gcode, MSG_BED_LEVELING, PSTR("G28\nG29"));
    #endif
```

You can use this until an official correction is made.

devekrehbiel, 2017-06-30:
Thanks for this. You guys are doing a great job of juggling between customer whining and making the perfect system.

devekrehbiel, 2017-06-30:
which file?

thinkyhead, 2017-07-02:
> I got the leveling option again on the LCD by inserting these lines:

A decent fix.

closed 2017-07-02 by thinkyhead

TullioC, 2021-01-23:
I know this post is very old, but the solution is perfect for me.
when Tanoo user write "see below", in which file insert this code?

 got the leveling option again on the LCD by inserting these lines:

    #elif PLANNER_LEVELING
      MENU_ITEM(gcode, MSG_BED_LEVELING, PSTR("G28\nG29"));

Tannoo, 2021-01-24:
Because this is old and I have not seen if the files have changed, it was put:
 "into ultralcd.cpp section like so (starting at approx. LN#2430)"