MarlinFirmware/Marlin #7175: Filament sensor LCD Monitoring

studiodyne, 2017-06-28:
Here is a code , to have  , a LCD MESSAGE , when , filament is detected or not . 

Filament sensor problem lcd show a message , if all ok  , the lcd show the welcome msg

Very usefull !
- If you have a led on your sensor , you don't know if the motherboard have the information 
- Because of the print security priority , the firmware can start a print without ask you about sensor bad state and will run a print without filament warning and impossible to know that
- In the case of an assembling or creating a printer , you need a monitoring to try your filament runout guiding and detection system 
- And when you don't know if your sensor is true or false active state , you can test it now

I understand nothing on this website , so i ask ' ROXY , thinkyhead ' and many other to write a new topic , and to make my contribution the more visible as possible , may be , you can add it , on the release candidate , the code is poor , very light and no crashing possibility
I use only one bool data allocation , and one constant ' the message '
I use it only when not printing to not use the ressource when printing

Here is the code  

To use it , you know that you need to activate all the filament runout switch , and the pins that you use 
*****************************************************************************************
In  your ' LANGUAGE.H ' 
// french version , but you can place your own
#define FILAMENT_RUNOUT_STATUS_MSG          "..Filament absent.. " 
********************************************************************************************
in your ' MARLIN.CPP'  line 9920

bool FILAMENT_RUNOUT_STATUS=0; // one bit global variable for minimal memory allocation

void manage_inactivity(bool ignore_stepper_queue/*=false*/) {  // existing code 
  
  #if ENABLED(FILAMENT_RUNOUT_SENSOR) 
  // Only when not printing 
   if ( !IS_SD_PRINTING ) { 

  // Reading the filament warning pin or pins " i use 2 pins , better for testing each separatly'
  if (
      (READ(FIL_RUNOUT_PIN)||(READ(FIL_RUNOUT_PIN2)) == FIL_RUNOUT_INVERTING)
        )

 // Screen Message to show filament default and status =1            
      { LCD_MESSAGEPGM(FILAMENT_RUNOUT_STATUS_MSG); FILAMENT_RUNOUT_STATUS=1;}

 // If filament pins now ok and status =1,  new screen message ' welcome '     
   else if ( FILAMENT_RUNOUT_STATUS  )
             
      { LCD_MESSAGEPGM(WELCOME_MSG); FILAMENT_RUNOUT_STATUS=0;}
       
   }       
   #endif
***********************************************************************************************************

Hope it will be usefull for all

My first contribution ;-)

Tannoo, 2017-07-01:
Issuing an M119 will already report the filament runout sensor status.

ebraiman, 2017-11-16:
studiodyne,
 
I had a similar issue with knowing what was occurring on servo pins.  It appears my printer LCD and a pin conflict in pins_RAMPS.h kept me working in circles.  I've enclosed a video I made after figuring out the issue.  Let me know if this helps ...
https://youtu.be/PiojWy5PDZE

Regards,
Ed

closed 2018-02-09 by thinkyhead