MarlinFirmware/Marlin #7176: #error "LCD_BED_LEVELING requires MESH_BED_LEVELING or PROBE_MANUALLY."

Nickduino, 2017-06-28:
When I turn the LCD bed leveling on **with PROBE_MANUALLY turned on as well**, I get a compilation error.

I have to activate MESH_BED_LEVELING **and** PROBE_MANUALLY for it to work.

Tannoo, 2017-06-29:
That is because `PROBE_MANUALLY` needs one of the following defined:
```
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING
```

Tannoo, 2017-06-29:
This is the sanity check:
```
/**
 * LCD_BED_LEVELING requirements
 */
#if ENABLED(LCD_BED_LEVELING)
  #if DISABLED(ULTIPANEL)
    #error "LCD_BED_LEVELING requires an LCD controller."
  #elif DISABLED(MESH_BED_LEVELING) && !(HAS_ABL && ENABLED(PROBE_MANUALLY))
    #error "LCD_BED_LEVELING requires MESH_BED_LEVELING or PROBE_MANUALLY."
  #endif
#endif
```
`!(HAS_ABL && ENABLED(PROBE_MANUALLY))` is what's giving you the error. It's looking for a type of leveling to be enabled also.

Nickduino, 2017-06-29:
Could it be that the documentation isn't completely clear?
I thought I could do a manual bed leveling (with the bed screws) asking the printer only to park the nozzle at the four corners.

Tannoo, 2017-06-30:
I do that also, but it's so rare, that I do it manually -- 
`G28 or Auto Home from LCD.
`G1 X10 Y10`, adjust.
`G1 X190 Y10`, adjust.
`G1 X190 Y190`, adjust.
`G1 X10 Y190`, adjust.

I do it that way because setting up bed leveling to do 4 points requires changing the firmware settings accordingly and flashing. Then changing things back for the actual leveling.

Tannoo, 2017-06-30:
I do use Unified Bed Leveling now. As a lot can be done on-the-fly without changing the firmware.

Roxy-3D, 2017-06-30:
You can do a partial G29 P1 T U using UBL to help with the physical adjustment of the bed.  The U is for Uneven bed.  It will spread the probe points out as far as possible.  When you have enough points sampled...  You stop it and physically level the bed.   And then do it again until the bed is as level as you can get it.

Nickduino, 2017-06-30:
> I do it that way because setting up bed leveling to do 4 points requires changing the firmware settings accordingly and flashing. Then changing things back for the actual leveling.

I don't get you.

Tannoo, 2017-06-30:
Lol

thinkyhead, 2017-07-02:
I believe the issue in the OP is now patched in `bugfix-1.1.x`.

boelle, 2019-02-19:
@Nickduino do you still have the problem with latest bugfix 2.0 ?

Nickduino, 2019-02-20:
Wow, I haven't tried that in quite a while (I've since created a "leveling gcode")

closed 2019-02-21 by thinkyhead

disappear123df, 2019-10-24:
how did you guys solve it

Youri-anet-a8, 2019-12-22:
idk