MarlinFirmware/Marlin #7177: Pause Printing Issue

devekrehbiel, 2017-06-28:
Error: KILL caused by too much inactive time - current command: M105 

I get this about 15-30 seconds after pausing a print. Not enough time to change the filament! I am using OctoPrint to pause so maybe Marlin is not causing it. Just hoping someone here has an answer. My pause code does not spool down the steppers or heat.

Bob-the-Kuhn, 2017-06-28:
Sounds like you've done some custom coding.  

To tell Marlin that the code hasn't gone off into the weeds, add` previous_cmd_ms = ms; // refresh_cmd_timeout()` to your code.  You'll need to update this periodically.

devekrehbiel, 2017-06-28:
Im learning.. so add that anywhere in the configuration.h file? When I do that I get
exit status 1
'previous_cmd_ms' does not name a type
The only code I added to Octoprint since I am using the Pi for the sensor is:
M117 Print Paused
G91
G1 Z10 E-10 F1000
G90

Which is insufficient because I want the heat turned off and then on again for resume, but havent learned that much yet. Thank You Bob!

devekrehbiel, 2017-06-28:
Just after I hit Pause, less than 30 seconds later I get:  KILLED, Printer Halted, Please Reset. The extruder and bed heat are still on until the KILL. Im afraid to wander into other config files on my own just yet.

Bob-the-Kuhn, 2017-06-29:
Sorry - I thought you had added code to Marlin.

Please copy the log from Octoprint, put it into a .TXT and attach it your reply.

I'm hoping there is a clue in the log.

fiveangle, 2017-06-29:
Do you have HOST_KEEPALIVE_FEATURE enabled ?

devekrehbiel, 2017-06-29:
I wish I was that talented Bob. I do NOT have Keep Alive enabled because something doesnt like it. I get the bed heating, then when it goes to E1 heating, it goes back to idle. So no. BUT if anyone else runs across this problem, I have a solution. Hopefully this will help you Marlin developers too. If you put the code G85 S0 in the Pause code, then G85 S30 in the Resume code, everything works great. Thanks to all of you for responding.

Don't forget to fix the problem of Bed Leveling not being in the LCD Prepare Menu just below Home. I use Bi-Linear Leveling so its dead to me right now. Thanks again!

benlye, 2017-06-29:
> Don't forget to fix the problem of Bed Leveling not being in the LCD Prepare Menu just below Home. I use Bi-Linear Leveling so its dead to me right now. Thanks again!

If you're desperate to have bed levelling in the LCD menu, enable user menus and add an entry for G29.

In configuration_adv.h:
```
#define CUSTOM_USER_MENUS
#if ENABLED(CUSTOM_USER_MENUS)
  #define USER_SCRIPT_DONE "M117 User Script Done"

  #define USER_DESC_1 "Level Bed"
  #define USER_GCODE_1 "G29"
...
```

devekrehbiel, 2017-06-30:
I did that exactly as is written above. I DO get a Custom Commands menu, but when I choose Level Bed, nothing happens.  It says user script done, but nothing happened. No worries though, sounds like someone is working on putting our menu selection back, so I can wait. Thanks to all of you!

thinkyhead, 2017-07-02:
> nothing happens

Maybe this would've worked…?
```
  #define USER_GCODE_1 "G28\nG29"
```
… Anyway, continuing to patch up issues. The bed leveling menu item is already patched, but the question of the KILL is still up in the air.

> If you put the code `G85 S0` in the Pause code, then `G85 S30` in the Resume code, everything works great.

That's not a Marlin G-code. What does it do?

devekrehbiel, 2017-07-02:
I meant to say M85. Sorry. If you do not turn off the inactivity timer, it KILLs. I noticed that in my start code (that I didnt realize I had in Cura), it calls for that. Its why I was having the problem. I hear its a good idea to have that in the startup, but you should know its there so in Pause and Resume circumstances you can turn it off then back on. Nothing wrong in Marlin that I know of. Since I use OctoPrint, for the pause/resume I can turn M85 off and on easily from the Gscript settings. I am still working on proper pause and resume gcode.

Drayson80, 2017-07-16:
did also a test with some custom code and also at me nothing happens...
just palin G28 works, but it seems as more to enter, as more marlin or the arduino quits working...

my trials:

#define USER_DESC_1 "3-Point Leveling"
#define USER_GCODE_1 "G21\nG90\nM107\nM117 Start Leveling - Homing X/Y\nG28 X0 Y0\n\nM117 Start Extended Bed Leveling\nM117 Front Left\nG1 X42 Y15 F9000\nG28 Z0\nM0\nM117 Front Right\nG1 Z5.0 F180\nG1 X148 Y15 F9000\nG1 Z0.0 F180\nM0\nM117 Back Middle\nG1 Z5.0 F180\nG1 X95 Y180 F9000\nG1 Z0.0 F180\nM0\nG00 Z10.0 F180\nM117 Extended Front Left\nG1 X15 Y15 F9000\nG1 Z0.0 F180\nM0\nM117 Extended Front Right\nG1 Z5.0 F180\nG1 X175 Y15 F9000\nG1 Z0.0 F180\nM0\n\nM117 Extended Middle\nG1 Z5.0 F180\nG1 X95 Y97.5 F9000\nG1 Z0.0 F180\nM0\n\nM117 Homing X/Y\nG00 Z10.0 F180\nG28 X0 Y0 F9000\nM117 Leveling Done\nG00 Z20.0 F180"

  #define USER_DESC_5 "Home"
  #define USER_GCODE_5 "G28"

the user desc 5 worked, the other not...

closed 2018-02-06 by thinkyhead