MarlinFirmware/Marlin #7178: linear Bed leveling & _MIN_PROBE_REPEATABILITY_TEST too large

naseweiss01, 2017-06-28:
Hello,
I am using an Anet 1.0 board with Skynet 2.4.5 (Marlin 1.1.3).
When trying to flash the firmware with aoutbed leveling & _MIN_PROBE_REPEATABILITY_TEST defined, the sketch was 99% of the available space & overwrote the boot loader. 
I am back in life with an AVR ISP but the issues seem to be the too large firmware.
I found other people in the net, which had the same problem with the repeatability test enabled.

Thanks,

Naseweiss01

benlye, 2017-06-28:
@naseweiss01, you need to use the latest version of the Anet board definition for the Arduino IDE, which you can get here:
https://github.com/SkyNet3D/anet-board

It has been updated to prevent the bootloader being overwritten by too-large flashes.  I'd recommend you flash the Optiboot bootloader to give you a little more space as well.

The Atmega1284-based Anet board doesn't have enough flash space for many of Marlin's more advanced features.  You need to pick and choose what to compile.

naseweiss01, 2017-07-01:
Hello,
indeed the new Anet board definition warns when the firmware is too large.
V1.1.3 is anyway to large even with optiboot for repeatablility enabled.
However I am now on the bugfix branch - there its small enough.

fiveangle, 2017-07-08:
@naseweiss01 - close ?

Roxy-3D, 2017-07-08:
Also...   This isn't how I want to operate...  But the M48 Z-Repeatability test tends to be pretty constant.   You can turn it on, and turn off other things like the LCD Panel just to get a number and know what you are dealing with.   And then turn the M48 off and turn back on the LCD Panel.

closed 2018-02-09 by thinkyhead