MarlinFirmware/Marlin #7180: Merlin No Z Travel

Justinjhcs1, 2017-06-28:
I have done all kinds of setting tricks but Z is still stuck... when I reload Sprinter again It's back.
settings are not wrong It just will not Travel it just sits there and HUMMS....

pmbroth, 2017-06-28:
I noticed with the latest build on my Marlin Gen 1.4 board, with the mkstsft 32, If I select Z to move up or down will not work,  Will work with the regular reprap 64128 controller.  Rolling back to 2 days ago build works fine

Justinjhcs1, 2017-06-29:
probably a Hardware Incompatibility I bought this Controller in 2011 Arduino Mega/RAMPS-1.4 and drivers for the Steppers... Zero problems with sprinter but nothing but headache with Merlin.... Merlin RC was better but Z still sat there and Hummed I though it was stuck... I am assuming that the drives are not getting enough current right? or the Wrong Pulse Current from the Controller.

pmbroth, 2017-06-29:
Mine is not hardware, when I roll back to 2 days ago bugfix11x build all is fine.  Something changed.

Andrei-Pozolotin, 2017-06-29:
https://github.com/MarlinFirmware/Marlin/issues/7166

Tannoo, 2017-06-29:
#7183 fixed it.

closed 2018-02-09 by thinkyhead