MarlinFirmware/Marlin #7191: Question about DELTA_CALIBRATION_RADIUS

JanVelleman, 2017-06-30:
Hi,

I have a quick question about DELTA_CALIBRATION_RADIUS, this is defined as follows:

#define DELTA_CALIBRATION_RADIUS ((DELTA_PRINTABLE_RADIUS) * 0.869) // mm

Thu 0.869 value is this somehow "fixed" or can I change this value? Why is it standard 0.869, is there a special reason for this??

LVD-AC, 2017-07-03:
You can set it to any value you like with max DELTA_PRINTABLE_RADIUS) * 0.869; this value is derived from the fact G33 P7 (maximum points) probes at DELTA_CALIBRATION_RADIUS * 0.85 *0.95 *1.05 and *1.15. You can set it even to higher values up to DELTA_PRINTABLE_RADIUS but then the higher density grids (P5, P6 and P7) won't run.

LVD-AC, 2017-07-03:
Close issue?

closed 2017-07-03 by JanVelleman