MarlinFirmware/Marlin #7192: Thinky has been quiet...

Roxy-3D, 2017-07-01:
It is possible we have a new folder and file layout Sunday evening!...

My wish list is:
- Not just an LPC1768 HAL
- I want some (any) 20X4 LCD display to compile clean with the Re-ARM board.

Add your wish list item here!

fiveangle, 2017-07-01:
- super-efficient realtime PROGMEM compression so 128k boards can run current marlin features ;) bwahahahahhaa

Tannoo, 2017-07-01:
All of Marlin working on the Re-Arm.

Tannoo, 2017-07-01:
Easier form-based configuration of Marlin (mainly for new users).

TGMods, 2017-07-01:
> All of Marlin working on the Re-Arm.

Especially the LCDs particularly the miniVIKI as I bought on especially for for Re-Arm

Tannoo, 2017-07-01:
A new `C1 Sx` for remote coffee machine control. ;)

p3p, 2017-07-01:
@Roxy-3D The only reason the 20x4 displays don't work (I think) is I had to strip out the Arduino Print API from U8glib, (it was the quickest solution to make graphical displays work), It shouldn't be very hard to add back into https://github.com/p3p/u8glib_arm_platformio when I get round to it. I've been away for a bit but I should have time for more work on the LPC1768 HAL next week.

thinkyhead, 2017-07-02:
I'm on the case at the moment, just scanning issues to see if anything serious needs a patch ahead of doing a quick 1.1.4 release. Then, diving into the current HAL work to make sure I understand it well. The big code re-organization is tomorrow's big half-day task, which I'll probably get a start on tonight.

Roxy-3D, 2017-07-02:
> I want some (any) 20X4 LCD display to compile clean with the Re-ARM board.

> The only reason the 20x4 displays don't work (I think) is I had to strip out the Arduino Print API from U8glib,

It is OK if the first version of the LPC1768 HAL doesn't support everything.  It already supports the Graphical displays.  It would be good if it supported at least some of the 20x4 displays.

But I have other agendas.   It turns out the place I can put my Re-ARM board has a 20x4 display.  So...  I at least want that to compile.  I can do the debugging if it needs it.  But it would be nice if the 20x4 LCD Displays at least compiled.

If I can get that display to run...  I can start adding to and fixing LPC1768 HAL issues.

closed 2017-07-26 by thinkyhead