MarlinFirmware/Marlin #7193: Support: BILINEAR with PROBE_MANUAL "ABL Adjustment" changes ?

fiveangle, 2017-07-01:
The [BILINEAR bed leveling docs](http://marlinfw.org/docs/features/auto_bed_leveling.html) appear to be sorely outdated (I will try to fix if I can understand how it works eventually):

> Currently, Marlin doesn’t save the bed leveling results to EEPROM, and it throws away the probe results on G28. So the machine must always do a new G29 after any G28

I've been told this is not true, and observations in terminal corroborate.

[G29 gcode docs](http://marlinfw.org/docs/gcode/G029-abl.html) appear more up-to-date but are incomplete with respect the various options.

In this discussion I am referring to `AUTO_BED_LEVELING_BILINEAR` with `PROBE_MANUALLY`.

Following the steps G28, G29 (with manual check of 5x5 grid) results come out to ("`<--------`" denoting my questions/comments):

<details><summary>Terminal Output</summary>

```
Recv: Grid probing done.
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.772 -0.810 -0.833 -0.843 -0.849
Recv:  1 -0.904 -0.929 -0.970 -0.959 -0.956
Recv:  2 -0.908 -0.929 -0.925 -0.937 -0.968
Recv:  3 -0.876 -0.840 -0.874 -0.889 -0.955
Recv:  4 -0.890 -0.882 -0.899 -0.900 -0.945
Recv: 
Recv: X:152.00 Y:152.00 Z:0.00 E:0.00 Count X:3040 Y:3040 Z:-479
Recv: ok P15 B3
[...]
Send: M500 <--------assumed this saved grid info to EEPROM
Recv: echo:Settings Stored (529 bytes; crc 21334)
Recv: ok P15 B3
[...]
Send: G29 Q <------queried leveling state
Recv:   current_position=(152.00, 152.00, 0.00) : >>> gcode_G29
Recv: Machine Type: Cartesian
Recv: Probe: PROBE_MANUALLY
Recv: Auto Bed Leveling: BILINEAR (enabled) <----shows enabled
Recv: ABL Adjustment Z-0.95 <------------------shows -0.95mm Z adjustment
Recv: Manual G29 idle
Recv: ok P15 B3
[...]
Send: M420 S1
Recv: echo:Bed Leveling On
Recv: echo:Fade Height Off
Recv: ok P15 B3
[...]
Send: G29 T   <------------assumed would output topology report
Recv: ok P15 B3
[bot moved to grid position #1 and awaited further instruction - this was unexpected.  Only documentation for G29 T command shows:

[T<flag>] - Generate a Bed Topology Report (AUTO_BED_LEVELING_LINEAR)

But gives no indication what <flag> denotes.  As usage standards aren't used such as "[ ]" to denote optional arguments, it is unclear what is required here.

I suspect the parser ignored the "T" without the <flag> argument and interpretted this as "G29" so began reprobe of grid.

[...]
Send: G28  <---------send autohome in order to exit grid re-probe process
Recv: echo:busy: processing
Recv: X:76.00 Y:76.00 Z:0.00 E:0.00 Count X:1520 Y:1520 Z:0
Recv: ok P15 B3
Send: G29 Q  <----------------queried state again
Recv:   current_position=(76.00, 76.00, 0.00) : >>> gcode_G29
Recv: Machine Type: Cartesian
Recv: Probe: PROBE_MANUALLY
Recv: Auto Bed Leveling: BILINEAR (disabled)  <------ disabled - I assume G28 disables bed leveling automatically ?
Recv: 
Recv: Manual G29 idle
Recv: ok P15 B3
[...]
Send: M420 S1   <----------------enabled bed leveling again
Recv: echo:Bed Leveling On
Recv: echo:Fade Height Off
Recv: ok P15 B3
[...]
Send: G29 Q
Recv:   current_position=(76.00, 76.00, 0.00) : >>> gcode_G29
Recv: Machine Type: Cartesian
Recv: Probe: PROBE_MANUALLY
Recv: Auto Bed Leveling: BILINEAR (enabled)
Recv: ABL Adjustment Z-0.93  <-----------ABL Adjustment is now changed to +0.02mm from previous (???)
Recv: Manual G29 idle
Recv: ok P15 B3
[...]
```
</details>

...
Could someone help fill in the blanks here ?  I'll try to fix the docs when/if I get to the bottom of this (if this is WAD)

Thanks !

-=dave

fiveangle, 2017-07-01:
Looking through the code for what seems like forever, it appears there's no way to report the stored probe values so I'm giving up on this and going back to MBL, since without the ability to see the stored values, the feature is pretty useless as if there is any doubt in values (like now, my printing is about 1mm above the bed), you need to reperform a tedious manual probe of every point.

closed 2017-07-01 by fiveangle

fiveangle, 2017-07-02:
BTW- everywhere I search on the big I'net for info on this, all I find is confusion, and @Roxy-3D suggesting everyone try UBL and "don't look back" - lol

thinkyhead, 2017-07-02:
This is slightly more up to date.

http://marlinfw.org/docs/gcode/G029-abl.html

thinkyhead, 2017-07-02:
> Looking through the code for what seems like forever, it appears there's no way to report the stored probe values with BILINEAR + PROBE_MANUAL

`M420 V` should print them. See http://marlinfw.org/docs/gcode/M420.html

fiveangle, 2017-07-02:
Yeah, tried that one too, but the Bot reacts as if I sent it G29 without any parameters:

```
Send: G29 V
Recv:   current_position=(38.00, 0.00, 0.00) : >>> gcode_G29
Recv: Machine Type: Cartesian
Recv: Probe: PROBE_MANUALLY
Recv: Auto Bed Leveling: BILINEAR (disabled)
Recv: 
Recv: Save X1 Y0 Z0.00
Recv: ok P15 B3
[...]
Send: G29 V
Recv:   current_position=(76.00, 0.00, 0.00) : >>> gcode_G29
Recv: Machine Type: Cartesian
Recv: Probe: PROBE_MANUALLY
Recv: Auto Bed Leveling: BILINEAR (disabled)
Recv: 
Recv: Save X2 Y0 Z0.00
Recv: ok P15 B3
```

I suspected it may be some FASTER_GCODE_PARSER weirdness, but then when trying with the old parser, it wouldn't compile:

```
In file included from /var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_54543/sketch/Marlin_main.cpp:259:0:
/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_54543/sketch/gcode.h: In static member function 'static volatile bool GCodeParser::seen(char)':
gcode.h:155: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
       if (b) value_ptr = DECIMAL_SIGNED(p[1]) ? &p[1] : (char*)NULL;
                        ^
Using library U8glib at version 1.19.1 in folder: /Users/speedster/Documents/Arduino/libraries/U8glib 
Using library Wire at version 1.0 in folder: /Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/avr/libraries/Wire 
invalid conversion from 'const char*' to 'char*' [-fpermissive]
```

So I decided I was probably too far off the beaten path and that I should get back home for dinner before sundown ;)

thinkyhead, 2017-07-02:
If `G29 V` doesn't report the leveling grid with bilinear, it means the leveling data is bogus and you do need to re-level to generate valid data.

thinkyhead, 2017-07-02:
> I don't want to be known as "that guy who always finds the broken tedious little things" :)

Honestly, we need exactly what you're doing. Muchas gracias!

fiveangle, 2017-07-02:
> If G29 V doesn't report the leveling grid with bilinear, it means the leveling data is bogus and you do need to re-level to generate valid data.

It's odd then why G29 S1 succeeds.  If G29 V fails, then G29 S1 should fail as well, no ?

thinkyhead, 2017-07-03:
For ABL with Bilinear, read the documentation at http://marlinfw.org/docs/gcode/G029-abl.html — just updated with some additional notes.

When using ABL Bilinear with `PROBE_MANUALLY` you just use `G29` repeatedly — there's no `S` parameter. (MBL does have an `S` parameter.)

fiveangle, 2017-07-03:
Big help Scott.  I'm sure this will help a lot of people down the road.  I'll try to improve if I find opportunities.  Thanks !

fiveangle, 2017-07-04:
M420 V doesn't print grid, but is otherwise ignored. V4 does print more info as leveling
Progresses. But no way to validate existing mesh or even view it. The reason this is critical is that one has no way to prevent this from happening (identical GCODE as printed the night before):

[image]

if a relatively technical user can't get BILINEAR working consistently without destroying $70 beds and $20 ultem sheets,  I call it broken :/