MarlinFirmware/Marlin #7194: Endstop Triggered but not working.

NovemberCo, 2017-07-01:
Hi, i have this issue where the X endstop wont work if the Y endstop its triggered first. The printer was working fine until i shut it off... and when i turned it back on, this error apeared.  If i move the carriage in the Y direction and release the Endstop then the X endstop works fine. Its like the X endstop has to be triggered first than the Y endstop or it wont work... If anyone has any idea i would apreciate the help.

Thanks!

Tannoo, 2017-07-01:
Power cycle it again see if it persists.

NovemberCo, 2017-07-01:
Hi... already turn it on and off like 6 times.. and its happening the same.

Tannoo, 2017-07-01:
Have you tried re-flashing Marlin?

NovemberCo, 2017-07-01:
I have re-flashed Marlin like 2 weeks ago... tomorrow im going to check that thanks alot

Bob-the-Kuhn, 2017-07-01:
I'm wondering if maybe the X & Y endstops are shorted together.

Using M119 - are X & Y always the same?

thinkyhead, 2017-07-02:
If you have any extra endstop plugs, try connecting the endstops to different endstop plugs (changing the pin numbers appropriately), re-flash, and see if it changes the behavior.

NovemberCo, 2017-07-02:
Hi sorry i didnt answer... im using the printer as it is right now so didnt have time to check... tomorrow im going to put hands on the thing and see if anything has changed.. Thanks Alot for the help!

NovemberCo, 2017-07-03:
Hi, i tested the endstop, Using the m119 with the X and Y endstop triggered (looking at the led in the endstop) this is the result,
[image]

The X endstop appears Open when its triggered in the LED

Tannoo, 2017-07-03:
Try swapping the X and Y enstop connections and issue M119 again. Looking to see if it's a bad endstop or connection.

NovemberCo, 2017-07-03:
The X endstop wires are bad :/ thanks alot for the help.. and sorry that i bring this issue that its not relative to the firmware.

Bob-the-Kuhn, 2017-07-03:
No need for sorry.  

We enjoy solving problems.  Doesn't matter if the cause is hardware or firmware or documentation.  

Hardware and firmware are so tightly intertwined that you have to be knowledgeable in both in order to troubleshoot either.  Lots of the time you can't tell if it's hardware or firmware until you get close to the end.

Hope to hear from you again.

closed 2017-07-03 by NovemberCo