MarlinFirmware/Marlin #7195: HOMING ISSUE USING MARLIN 1.02 WITH PRUSSA 3D PRINTER

mandeep255, 2017-07-01:
I have an issue with Z Homing using inductive sensor. The issue cropped up only when I upgraded to MARLIN 1.02.
The Z homes above the bed more than required, The inductive sensor is triggering perfectly, working perfectly with previous version. It now homes but rises more than required. PL help!

Bob-the-Kuhn, 2017-07-01:
I'm not sure what you mean by "rises more than required".  

Is it: "In the old software, after G28, G0 Z0 resulted in the nozzle being x mm above the bed but with the new software, after G28, G0 Z0 results in the nozzle being x + y mm above the bed (where y is a positive number)"

Most inductive sensors use a high impedance resistor network to interface to the controller.  The internal pullup can cause problems if it's enabled.  The following will disable the internal pullup on the inductive sensor.  Note that both ZMIN & ZMIN_PROBE have to be disabled.
```
#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif
```

mandeep255, 2017-07-03:
Thanks Bob, 
The sequence of Homing is under
nozzle moves to center of the bed
Extruder moves down till proximity sensor is triggered
The extruder rises up to a certain height , approx 10mm.
The Z probe  axis reads 0 on the Repetier Host
While printing the nozzle prints in the air, ie above  the surface.
Endstoppullup had been disabled since I was using a NPN sensor.

Autolevelling is on.

Thanks and regards

Bob-the-Kuhn, 2017-07-04:
Please do a` M420 V`, copy the result to a .TXT file and then drop the .TXT file onto your reply.

I expect that you need to adjust your Z_PROBE_OFFSET_FROM_EXTRUDER.  You can use the `M851 Zxxx` to set this offset.  You may need to repeat G29 for the new value to take effect.

Don't forget update the configuration.h file and to do a `M500` command to save the new offset value to EEPROM.

mandeep255, 2017-07-05:
Thanks Bob, will revert back.

mandeep255, 2017-07-06:
[420Vresults.txt](https://github.com/MarlinFirmware/Marlin/files/1128046/420Vresults.txt)
The file is uploaded, I performed a G28 then a G29 followed by M420V
regards

mandeep255, 2017-07-06:
[PrinterVideo.zip](https://github.com/MarlinFirmware/Marlin/files/1128947/PrinterVideo.zip)
Pl also see the att videos, one shows the homing sequence of my printer and second for auto bed levelling, I think the sequence is also incorrect.  regards

mandeep255, 2017-07-06:
Firmware now used Marlin 1.1.4

mandeep255, 2017-07-07:
[configH.txt](https://github.com/MarlinFirmware/Marlin/files/1129703/configH.txt)
copy of config.h 
thanks

closed 2018-02-09 by thinkyhead