MarlinFirmware/Marlin #7198: Homing issue big bug ! In 1.1.3 and not in RC8

studiodyne, 2017-07-01:
Hi 
When homing with 1.1.3 ,   and  boundaries setted like ' XMIN=30 ,  when the carriage is on the xmin switch , then firmware is on XMIN , on the lcd we can see ' 30 ' and we are on the switch in the zero position , i have made my printer on december with RC8 , today i decide to finish with the last stable marlin , and i make all parameters , BIG BUG OF HOMING with software limits

RC8 WORKS FINE

I have tried three times , with three new clean 1.1.3 , to set the minimum parameters to be sure it's me or the software

A fix  ?

fiveangle, 2017-07-01:
please try RCBugFix1.1.1 to see if a fix already exists

studiodyne, 2017-07-01:
HOMING ISSUE CONFIRMED

The software endstops are unfunctionnals  

Return to RC8

Very sad ... i loved the new axis move menu

Thks

Bob-the-Kuhn, 2017-07-01:
Did you try bugfix-1.1.x (https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x)?

It has some bug fixes that are not in 1.1.3

Did you transfer all the machine specific items to the new config files?  Using the one from RC8 can cause some strange problems if used (without modifications) with 1.1.3 or bugfix-1.1.x.

Can you describe the failure mode in more detail?  A copy of the log with your notes added would be very helpful in understanding the issue.

You could also ZIP up configuration.h & configuration_adv.h and attach them to the reply.

studiodyne, 2017-07-01:
I have downloader , the 1.1.x version zip and make a new clean firm
I have entered with my keyboard all parameters ' some basics ' directions of motors , bltouch , filament runout , and my software bed limits

When i make the homing ,  the carriage is on the switch ' x ' and it's the zero hardware position , but , the firmware show ' 11 ' , 11 is my xmin limit
I have switched off the safe homing , i have switched all is possible but nothing happened
I'm working on RC8 since december 2016
Now i have finished my 400x400x400 printer , and i want to have the last software , but , i'm very sad  lolll 
RC8 is stable , perfect !!! But , i love the new menus and options

Bob-the-Kuhn, 2017-07-01:
The configuration.h file doesn't give the whole story on the min/max positions.  Here's my notes:
```
#define X_MIN_POS -50  // if homing in the - direction, G28 sets X position to this value after homing
#define Y_MIN_POS -70  // if homing in the - direction, G28 sets Y position to this value after homing
#define Z_MIN_POS 0    // if homing in the - direction, G28 sets Z position to this value after homing
#define X_MAX_POS 330  // if homing in the + direction, G28 sets X position to this value after homing
#define Y_MAX_POS 280  // if homing in the + direction, G28 sets Y position to this value after homing
#define Z_MAX_POS 430  // if homing in the + direction, G28 sets Z position to this value after homing
```

studiodyne, 2017-07-01:
okkkkkkkkkkkkkkkkkkkkkkkkkkkkkk

studiodyne, 2017-07-01:
BOB THE DIABOLO !

studiodyne, 2017-07-01:
all my homing directions are negatives

studiodyne, 2017-07-01:
Please make a PR , to have a clean code , that have the same arithmetic psychology of the parent code lol

studiodyne, 2017-07-01:
Explaination 
Cartesian printer , 
x0 is on the left , and switch on the left
Y0 is on the front , and the switch is on the top
Settings of directions homing are   x --> -1 and y --> -1
If i want a xmin software limit , i have to set ' xmin_pos=-11  ..... ???
I have a print running , i will try later

thkyou

Bob-the-Kuhn, 2017-07-01:
I'm assuming that X_MIN_POS is set to -11 because the nozzle needs to move 11 mm to the right in order to be at the bed's beginning position.

If that case MIN_SOFTWARE_ENDSTOPS won't keep the nozzle within the bed.

studiodyne, 2017-07-01:
It's ok , but not the same arithmetic than the old firm
Now , the gcode file see a zero , the lcd see a zero , it's more clean 
The old version show 11 , but the printing make it for zero
Now ilt's more clear , the lcd and the printing have the same arithmetic

studiodyne, 2017-07-01:
I ve tried it now , it's perfect , i prefer like this
But write a PR , to have the coments , if not , you will flood this forum just for this

I'm happy , i don't want to sleep , i make my new release

THK YOU VERY MUCH

studiodyne, 2017-07-01:
but by trying it now , it's a big problem , now , i don't have the hardware protection of the machines , to avoid to explode the clamp on the bed , and i have no solution to make a hardware stop like before on a defined area
If i have some clamp on x5 , i don't want the carriage go on x5 i just want it starts from x11

And now , we have no protection for the carriage ....

studiodyne, 2017-07-01:
The old psychology of the endstop given us , a hardware true limit with no possibility to cross it
a lot of people have some clamps on the bed , ultimaker have it too 

What is your solution to have hardware area limits with this new release ?

studiodyne, 2017-07-01:
It's ok , i have worked on this release , i have some suggestions for you
1-) If we put a negative x min its ' of course ' to place a ' PERFECT X ZERO LINE' , so if we just need the area from 'zero' ,  the homing have to make this
G1 X0 or G1Y0 or G1X0Y0 just after the probing , and without go down with z axis
Because , we have to fly over the clamps on the bed that are in the negative zone 
2-) The lcd move axis must have ' zero ' as limit 'min' , just to avoid to go in the negative zone
3-) the max software endstop works fine , no need to modify

You have to make the new ' SAFE HOMING' , that have a gcode script just after , to positionate the carriage on the positive area
And now it's a sure technology 

If all is respected in the slicer , the gcode will never go in negative zone

studiodyne, 2017-07-01:
For my personnal use , i have just set the min ' lcd moves x and y ' to zero , and it's perfect , the lcd cannot' move in negative zone , the slicer have only positive values so , for me , all is ok

Bob-the-Kuhn, 2017-07-01:
Glad to hear you've worked out a useable system for your application.

----

Right now there is no method of keeping the nozzle within the printable area if a bad command is issued.

I took a look around the code and found out that there are multiple functions that use the travel limit.  Creating/modifying code to keep the nozzle safely within the print area is a non-trivial task.

There are cases where you want the nozzle to be able to travel outside the printable area.  G28 and G29 fall into this class.

I'll open an issue just for this discussion.

studiodyne, 2017-07-01:
Here is a listing , with  new release problem versus old release system

1-) Actual release : Impossible to have a hardware limiter for the machine in the negative zone 
     Old release : Limiter works , just the start of the bed not named ' zero' , and have the xminpos as value
Very dangerous for all machine that need to define a safe area ' clamp on the bed for example' 

2-) Impossible to make a good probing ... If i define the xmax boundarie , on a carriage where we have one extruder + bltouch on the center + second extruder on the right . Why , because of the restricted xmaxpos software endstop , the firmware have defined a strict area , and when my FIRST EXTRUDER is to the max , my bltouch is 17mm to the right , the firmware not allow me to probe on the XMAXPOS+ xoffset bltouch
Result , i must use ' xmaspos ' for the probing max zone , and , i loose17mm of probing space
on a bltouch placed on the right after 2 extruders , the probing can loose 70mm

3-) The comments of the boudaries say ' boundarie limit ' , yes it's true , but only for one extruder , i we are in 2 extruders + bltouch configuration , we loose space

The old sytem can place the probe on xmax + offset ,  the new release is strict and can not made this

This code is not clear , two many separates laws , not made to help

Solution . 
Keep this psychology , to have the real zero on the real zero of the bed
Allow the probe to cross the limits to adjust the area of probing
Give a true functionnal hardware restriction to the ENDSTOPSOFTWARELIMITER to make the ' safe area 
Keep z axis up !  a defined height when in the negative zone to avoid collisions with the bed and the clamps

studiodyne, 2017-07-01:
Hardware safe area is the most important thing we need
How many stupid error we make everytime , on the slicer or anywhere 
The machine must have this
for the moment i return to RC8 
I FOLLOW the evolution

studiodyne, 2017-07-01:
to have the perfect use for me 

The liberty of the probe  : Conditional post.h  line after 800 ,  min to max 'conversion'  lol
For lcd move restriction : ultralcd cpp , line 2391   .. Min = 0   lol 
and LIFE IS BEAUTIFULL

thinkyhead, 2017-07-02:
> I'm assuming that X_MIN_POS is set to -11 because the nozzle needs to move 11 mm to the right in order to be at the bed's beginning position.

That is the correct way to do it.

> [In] that case MIN_SOFTWARE_ENDSTOPS won't keep the nozzle within the bed.

The software endstops are designed to prevent movement outside the physical limits, not the bed, so you're all good.

studiodyne, 2017-07-02:
@thinkyhead    

Yes , hours after hours i understand the true functionnality of the  MIN/MAX_SOFTWARE_ENDSTOPS and i understand now , one of this have no utility in some configurations

I realize the true problem is the way we understand this functionnality 7
May be , some comments to avoid many people here with the exactly same problem 
Like this ::
Be carreful , MIN/MAX_SOFTWARE_ENDSTOPS not made for make a bed safe area , but just to make a movement limitation in the homing blind direction that have no hardware switch or probe. And you must understand that only one software limitation by axis is effective , the other direction is limited by the endstop switch ' of course '

And now , All people have spirit illuminations loll

'Don't forget us , we don't have your brains '  👍

studiodyne, 2017-07-02:
Thk you very much to @Bob-the-Kuhn  , that have given me the solution to keep alive my 1.1.3 and use it now !

studiodyne, 2017-07-02:
@Bob-the-Kuhn  
Can you make us for this release 
4 lines of bed boundaries ,xmin , xmax , ymin , ymax
When homed , impossible to go outside of these , lcd or gcode or nothing , only the bltouch have the right of crossing

If you can 👍

Bob-the-Kuhn, 2017-07-02:
I can't promise this idea will be accepted.

I will start a discussion on it.

It'll be at least a week before it's implemented.  Usually in the 2-4 week time frame.

I'll update this thread as progress happens.

studiodyne, 2017-07-04:
hI , i'm trying this new firm , I LOVE IT
baby step , linear pressure , and firm retract , all the menus are now more easy and more and more

How , to ?

I love the firmware retract ok for it , but i want to know how to call the firmware tool change retract , i use the old way  but , with my ' Y ' dual extruder , the recovery retract must be adjusted and i need to have the possibility to set it during a print

Can you tell me how to  .... G10 and G11 ok ...' 3mm ' for me , but if i want to make a tool change retractation i need 60mm retract and recovery ,  how to call the firmware swap tool recovery by gcode please

Thks

Bob-the-Kuhn, 2017-07-04:
Sorry - firmware retract, G10 & G11 is something I've never worked with.

Best thing to do would be to open up a new issue specifically for that.  That'll get the people with more printing experience looking at it.

studiodyne, 2017-07-04:
I ask you to try this amazing thing
Replace all your retract gcode by g10 and g11 and see what happens
We set our retract parameters without a true idea of what will happen
With this g0 g11 you can change retract in real time and test all setting you want to find what is perfect for you
I will search in marlin.cpp , what are the parameters of g10 g11 to see if i can have informations

Thks you

studiodyne, 2017-07-04:
g10 s1 to have tool change big retract , g11 to recover ' same as g10 ' because recover is just recovering to zero

Yeah , i try it tomorow

thinkyhead, 2017-07-04:
With both `MIN_SOFTWARE_ENDSTOPS` and `MAX_SOFTWARE_ENDSTOPS` all `G0` and `G1` movement is constrained to the values set for `*_MIN_POS` and `*_MAX_POS`. You will find that with these options set you cannot initiate a bad movement with these commands, or using your host's movement buttons.

That constraint is performed at a high level as part of parsing the G-code coordinates. It is simpler and easier to constrain movement there, because at that point it's always a single linear movement.

In order to constrain movement for every type of move, we would need to do so at the planner level, and here we fall into a dangerous area, where the lower-level position stored in terms of stepper steps loses synchronization with the high level `current_position` variable that allows us to know where the nozzle is positioned.

While in theory it would be a cool thing to add a constraint on movement at the lowest level, in practice this opens up various complex difficulties and increases computational load. So we have only opted to concatenate movements by `G0` and `G1` so that we can continue to know the position of the nozzle without needing to re-poll the planner and stepper classes.

thinkyhead, 2017-07-04:
When testing your software endstops, use `M211`…
```cpp
/**
 * M211: Enable, Disable, and/or Report software endstops
 *
 * Usage: M211 S1 to enable, M211 S0 to disable, M211 alone for report
 */
```

studiodyne, 2017-07-05:
@thinkyhead 
Thks for the explanations 👍 
But physical edges and bed edges are two separates data .
When we set the MIN_SOFTWARE_ENDSTOPS , on the same side of the endstop switch , but with a bed that not start on the ' ZERO , the position of the endstop switch' , all is different
The manual lcd go on this negative zone , because LCD MANUAL MOVE don't stop at zero 
'i have fixed this ' 👍 to avoid the risk to be in negative zone
For a dual extruder in line , the MAX_SOFTWARE_ENDSTOPS , is for the max position of extruder 2 to have secure movement of all the carriage, but if we have BLTOUCH between extruder1 and extruder 2 , we dream of the probe can reach the extruder 2 coordonates for probing at the maxe
'I have fixed by changing the  ' max  choosen selection ' of the probe movement

I ve tried the slicer and no risk with it , when  we define the area in the setttings , the slice don't produce negative coordonates , it runs only between coordonates ,

studiodyne, 2017-07-05:
I have a question , serious question 
I use marlin , and i  see it works , it has a true real ' performance ' , it's a very big job
I want to make a donation but for who  ?
Who can accept a donation in a free communauty

Thks

studiodyne, 2017-07-06:
@Bob-the-Kuhn  @thinkyhead 

hI , here are my results , MARLIN works !!!! 
Tested : 
babystepping ! 
Linear advance ' k=15 for 0.6 nozzle , direct drive , 1.75mm and higH pressure extruder ' flexxion ' , this algorithm not very ' miraculous ' in the case of big nozzle , the pressure is always low but i love to use it , the print look is better than all ' simplify checked boxes and options '
G10 AND G11 + G10 S1 and G11 S1 : It works , but i have two issues to report 
1-) The feedrate of the ' swap ' recovery , is the same of the normal retract , in my case , i have 60mm of retract distance ' because of a single nozzle  ' Y ' extruder ' , and it's a big problem , the speed of the recovery is adjusted to have a perfect extrusion in the ' RESTART' , but , when swapping extruder ' tool change ' , i'm forced to use the very low feedrate to recover ' 60mm' of filament
I have fixed it by creating a constant ' FIRMWAIRE RETRACT_RECOVER_SWAP_FEEDRATE_MULTIPLIER ' , and i have added it on the 'recover_feedrate'  with a condition , if swapping i multiply the speed
Easyto code and it runs 

The second problem is a true problem , when you activate ' firmware retract' on the machine , we can not use the old gcode  with negative extrusion , the firmware replace all negative extrusion by the g10 ' retract '  parameters

I explain
If i want to initialize an extruder with my own gcode , i can use 
G1 E40   ' because positive '
If i want to push  G1 E-40 , the firmware replace it by a G10 , and g10 = G1 E-3 

for the moment it's not a problem because i only use G10/G11 and G10 S1 / G11 S1

Solution 
For the swap feedrate recover of a G11 S1 ,  make a new line for the lcd menus , add this value in the swap recovery retract engine

For the Negative extrusion replaced by G10 when FirmwareRetract is turned on , i trust you to repair this 

AND TO FINISH

I LOVE MARLIN , it's a beautifull job 
I want to thks all the worked of this project , it's a big big job !

Thks for all , i have my printer with MARLIN INSIDE !!!  
Marlin Inside !!! Yes !

studiodyne, 2017-07-09:
@Bob-the-Kuhn @thinkyhead
New results

The 'Firmware Retract' and the ' AUTORETRACT' are separates process and that explain the trigger phenomen that i said before

Please add this comments on the firmware retract to inform all the users

1- Firmware Retract AUTO is an automated retraction if ' RETRACT_MIN ' is reached by a GCODE with negative extrusion , for example if  G1 E -2 , it is superior of the ' Retract min =0.2 ' so , it will replaced by a G10 'firmware retract '  , in case of positive extrusion , no trigged process

2- G10 and G11 normal retract
3- G10 S1 and G11 S1 , are for tool change retract and recovery , the feedrates are the same of the normal

4- Be careful with Firmware retract with Z height movement , because some slicers can put a G92 Z0 sometimes and your Z axis will be false for the rest of the print ,  it's always on the initialisation and starting gcode script or when you write your new own script

Please , for all the users , add this comments , i have lost one week to understand this beautifull functionnality

thks

studiodyne, 2017-07-10:
I'm forced to make a new issue for g10 and g10
Too much problems ...

studiodyne, 2017-07-10:
I have worked all night long , to learn language , and make war with each line of code that never works ...
G10 and G11 are so bugged , or may be ' not finish ' , i can finish it , but i need to see the variables of the software , each time i write something , it's in blind mode , i write a line of code , i must upload the firmware and try it , it's too much
I want to put some codes that permit me to see the value of all the variables i work with , because , in blind mode , it's impossible to see what really happen and the flow of the values that i need to process in my code 

Can you give me a way to have an echo of all the variables i need  , if i can see all the values i can understand why i can't make a functionnal line of code

Thks

studiodyne, 2017-07-10:
I have found an idea to debugg 👍

Roxy-3D, 2017-07-10:
You can put SERIAL_ECHO() macros at points in the code to print out values.   That is how most of us debug difficult issues.

studiodyne, 2017-07-10:
@Bob-the-Kuhn @thinkyhead 

Here is the new ' DEBUGGED FIRMWARE RETRACT PROCESS '
Only for 2 extruders , but , the work is done , to make the code for 4 extruders is now more easy
25 hours of war !!  

#if ENABLED(FWRETRACT)

  void retract(const bool retracting, const bool swapping = false) {

    static float hop_height;
    // debugging
    /*SERIAL_ECHOLNPAIR("hop height ",hop_height);
    SERIAL_ECHOLNPAIR("retracting ",retracting);
    SERIAL_ECHOLNPAIR("swapping ",swapping);
    SERIAL_ECHOLNPAIR("active extruder ",active_extruder);
    SERIAL_ECHOLNPAIR("retracted 0 ",retracted[0]);
    SERIAL_ECHOLNPAIR("retracted 1 ",retracted[1]);
    SERIAL_ECHOLNPAIR("retracted_swap0 ",retracted_swap[0]);
    SERIAL_ECHOLNPAIR("retracted_swap1 ",retracted_swap[1]);
    SERIAL_ECHOLNPAIR("current_position z ",current_position[Z_AXIS]);*/
  
//can retract if not retracted and swap not active   
if(retracting == retracted[active_extruder]
&&(retracted_swap[0]==0)
&&(retracted_swap[1]==0))
return;

// cannot recover if not the active extruder or the active swap
if(!retracting
&&!retracted[active_extruder]
&&!retracted_swap[active_extruder]) 
return;

    const float old_feedrate_mm_s = feedrate_mm_s;
    

    set_destination_to_current();

    if (retracting) {

      feedrate_mm_s = retract_feedrate_mm_s;
      current_position[E_AXIS] += (swapping ? retract_length_swap : retract_length) / volumetric_multiplier[active_extruder];
      sync_plan_position_e();
      prepare_move_to_destination();

      if (retract_zlift > 0.01) {
        hop_height = current_position[Z_AXIS];
        // Pretend current position is lower
        current_position[Z_AXIS] -= retract_zlift;
        SYNC_PLAN_POSITION_KINEMATIC();
        // Raise up to the old current_position
        prepare_move_to_destination();
      }
    }
    else {

      // If the height hasn't been lowered, undo the Z hop
      if (retract_zlift > 0.01 && hop_height <= current_position[Z_AXIS]) {
        // Pretend current position is higher. Z will lower on the next move
        current_position[Z_AXIS] += retract_zlift;
        SYNC_PLAN_POSITION_KINEMATIC();
        // Lower Z
        prepare_move_to_destination();
      }
      
      //adding of a retract multipilier ' to add in your config.h ' 
      // To increase swapping recover by multiplying the recover feedrate
      feedrate_mm_s = swapping ? RETRACT_RECOVER_FEEDRATE_SWAP : retract_recover_feedrate_mm_s ;
      
      //feedrate_mm_s = retract_recover_feedrate_mm_s ; Original feedrate
      const float move_e = swapping ? retract_length_swap + retract_recover_length_swap : retract_length + retract_recover_length;
      current_position[E_AXIS] -= move_e / volumetric_multiplier[active_extruder];
      sync_plan_position_e();

      // Recover E
      prepare_move_to_destination();
    }

    feedrate_mm_s = old_feedrate_mm_s;
    
    retracted[active_extruder] = retracting;
   
       
// to set the same state ' swapping'  to all the extruders
// Because of Zlift ' very usefull in dual color ',
// all extruders must be synchronised on ONLY ONE ZLIFT
      if (swapping ) {          
      if (retracting){retracted_swap[0] = 1;retracted_swap[1] = 1; }
      else {retracted_swap[0] = 0;retracted_swap[1] = 0;  }
      }

     // debugging
    /*SERIAL_ECHOLNPAIR("-hop height ",hop_height);
    SERIAL_ECHOLNPAIR("-retracting ",retracting);
    SERIAL_ECHOLNPAIR("-swapping ",swapping);
    SERIAL_ECHOLNPAIR("-active extruder ",active_extruder);
    SERIAL_ECHOLNPAIR("-retracted 0 ",retracted[0]);
    SERIAL_ECHOLNPAIR("-retracted 1  ",retracted[1]);
    SERIAL_ECHOLNPAIR("-retracted_swap0 ",retracted_swap[0]);
    SERIAL_ECHOLNPAIR("-retracted_swap1 ",retracted_swap[1]);
    SERIAL_ECHOLNPAIR("-current_position z ",current_position[Z_AXIS]);*/

  } // retract()

#endif // FWRETRACT

thinkyhead, 2017-07-10:
> i must upload the firmware and try it , it's too much

Welcome to Marlin development!

thinkyhead, 2017-07-10:
Thanks for looking more closely. I will use your sample as a guide to making `FWRETRACT` work correctly on more types of setups.

studiodyne, 2017-07-10:
I have leaved comments , and debugging lines
The code is made to be easy to understand and all conditions is written like a child , all is made for the engineer who want to implement it in the future release without wasting hours to try and understand

Thks for Marlin , this firmware is a very BIG JOB !

thinkyhead, 2017-07-10:
So… to extend for more extruders, it looks like in the first `if` block it needs to check whether `retracted_swap` is set for **any** extruder, not just 0 and 1. Does that make sense?

thinkyhead, 2017-07-10:
> for the engineer who want to implement it in the future release without wasting hours to try and understand

Always appreciated. In my 40 years of coding, I find that very few coders write good comments.

studiodyne, 2017-07-10:
The rule for many extruders is simple 👍 
Only one zheight for all , 
all can normal recover  , you can be strict but , to avoir some risks of bad gcode that push or forget a g10
Only one can swap recover , because ' in the case of single nozzle , need to load and unload '

For more extruders ,  use ' do while ' , and loop the extruders[i] loll good luck
The 0 and 1 is for a better visibility of the process , if i made a very short compressed and compacted code , impossibe to gain time to implement it
All is made to be transmitted without wasted time
You can work quickly with this type of writting !

I have past 25 hours , one night without sleeping , just for somes lines ..... I don't want to imagine , the big job marlin is ........

studiodyne, 2017-07-11:
I have one good , and one bad news
The good is the rewritten new Firmware retract that i have made with more security in the case of a bad produced code by ' batch converting'

The bad new , is that , the print have an horrible bug not in the beginning but after some layers , i can see , over extrusion , the outline is larger  like an overextrusion ' the print is larger , '  the z height or overextrusio problems appears and print dye , the nozzle become lower of the last layer 
I'm sure it's a z calculation that dye ..... 
I have tried the same print with only gcode ' exactly the same movements' but no g10/g11  and it's works fine , and with linear advance

I don't know if it's my code that make a sad z height , or an overextrusion ..... but it happens after 20 or 30 layers , and not at the beginning

I don't know what is the problem , for the moment , i have printed some same objects , and when no g10/g11 , the print is sucessfull
I have visualized the produced gcode in a visualiser to verify , and all is ok , no overlaping
2 prints with g10g11 and same problems , but not at the same layer , the first on layer 20 , the second print on layer 30 
The sequenced gcode is the same of the 20 or 30 layers before , no new gcode or something else

I have tried to make a new secure code , to be sure , i don't make some problems 
I'm very sad

here my new code

...cpp
#if ENABLED(FWRETRACT)

  void retract(const bool retracting, const bool swapping = false) {

    static float hop_height;
    static bool zlift_activated=0;
    // debugging
    /*SERIAL_ECHOLNPAIR("hop height ",hop_height);
    SERIAL_ECHOLNPAIR("retracting ",retracting);
    SERIAL_ECHOLNPAIR("swapping ",swapping);
    SERIAL_ECHOLNPAIR("active extruder ",active_extruder);
    SERIAL_ECHOLNPAIR("retracted 0 ",retracted[0]);
    SERIAL_ECHOLNPAIR("retracted 1 ",retracted[1]);
    SERIAL_ECHOLNPAIR("retracted_swap0 ",retracted_swap[0]);
    SERIAL_ECHOLNPAIR("retracted_swap1 ",retracted_swap[1]);
    SERIAL_ECHOLNPAIR("current_position z ",current_position[Z_AXIS]);
    SERIAL_ECHOLNPAIR("ZLIFT  ",zlift_activated);*/
    
// cause the firmware make no swap status to all extruders
// need to allow the swap load and unload in singlenozzle
if (retracted_swap[0]||retracted_swap[1] ) {       
      retracted_swap[0] = 1;retracted_swap[1] = 1; }

  
//IF ALREADY RETRACTED and retract asked ---then return because not needed and if retract_zlift exist of course
if(retracting &&  zlift_activated ) 
return;
//IF ALREADY RETRACTEDDOWN and retract asked ---then return because not needed and if retract_zlift exist of course
if(!retracting &&  !zlift_activated && retract_zlift) 
return;

//can retract up if  retracted and swap  active  and reinitilise the status lost ' cause the firmware low the swap status 
    if(retracting &&!swapping )
    {
      if ((retracted_swap[0]==1)||(retracted_swap[1]==1))    
   {retracted[active_extruder]=0; retracted_swap[0] = 1;retracted_swap[1] = 1;
   return;}
   };
    
// can retract the other ' for bad slicer or batch'
    if(!retracting && !swapping)
    {
      if (!retracted[active_extruder])
      retracted[active_extruder]=1;retracted[!active_extruder]=0;
    }

//if swapped no else possibility to unswap because if a g10 appear , the retracting status will be high and the 
//load/unload of the swapping process is down for eternity lol
 // if( retracted_swap[0]||retracted_swap[1]&& !swapping)  return; 

[... 61 lines cut ...]
   
       
// to set the same state ' swapping'  to all the extruders
// Because of Zlift ' very usefull in dual color ',
// all extruders must be synchronised on ONLY ONE ZLIFT
      if (swapping ) {          
      if (retracting){retracted_swap[0] = 1;retracted_swap[1] = 1; }
      else {retracted_swap[0] = 0;retracted_swap[1] = 0;  }
      }
      

      //debugging
    /*SERIAL_ECHOLNPAIR("-hop height ",hop_height);
    SERIAL_ECHOLNPAIR("-retracting ",retracting);
    SERIAL_ECHOLNPAIR("-swapping ",swapping);
    SERIAL_ECHOLNPAIR("-active extruder ",active_extruder);
    SERIAL_ECHOLNPAIR("-retracted 0 ",retracted[0]);
    SERIAL_ECHOLNPAIR("-retracted 1  ",retracted[1]);
    SERIAL_ECHOLNPAIR("-retracted_swap0 ",retracted_swap[0]);
    SERIAL_ECHOLNPAIR("-retracted_swap1 ",retracted_swap[1]);
    SERIAL_ECHOLNPAIR("-current_position z ",current_position[Z_AXIS]);
    SERIAL_ECHOLNPAIR("ZLIFT  ",zlift_activated);*/
  } // retract()

#endif // FWRETRACT

studiodyne, 2017-07-11:
here my new code

...cpp
   #if ENABLED(FWRETRACT)

   void retract(const bool retracting, const bool swapping = false) {

    static float hop_height;
    static bool zlift_activated=0;
    // debugging
    /*SERIAL_ECHOLNPAIR("hop height ",hop_height);
    SERIAL_ECHOLNPAIR("retracting ",retracting);
    SERIAL_ECHOLNPAIR("swapping ",swapping);
    SERIAL_ECHOLNPAIR("active extruder ",active_extruder);
    SERIAL_ECHOLNPAIR("retracted 0 ",retracted[0]);
    SERIAL_ECHOLNPAIR("retracted 1 ",retracted[1]);
    SERIAL_ECHOLNPAIR("retracted_swap0 ",retracted_swap[0]);
    SERIAL_ECHOLNPAIR("retracted_swap1 ",retracted_swap[1]);
    SERIAL_ECHOLNPAIR("current_position z ",current_position[Z_AXIS]);
    SERIAL_ECHOLNPAIR("ZLIFT  ",zlift_activated);*/
    
    // cause the firmware make no swap status to all extruders
    // need to allow the swap load and unload in singlenozzle
    if (retracted_swap[0]||retracted_swap[1] ) {       
      retracted_swap[0] = 1;retracted_swap[1] = 1; }

  
     //IF ALREADY RETRACTED and retract asked ---then return because not needed and if retract_zlift                             exist of course
if(retracting &&  zlift_activated ) 
return;
//IF ALREADY RETRACTEDDOWN and retract asked ---then return because not needed and if retract_zlift exist of course
if(!retracting &&  !zlift_activated && retract_zlift) 
return;

//can retract up if  retracted and swap  active  and reinitilise the status lost ' cause the firmware low the swap status 
    if(retracting &&!swapping )
    {
      if ((retracted_swap[0]==1)||(retracted_swap[1]==1))    
   {retracted[active_extruder]=0; retracted_swap[0] = 1;retracted_swap[1] = 1;
   return;}
   };
    
// can retract the other ' for bad slicer or batch'
    if(!retracting && !swapping)
    {
      if (!retracted[active_extruder])
      retracted[active_extruder]=1;retracted[!active_extruder]=0;
    }

//if swapped no else possibility to unswap because if a g10 appear , the retracting status will be high and the 
//load/unload of the swapping process is down for eternity lol
 // if( retracted_swap[0]||retracted_swap[1]&& !swapping)  return; 

    const float old_feedrate_mm_s = feedrate_mm_s;
    

    set_destination_to_current();

    if (retracting) {// IF ZLIFT NOT NEEDED TO RETRACT ANY E=1;

      feedrate_mm_s = retract_feedrate_mm_s;
      current_position[E_AXIS] += (swapping ? retract_length_swap : retract_length) / volumetric_multiplier[active_extruder];
      sync_plan_position_e();
      prepare_move_to_destination();

      if (retract_zlift > 0.01) {
        hop_height = current_position[Z_AXIS];
        // Pretend current position is lower
        current_position[Z_AXIS] -= zlift_activated ?  0: retract_zlift;// ONLY ONE ZLIFT ABSOLUTE CONDITION
        zlift_activated=1;
        SYNC_PLAN_POSITION_KINEMATIC();
        // Raise up to the old current_position
        prepare_move_to_destination();
      }
    }
    else {

      // If the height hasn't been lowered, undo the Z hop
      if (retract_zlift > 0.01 && hop_height <= current_position[Z_AXIS]) {
        // Pretend current position is higher. Z will lower on the next move
        //current_position[Z_AXIS] += retract_zlift;
        current_position[Z_AXIS] += zlift_activated ?  retract_zlift :0;// ONLY ONE ZLIFT ABSOLUTE CONDITION
        SYNC_PLAN_POSITION_KINEMATIC();
        // Lower Z
        prepare_move_to_destination();
        
      }
      if (retract_zlift > 0.01 && hop_height > current_position[Z_AXIS]) {
        //If on a slicer with no firmware retract compatibility and 
        // you use a batch processing , sometimes , a G10 can appear before
        // a Z MOVE ' simplify3d' and it is need to actualise the hop_height
        current_position[Z_AXIS] += zlift_activated ?  retract_zlift :0;// ONLY ONE ZLIFT ABSOLUTE CONDITION
        SYNC_PLAN_POSITION_KINEMATIC();
        prepare_move_to_destination();
      hop_height = current_position[Z_AXIS];
      }
      //adding of a retract multipilier ' to add in your config.h ' 
      // To increase swapping recover by multiplying the recover feedrate
      feedrate_mm_s = swapping ? RETRACT_RECOVER_FEEDRATE_SWAP : retract_recover_feedrate_mm_s ;
      
      //feedrate_mm_s = retract_recover_feedrate_mm_s ; Original feedrate
      const float move_e = swapping ? retract_length_swap + retract_recover_length_swap : retract_length + retract_recover_length;
      current_position[E_AXIS] -= move_e / volumetric_multiplier[active_extruder];
      sync_plan_position_e();

      // Recover E
      prepare_move_to_destination();
      zlift_activated=0;
      
    }

    feedrate_mm_s = old_feedrate_mm_s;
    
    retracted[active_extruder] = retracting;
   
       
// to set the same state ' swapping'  to all the extruders
// Because of Zlift ' very usefull in dual color ',
// all extruders must be synchronised on ONLY ONE ZLIFT
      if (swapping ) {          
      if (retracting){retracted_swap[0] = 1;retracted_swap[1] = 1; }
      else {retracted_swap[0] = 0;retracted_swap[1] = 0;  }
      }
      

      //debugging
    /*SERIAL_ECHOLNPAIR("-hop height ",hop_height);
    SERIAL_ECHOLNPAIR("-retracting ",retracting);
    SERIAL_ECHOLNPAIR("-swapping ",swapping);
    SERIAL_ECHOLNPAIR("-active extruder ",active_extruder);
    SERIAL_ECHOLNPAIR("-retracted 0 ",retracted[0]);
    SERIAL_ECHOLNPAIR("-retracted 1  ",retracted[1]);
    SERIAL_ECHOLNPAIR("-retracted_swap0 ",retracted_swap[0]);
    SERIAL_ECHOLNPAIR("-retracted_swap1 ",retracted_swap[1]);
    SERIAL_ECHOLNPAIR("-current_position z ",current_position[Z_AXIS]);
    SERIAL_ECHOLNPAIR("ZLIFT  ",zlift_activated);*/
  } // retract()

#endif // FWRETRACT

studiodyne, 2017-07-11:
Can you give me the way to make a recording of all i need during a print

Like x y z e ,  and the  machine movement of the planner
To know when , and how , and why , an overextrusion or ' a bad z ' comes later in the print

thks

studiodyne, 2017-07-11:
i will deactivate data serial transmiting , and make clean all conditions
to see what happen

studiodyne, 2017-07-12:
less process conditionals

// cause the firmware make no swap status to all extruders
// need to allow the swap load and unload in singlenozzle
if (retracted_swap[0]||retracted_swap[1] ) {       
      retracted_swap[0] = 1;retracted_swap[1] = 1; }
      
 if(retracting) 
    {
  //IF ALREADY RETRACTED and retract asked ---then return because not needed and if retract_zlift exist of course
  if( zlift_activated ) return; 
  //can retract up if  retracted and swap  active  and reinitilise the status lost ' cause the firmware low the swap status
  if(!swapping )
    {if (retracted_swap[0]||(retracted_swap[1])){retracted[active_extruder]=0; retracted_swap[0] = 1;retracted_swap[1] = 1;return;}}
    }
        else
    {
  //IF ALREADY RETRACTEDDOWN and retract asked ---then return because not needed and if retract_zlift exist of course
  if(  !zlift_activated && retract_zlift) return;
  // can retract the other ' for bad slicer or batch'
  if(!swapping)
    {
      if (!retracted[active_extruder])
      retracted[active_extruder]=1;retracted[!active_extruder]=0;// can retract the other ' for bad slicer or batch'
    }
    }

studiodyne, 2017-07-12:
The bad news if , i'm sure the low ressources make crash the print , i use all options , bltouch , mesh , advance, fwretract ...
When i print with direct gcode ' no g10/g11' , all is perfect
When i take the same file and i batch to Firmware Retract 
It works on the 20 /30 first layers , after , the print have a too low z , overextrusion , print dye

Do you think it's the firmware retract bugged , or very low ressources
Tomorrow il will try at low speed , i will see

Roxy-3D, 2017-07-12:
I doubt it is too low of resources.   If you turn on the M100 Free Memory Checker, you will know for sure.  Initialize the M100 Free Memory Checker after boot up...  Let it print 10 layers.   And see how much free memory it finds.

I don't know...   But this kind of feels like how a 'Bad Pointer' problem feels.  It would be worth while to make sure every where you are passing pointers, you are doing it correctly.

studiodyne, 2017-07-12:
My code is just a behavior code.
Some confitionals and its all 

I have  800 steps my z axis ...and 2 motors 
My feedrate limit is 12 mms ... if more i loose steps .. may be i loose steps because overloaded with too many zlift....
I don't know ..
HELP :-)

fiveangle, 2017-07-12:
800 steps on Z does not seem like much, but you should be able to drop this in half to 400 and still get excellent layer height / bed levelling control.  I dropped my PB from 2020 to 507 steps/mm and can now run aggressive Z-lift with LIN_ADVANCE, LCD, and MBL (still working on getting BILINEAR working) at 120mm/s with 1,000mm/s^2 print acceleration and no print slowdown set in slicer.

Calcs show it should be hitting 120mm/s through the middle 2" radius on bed traverse paths, and it definitely prints around 2x speed of standard 40mm/s speed with same quality.  Nice thing is it doesn't realy increase noise when it's just the Z-axis since it's still high steps/mm (400 in your case).

Give it a try if your steppers support going 1/2 down from where you are now.

studiodyne, 2017-07-12:
Thks 👍  
Please explain me how to decrease the step per mm ,  i have to set it , on the motor driver of the motherboard , or direct in the firmware , i'm beginner in this type of elecrtronics
If i want to have 400 steps for my z axis '  two motors on same drivers wired together ' , how i do this  ?
Thks

fiveangle, 2017-07-12:
I don't see your config* files anywhere here.  Please share them (you may have to zip them to upload).

studiodyne, 2017-07-12:
I'm talking about 'microstepping' , to reduce the number of pulses to drive a motor
If i reduce the pulse i reduce the processing and time to process

It's not in the firm , it's some switch on the driver board  ? Isn't it  ?

studiodyne, 2017-07-12:
1/4 of revolution = 1mm on my z axis
Stepmotors have 200 steps
If i set no microstepping , i have 50 steps for 1mm and 0.05 mm resolution 
It's perfect for printing , printing don't need more , high resolution not needed , so 0.05 if perfect
I want to set this resolution
I have switch under the driver board
Possible ?

[16 more replies]

closed 2018-02-06 by thinkyhead