MarlinFirmware/Marlin #7200: E0_AUTO_FAN_PIN don't work

htaxil, 2017-07-01:
Hi,

I can'n operate my HotEnd fan automatically with parameter E0_AUTO_FAN_PIN.

Motherboard set : 
``` cpp
#define MOTHERBOARD BOARD_RAMPS_14_EFF
```

Pin binded : 
``` cpp
#elif ENABLED(IS_RAMPS_EFF)                    // Hotend, Fan, Fan
  #define FAN_PIN        RAMPS_D9_PIN
  #define FAN1_PIN       RAMPS_D8_PIN
```

Fan Pin assigned : 
``` cpp
#define E0_AUTO_FAN_PIN FAN1_PIN
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 40
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed
```

But nothing happens on the D8 output when I exceed the temperature threshold ... 
For information I tested on two circuits (Arduino + RAMPS), it's the same result.

Bob-the-Kuhn, 2017-07-01:
`#define E0_AUTO_FAN_PIN FAN1_PIN` works as expected in pins_RAMPS.h but not in Configuration.h.

`#define E0_AUTO_FAN_PIN 8` works as expected in both places.

Let me know if that works or not.

htaxil, 2017-07-02:
_#define E0_AUTO_FAN_PIN 8_ do not work

I do not think that FAN1_PIN is not usable in the configuration_adv.h file since it is used to control the cooling of the drivers.

```cpp
/**
 * Controller Fan
 * To cool down the stepper drivers and MOSFETs.
 *
 * The fan will turn on automatically whenever any stepper is enabled
 * and turn off after a set period after all steppers are turned off.
 */
//#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  //#define CONTROLLER_FAN_PIN FAN1_PIN  // Set a custom pin for the controller fan
  #define CONTROLLERFAN_SECS 60          // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255        // 255 == full speed
#endif
```

Bob-the-Kuhn, 2017-07-02:
Please ZIP up your config files and drop the ZIP file on your reply.  I'll play with my RAMPS unit & get the fans setup the way you want.

What fans do you have and how do you want them to operate?

Sounds like you want logical pin 9 to cool the E0 extruder.

Is there a second fan in your system?

htaxil, 2017-07-03:
Here the zip file of my configuration.

I've got two fans : 
 - One to ventilate the print : plugged on D9 (and works well)
 - One to cool the heat sink of the hotend that I want to drive automaticly with the parameter E0_AUTO_FAN_PIN binded to the D8 output. (but do not work)

For information, I have configured this function on another printer with a RUMBA board and Marlin 1.1.0-RCBugFix, and it works well.

[Marlin 1.1.3.zip](https://github.com/MarlinFirmware/Marlin/files/1120486/Marlin.1.1.3.zip)

Bob-the-Kuhn, 2017-07-04:
The D8 pin is fully functional.  When the extruder temperature goes above EXTRUDER_AUTO_FAN_TEMPERATURE then D8 turns on.  When the extruder temperature drops below then D8 is off.

I didn't make any changes to your files.

htaxil, 2017-07-05:
Hi Bob,

At the end, that works !

Thank you very much for having tested my configuration and validated that it worked. It allowed me to stop looking for the cause of the problem in the configuration files !
The problem was that D8 gets it's power from the 2nd input connector, normally dedicated to the power supply of the heating bed, and that I had not wired!

I wired the two inputs in parallel and everything works perfectly!

closed 2017-07-05 by Bob-the-Kuhn