MarlinFirmware/Marlin #7202: Is probe repeatability test valid with PROBE_MANUAL ?

fiveangle, 2017-07-01:
If so, it's failing:

```
Marlin_main.cpp: In function 'void gcode_M48()':
Marlin_main.cpp:6822: error: 'probe_pt' was not declared in this scope
     const float t = probe_pt(X_probe_location, Y_probe_location, stow_probe_after_each, verbose_level);
                                                                                                      ^
Marlin_main.cpp:6944: error: expected primary-expression before ')' token
     if (STOW_PROBE()) return;
                     ^
Using library U8glib at version 1.19.1 in folder: /Users/speedster/Documents/Arduino/libraries/U8glib 
Using library Wire at version 1.0 in folder: /Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/avr/libraries/Wire 
'probe_pt' was not declared in this scope
```

I would think it isn't valid, and the [repeatability probe test](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/SanityCheck.h#L602-L604) within SanityCheck.h needs to be moved into the PROBE_DEFINED section of the conditional block so it will thrown an error.

Does someone know ?

thinkyhead, 2017-07-02:
A simple test for the combination of repeatability test plus probe_manually would work. Maybe `HAS_REAL_PROBE` and `CAN_PROBE` should replace `HAS_BED_PROBE` — each in the appropriate instance.

thinkyhead, 2017-07-02:
Oh wait, I see `HAS_BED_PROBE` already excludes `PROBE_MANUALLY`…

thinkyhead, 2017-07-02:
#7212 should prevent this.

closed 2017-07-02 by thinkyhead

Roxy-3D, 2017-07-02:
> Oh wait, I see HAS_BED_PROBE already excludes PROBE_MANUALLY…

This feels like the wrong direction.   If a machine is setup to do manual probes...   It does have a way to measure points.   It isn't automatic.  It is a real nuisance.  But it does have a bed probe.

And here is why I'm making the point...   I think going forward we need to embrace what ever probes are available.   We can do automatic probing with 'real' probes.   But we can still do manual probing in locations that the real probe can not reach.   

Maybe a name change of HAS_BED_PROBE to HAS_MACHINE_CONTROLLED_PROBE or something similar makes sense?   Going forward, we probably want to be able to use what ever probes are available.

fiveangle, 2017-07-02:
@Roxy-3D - I see your logic here, and normally agree, but there's also the concept of letter of the law, and spirit of the law.  Is there any universe where a user would sit in front of a probe with a paper and beleive the values were different between each move of the nozzle ?  It just doesn't make sense in this case.

If a users mechanicals are failing and they want to see if the values change between movements of the nozzle, they will see it immediatley during the measurement, so running the test as though it were autonomous, and comparing the results afterward, doesn't add any value here.

Preventing this feature with PROBE_MANUAL add more benefit by helping guide unsavvy users to better understand the feature and what it is intended for if they enable with PROBE_MANUAL and observe the sanity check warning.  For the 1-in-1000 savvy users who may have some minor mechanical issue on their bot and they wish to run the process for OCD accademic reasons to see if their own tactile skills are up to snuff is really outside the scope, don't you think ?

[I'm now imagining the new latest game sweeping the nation... PROBE_MANUAL REPEATABILITY_TEST championships - How accurate is your business card ? :]