MarlinFirmware/Marlin #7203: EEPROM_CHITCHAT PROGMEM usage comment in Configuration.h

fiveangle, 2017-07-01:
The claim [here in Configuration.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Configuration.h#L994-L997) is ~1700bytes are saved, but I'm seeing just 86bytes (no change in SRAM usage):

```
#if ENABLED(EEPROM_SETTINGS)
  // To disable EEPROM Serial responses and decrease program space by ~1700 byte: comment this out:
  #define EEPROM_CHITCHAT // Please keep turned on if you can.
#endif
```

Could it be compiler efficency improved so much as to make this comment moot ?

thinkyhead, 2017-07-02:
A very outdated comment. Keep looking and I bet you will find several more.

closed 2017-07-02 by thinkyhead

fiveangle, 2017-07-02:
DISABLE_M503 behavior Confirmed - thanks Scott !