MarlinFirmware/Marlin #7213: 'CUSTOM_BOOTSCREEN_TIMEOUT' was not declared in this scope

Bergerac56, 2017-07-02:
Hello,

The last bugfix of today does not compile correctly for me if I do not add "#define CUSTOM_BOOTSCREEN_TIMEOUT   2500" just after line (13036 lcd_init();) in marlin_main.cpp (not sure it is the best place to add it...)

benlye, 2017-07-02:
Related to #7211?

petrzjunior, 2017-07-02:
@Bergerac56  What configuration file are you using? Some of the example ones or custom? Did you add custom booscreen (is `SHOW_CUSTOM_BOOTSCREEN` enabledin your config)?

Bergerac56, 2017-07-02:
Yes, the "bugfix" version is related to #7211. I use a quite modified configuration.h and configuration_adv.h and have developped a small bootscreen. I have different printers. One is a "Atlas Jacbot" I modified (other motors, hotend adaptation, electronic improvement,etc...) I use this printer for my tests.

When I try a new "bugfix" I compare the modified files with those of the "bugfix" version. And do the adaptations when needed (hoping always not to have missed something ;) )

I attach a zip of the modified files: 
[JacBot OdB.zip](https://github.com/MarlinFirmware/Marlin/files/1117810/JacBot.OdB.zip)

petrzjunior, 2017-07-02:
I replicated the problem, with newest `bugfix-1.1.x` and any example config using custom Bootscreen. There must be an `#include` problem, pasting `#include "_Bootscreen.h"` in `Marlin_main.cpp` fixes the bug.

Bergerac56, 2017-07-02:
Yes This is what I did too. I pasted the #include just after the header "lcd_init()

thinkyhead, 2017-07-03:
If omitted, I'll have it use 2500.

petrzjunior, 2017-07-03:
#7219 fixed compile error, but editing the _Bootscreen.h TIMOUT does nothing. I guess #includes must be reorganized somehow.

Bergerac56, 2017-07-04:
In the last "bugfix" of today (Latest commit df87278 13 hours ago) the LCD problem is still not completely solved. The program compiles but the delay is still  not respected (as already described above).

In order to minimize the modifications, and without beeing sure that it is the best place, I propose to add (marlin_main.cpp: line 13050)
   #include "_Bootscreen.h"

and to drop  (marlin_main.cpp: line 13043 to 13045)
  #ifndef CUSTOM_BOOTSCREEN_TIMEOUT
    #define CUSTOM_BOOTSCREEN_TIMEOUT 2500
  #endif

This way, all is fine at least for me (delay respected, no compilation errors with or without bootscreen)

The new "lcd_init();" would be then:

  lcd_init();
  
  #if ENABLED(SHOW_BOOTSCREEN)
    #if ENABLED(DOGLCD)                           // On DOGM the first bootscreen is already drawn
      #if ENABLED(SHOW_CUSTOM_BOOTSCREEN)
        **#include "_Bootscreen.h"                  // Include custom bootscreen details**
        safe_delay(CUSTOM_BOOTSCREEN_TIMEOUT);    // Custom boot screen pause
        lcd_bootscreen();                         // Show Marlin boot screen
      #endif
      safe_delay(BOOTSCREEN_TIMEOUT);             // Pause
    #elif ENABLED(ULTRA_LCD)
      lcd_bootscreen();
      #if DISABLED(SDSUPPORT)
        lcd_init();
      #endif
    #endif
  #endif

Bergerac56, 2017-07-05:
Still not definitively solved in the last bugfix ( ef94ddf)

Bergerac56, 2017-07-07:
Hello Thinkyhead,

Could you please modify lcd_init() as proposed above? 

Thanks a lot

closed 2017-08-17 by Bergerac56