MarlinFirmware/Marlin #7222: Help on Set the boundaries for probing

derboehsevincent, 2017-07-03:
Hi Guys, 
just a quick question, how do I calculate the correct values for the 
*_BED_POSITION if I have

#define X_PROBE_OFFSET_FROM_EXTRUDER -7 
#define Y_PROBE_OFFSET_FROM_EXTRUDER 70

Lord-Quake, 2017-07-03:
I see you haven't gotten an answer so let me explain how I determine the values.

First I take a piece of paper and mark a dot on it.
I then place it square on the build plate.
Next I lower the extruder manually until it slightly touches the paper.
I then place the dot that is on the paper right under the nozzle.
Holding the paper I make a outline around the sensor.
I remove the paper from the build plate and find the center of the sensor outline and mark it.
Then I make parallel lines (horizontal and vertical) between the center of the sensor and the dot from the nozzle.
Lastly I measure the distances and enter the values in the #define configuration you posted.

fiveangle, 2017-07-03:
This is how I do it in my `Configuration.h`:

```
  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION (X_PROBE_OFFSET_FROM_EXTRUDER <= MIN_PROBE_EDGE ? X_MIN_POS + MIN_PROBE_EDGE : X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER)
  #define RIGHT_PROBE_BED_POSITION (X_PROBE_OFFSET_FROM_EXTRUDER >= -MIN_PROBE_EDGE ? X_MAX_POS - MIN_PROBE_EDGE : X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER)
  #define FRONT_PROBE_BED_POSITION (Y_PROBE_OFFSET_FROM_EXTRUDER <= MIN_PROBE_EDGE ? Y_MIN_POS + MIN_PROBE_EDGE : Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER)
  #define BACK_PROBE_BED_POSITION (Y_PROBE_OFFSET_FROM_EXTRUDER >= -MIN_PROBE_EDGE ? Y_MAX_POS - MIN_PROBE_EDGE : Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER)
```

fiveangle, 2017-07-03:
Which reminds me I meant to submit a PR to add these calcs to `Conditionals.h` and then make manual configuration of these optional in `Configuration.h`.  Thanks for the reminder :)

fiveangle, 2017-07-03:
#7228

closed 2018-02-09 by thinkyhead