MarlinFirmware/Marlin #7223: M851 has no effect (coreXY/UBL/bugfix-1.1x)

joris57, 2017-07-03:
```
Send: M851 Z-0.4
Recv: echo:Z Offset : 0.00
Recv: ok
```
Is this because I'm expected to use G29 P6 C? If so, where could I have found this informaiton?

Roxy-3D, 2017-07-03:
The most likely reason for M851 not affecting things is you may not be updating your EEPROM with new values.   If you do a M851 be sure to do a M500 after validating the new setting.

Bob-the-Kuhn, 2017-07-03:
How old is your Marlin software?

For a short time Marlin wasn't handling the Z parameter correctly.  Please try the latest [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x) and see if that fixes the problem.

joris57, 2017-07-04:
The latter was indeed the cause. Updated to the latest bugfix and M851 works again as expected.

closed 2017-07-04 by joris57