MarlinFirmware/Marlin #7224: Pin re-assignment leads to unexpected behaviour

Eloque, 2017-07-03:
I am not sure this is actually a bug/issue, but it is something that is odd. Perhaps just a error on my part. On my RAMPS board, I can't use the standard Z-Axis pins so for quite a while I've been using the unused extruder pins to drive the Z-Axis. This is done by a simple pin reassignment in pins.h in an old version of Marlin.

```
// Otherwise the X moves the wrong way
#define X_MIN_PIN           2 // was 3, is 2

// Since the Z-socket is messed up, I can't use these for much
//    #define Z_STEP_PIN         46
//    #define Z_DIR_PIN          48
//    #define Z_ENABLE_PIN       62
    #define Z_MIN_PIN          18
    #define Z_MAX_PIN          -1

// Since I only have 1 extruder, I might as well use these for Z axis
//    #define E1_STEP_PIN        36
//    #define E1_DIR_PIN         34
//    #define E1_ENABLE_PIN      30

// These lines are the hack
    #define Z_STEP_PIN        36
    #define Z_DIR_PIN         34
    #define Z_ENABLE_PIN      30

    #define E1_STEP_PIN       46
    #define E1_DIR_PIN        48
    #define E1_ENABLE_PIN     62
// These lines are the end hack    

```

Now, in the latest version of Marlin, 1.1.3, Git Checkout yesterday, I needed to reassign those pins again, now in their own .h file

```
#define Z_STEP_PIN         36 // 46
#define Z_DIR_PIN          34 // 48
#define Z_ENABLE_PIN       30 // 62
#define Z_CS_PIN           44 // 40

#define E0_STEP_PIN        26
#define E0_DIR_PIN         28
#define E0_ENABLE_PIN      24
#define E0_CS_PIN          42

//#define E1_STEP_PIN        36
//#define E1_DIR_PIN         34
//#define E1_ENABLE_PIN      30
//#define E1_CS_PIN          44
```

But, this doesn't work anymore. First off, the Z-axis doesn't move, only gives a small jittery jerk when activated. And, this is the weird thing, when turning the printer on, and homing the X-Axis for the first time, the Z motors activate and jam a bit, then the printer resumes homing as usual. 

Right now, I can't use the printer using this version of Marlin. The older version, 1.0.2, does work, so electronically and physically the printer is seemingly in order. It just software. Can anyone advise?

FHeilmann, 2017-07-03:
From your text, I assume that you haven't used Marlin 1.1.x previously? If so, make sure you do the following things:

**Lower your speed, acceleration and jerk values of your z-axis significantly and then slowly increase them again until you are satisfied:**
The motion code underwent significant changes so your values may be too high. Also make sure that your values have the right unit (confusing mm/**s** and mm/**min** can cause issues here)

**Make sure your EEPROM is clean/ initialized properly:**
After your printer boots up, perform a `M502` to force loading the values from your configuration and then `M500` to save them to EEPROM. The layout changed quite a bit between Marlin `1.0.x` and Marlin `1.1.x` so there may be some issues resulting in erratic behavior.

From your description I am quite certain that, if you previously used Marlin `1.0.x`, your acceleration and/or speed and/or jerk values for your Z-Axis are too high, causing the Z-steppers to lock up/jam instead of running smoothly.  While you're at it, als make sure your `steps/mm` are correct.

Eloque, 2017-07-03:
I will do all that. Your assumption about using older Marlin earlier is correct. I will do the initialisation first. 
`
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80, 80, 4000, 495}
#define DEFAULT_MAX_ACCELERATION      {9000,9000,100,300} 
#define DEFAULT_ZJERK                 0.4     // (mm/sec)
`
Are the ones that I have now. Are those the ones that need revising?

FHeilmann, 2017-07-03:
your 9000 seems very high but thats for X and Y so it shouldnt be an issue. Assuming that you use a M5 rod for Z axis, the steps/mm are okay.

Which .h file did you use to make your changes?

FHeilmann, 2017-07-03:
your values seem fine to me. In which .h file are you making your changes?

Eloque, 2017-07-03:
I used to change it in pins.h, I now changed it in pins_RAMPS.h. 
I have not redefined the E1 Pins, I will do that. 

One thing I did find out, if I don't redefine like this
`#define Z_STEP_PIN         36 // 46`
The short burst to the Z motors doesn't happen at startup. So there is that. I will try the solution you suggested once I am back at the workbench.

Eloque, 2017-07-03:
The 'just for kicks' suggestion worked. Reassigning the pins removes the Z axis move on homing X.
Now trying to clean EEPROM.

Eloque, 2017-07-03:
Cleaning EEPROM yields this;

>>>M502
```
SENDING:M502
echo:Hardcoded Default Settings Loaded
>>>M500
SENDING:M500
Error:EEPROM disabled
[ERROR] Error:EEPROM disabled
```

Eloque, 2017-07-03:
Changing max feedrate solved the lockup/jam issue it seems.

`#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 }`
to
`#define DEFAULT_MAX_FEEDRATE          { 300, 300, 2, 25 }`
works
`#define DEFAULT_MAX_FEEDRATE          { 300, 300, 3, 25 }`
works as well

However, the issue remains, that on homing X or Y, Z gets raised. But, I beginning to think that's a feature not an issue.

FHeilmann, 2017-07-03:
`#define Z_CLEARANCE_BETWEEN_PROBES` is your friend here ;)

FHeilmann, 2017-07-03:
You have EEPROM disabled so you can't get a bad EEPROM anyhow :)

Eloque, 2017-07-04:
Reassigning the pins and changing the speed works and solves problems.
Issues can be closed, not a bug.

closed 2017-07-04 by Eloque